firmadyne/firmadyne #1: dict object has not attribute 'iteritem'

misterch0c, 2016-02-25:
Hi there,

firstly that project looks great (: but I'm trying to use it in Archlinux. So I dowloaded the binaries with your scripts, installed few dependency and on step 5 I get this error:

```
└┼─$─┤▶ ./scripts/tar2db.py -i 1 -f ./images/1.tar.gz
Traceback (most recent call last):
  File "./scripts/tar2db.py", line 100, in <module>
    main()
  File "./scripts/tar2db.py", line 97, in main
    process(iid, infile)
  File "./scripts/tar2db.py", line 75, in process
    file2oid = [(fdict[h], oid) for (h, oid) in oids.iteritems()]
AttributeError: 'dict' object has no attribute 'iteritems'

```

So apparently that object would be empty? Any idea why? I'm using Python 3 btw, with Python2.7 I got a "cored.

I get a different error with Python 2
```

└┼─$─┤▶ python2.7 scripts/tar2db.py -i 1 -f ./images/1.tar.gz
Traceback (most recent call last):
  File "scripts/tar2db.py", line 100, in <module>
    main()
  File "scripts/tar2db.py", line 97, in main
    process(iid, infile)
  File "scripts/tar2db.py", line 77, in process
    insertObjectToImage(iid, file2oid, links, cur)
  File "scripts/tar2db.py", line 61, in insertObjectToImage
    for x in links])
psycopg2.IntegrityError: insert or update on table "object_to_image" violates foreign key constraint "object_to_image_oid_fkey"
DETAIL:  Key (oid)=(1) is not present in table "object".

```

ddcc, 2016-02-25:
Hmm, is there anything in the `objects` table of the database? You can probably do:
* `psql -h 127.0.0.1 -U firmadyne`
* `SELECT * FROM objects;`

For the moment, you can probably skip the `tar2db` step. The only impact will be that the `webAccess.py` analysis won't work, because the database won't be loaded with the list of objects (e.g. files) in the tarball.

I'm currently traveling, but I'll take a closer look when I get back.

misterch0c, 2016-02-25:
Sorry I'm not used to postgres. And I feel it's definitely a postgresql issue.

```
dyne/database├────
└┼─$─┤▶ psql -h 127.0.0.1 -U firmadyne
psql: FATAL:  database "firmadyne" does not exist
```
but...
```
└┼─$─┤▶ sudo -u postgres createdb -O firmadyne firmware
could not change directory to "/home/unknown/code/firmadyne": Permission denied
createdb: database creation failed: ERROR:  database "firmware" already exists
```
And you said I can skip step 5? But then it looks like I cannot continue because I need `1`
```

└┼─$─┤▶ sudo ./scripts/makeImage.sh 1
[sudo] password for unkn0wn: 
Querying database for architecture... mipseb
----Running----
----Copying Filesystem Tarball----
Error: Cannot find tarball of root filesystem for 1!

```

ddcc, 2016-02-25:
Sorry, I think I gave you the wrong PostgreSQL line earlier; it should be:
*  `psql -h 127.0.0.1 -U firmadyne firmware`

Hmm, that's strange. Is there a `1.tar.gz` file under `firmadyne/images`? For the issue you're referring to in Kali, the `chroot` is using `busybox` provided by your distribution (typically for x86 or x86_64), not any embedded `busybox` provided by the firmware. From a regular terminal, does `which busybox` provide a usable `busybox` binary?

misterch0c, 2016-02-26:
Indeed that works with that command but the table seem empty. Gonna investigate this. And yes the .tar.gz is there

In Kali:
```
root@kali:~# which busybox 
/bin/busybox
```

ddcc, 2016-02-26:
Ok, thanks for the information. I tried spinning up a Arch system to check, but it's too different from Debian-based systems, so I don't think it can be supported without spending significant time on it, which I don't have. For example, it seems that `ifconfig` and `vconfig` have been deprecated in favor of `ip` in Arch, lots of dependencies like `kpartx` (`multipath-tools`) appear to be orphaned or unavailable in Arch, and `binwalk` doesn't support Arch either.

I'll take a look at Kali, but Ubuntu 14.04 should definitely work.

ddcc, 2016-02-27:
Ok, I've figured out the cause of the `/busybox` error. The package that needs to be installed should be `busybox-static` instead of `busybox`. Among others, I'm testing some fixes in the `bugfix` branch, those will be merged into `master` after I've tested them.

misterch0c, 2016-02-28:
Indeed installing `busybox-static` fixed the problem and after a git pull everything works as expected (: thank you

closed 2016-02-28 by misterch0c