firmadyne/firmadyne #2: run.sh is missing in scratch

mr-panda, 2016-02-26:
i am getting this error
chroot: failed to run command ‘/busybox’: No such file or directory

when i trying to internetwork
Querying database for architecture... Password for user firmadyne: 
mipseb
Running firmware 1 for 60 secs...
qemu-system-mips: pci_add_option_rom: failed to find romfile "efi-pcnet.rom"
qemu-system-mips: pci_add_option_rom: failed to find romfile "efi-pcnet.rom"
qemu-system-mips: pci_add_option_rom: failed to find romfile "efi-pcnet.rom"
qemu-system-mips: pci_add_option_rom: failed to find romfile "efi-pcnet.rom"
qemu-system-mips: pci_add_option_rom: failed to find romfile "vgabios-cirrus.bin"

and when i am going ahead with tutorial and trying to creat a tap device 
i am getting the error
bash: ./scratch/1/run.sh: No such file or directory

any guidance will be of great help

x70, 2016-02-26:
same here.
following tutorial for netgear, after executing inferNetwork.sh:
qemu: terminating on signal 2 from pid 20996
run.sh is missing too

ddcc, 2016-02-26:
What distribution are you running? Can you post the output of `ls -R ./firmadyne/scratch` and `find ./firmadyne -name run.sh`? Does `which busybox` return a valid path?

ddcc, 2016-02-27:
Ok, I've figured out the cause of the `/busybox` error. The package that needs to be installed should be `busybox-static` instead of `busybox`. Among others, I'm testing some fixes in the `bugfix` branch, those will be merged into `master` after I've tested them.

closed 2016-03-10 by ddcc

azwicker, 2016-03-15:
I've the same issue with the romfile. I've installed the packages vgabios and ipxe-qemu which contain the missing files but the error message remains. Any idea? I'm using Ubuntu 14.4.

ddcc, 2016-03-16:
I'm unable to reproduce the problem with QEMU `Debian 2.0.0+dfsg-2ubuntu1.22)`, but I believe this may be a packaging problem on some Debian-based systems. Does installing the `qemu-system-x86` and `seabios` packages fix the issue?

azwicker, 2016-03-17:
Thanks for the tip. The error message goes away after installing qemu-system-x86 and seabios.

huyanshuhan, 2016-03-22:
I reinstall the qemu by myself. The problem is solved.

azzeddineazzeddine, 2016-04-28:
Hi, i cant find the script run.sh too,
here is the result of your commands

`root@kali:~# ls -R ./firmadyne/scratch
./firmadyne/scratch:
1

./firmadyne/scratch/1:
1.tar.gz  image.raw  qemu.initial.serial.log
root@kali:~# find ./firmadyne -name run.sh
./firmadyne/scripts/run.sh
root@kali:~# ./firmadyne/scripts/run.sh
Error: Could not find 'firmadyne.config'!
root@kali:~#
`

Thanks