firmadyne/firmadyne #4: Binwalk 2.1.2b problem?

misterch0c, 2016-04-01:
Hi there,

in my issue last time I mentioned I was trying to set it up on arch. Now I'm doing it again on an up to date ubuntu and I have some issue with binwalk. I noticed that it's the 2.1.2b version, same as when I was trying on Archlinux. I only got it working with 2.1.1 in Kali linux.
Note that I can still extract the firmware with binwalk -e and it works fine with both versions.

Here's a log:
```
root@ce1754c6970d:/home/firmadyne# python2.7 sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk "WNAP320 Firmware Version 2.0.3.zip" images
>> Database Image ID: 1

/home/firmadyne/WNAP320 Firmware Version 2.0.3.zip
>> MD5: 51eddc7046d77a752ca4b39fbda50aff
>> Tag: 1
>> Temp: /tmp/tmpSSTucI
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True

Signature Exception: list index out of range
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 566, in main
    retval = self.run()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 177, in run
    self.scan_file(fp)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 160, in scan_file
    self.result(r=r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 461, in result
    getattr(self, dependency.attribute).callback(r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 174, in callback
    (extraction_directory, dd_file, scan_extracted_files) = self.extract(r.offset, r.description, r.file.path, size, r.name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 496, in extract
    output_directory = self.build_output_directory(file_name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 434, in build_output_directory
    subdir = basedir.split(self.directory)[1][1:]
IndexError: list index out of range
----------------------------------------------------------------------------------------------------

>> Recursing into archive ...

Signature Exception: list index out of range
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 566, in main
    retval = self.run()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 177, in run
    self.scan_file(fp)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 160, in scan_file
    self.result(r=r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 461, in result
    getattr(self, dependency.attribute).callback(r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 174, in callback
    (extraction_directory, dd_file, scan_extracted_files) = self.extract(r.offset, r.description, r.file.path, size, r.name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 496, in extract
    output_directory = self.build_output_directory(file_name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 434, in build_output_directory
    subdir = basedir.split(self.directory)[1][1:]
IndexError: list index out of range
----------------------------------------------------------------------------------------------------

>> Recursing into compressed ...
>> Cleaning up /tmp/tmpSSTucI...
```

ddcc, 2016-04-01:
Looks like a binwalk bug, I forwarded the report upstream over there.

misterch0c, 2016-04-02:
Fixed by latest commit on binwalk. Thx for forwading ddcc

closed 2016-04-02 by misterch0c