firmadyne/firmadyne #6: Can't find ./scratch/1/run.sh !

azzeddineazzeddine, 2016-04-28:
Hi, i cant find the script ./scratch/1/run.sh too,
here is the result of your commands
Thanks

`root@kali:~/firmadyne# ./scripts/inferNetwork.sh 1 `
`Querying database for architecture... Password for user firmadyne:  `
`mipseb`
`Running firmware 1: terminating after 60 secs...`
`main-loop: WARNING: I/O thread spun for 1000 iterations`

and
`root@kali:~/firmadyne# ./scratch/1/run.sh`
`-bash: ./scratch/1/run.sh: No such file or directory`

ddcc, 2016-04-28:
Is this with the tutorial or a different firmware image? Can you post the contents of the qemu.initial.serial.log file?

azzeddineazzeddine, 2016-05-02:
I used the same fimware

here is the content of qemu.initial.serial.log file :+1: +1: 

`[    1.012000] Intel(R) PRO/1000 Network Driver - version 7.3.21-k5-NAPI`
`[    1.012000] Copyright (c) 1999-2006 Intel Corporation.`
`[    1.012000] e1000e: Intel(R) PRO/1000 Network Driver - 1.0.2-k2`
`[    1.012000] e1000e: Copyright (c) 1999-2008 Intel Corporation.`
`[    1.012000] pcnet32.c:v1.35 21.Apr.2008 tsbogend@alpha.franken.de`
`[    1.016000] PCI: Enabling device 0000:00:0b.0 (0000 -> 0003)`
`[    1.016000] PCI: Setting latency timer of device 0000:00:0b.0 to 64`
`[    1.016000] pcnet32: PCnet/PCI II 79C970A at 0x1020, 52:54:00:12:34:56 assigned IRQ 10.`
`[    1.016000] eth0: registered as PCnet/PCI II 79C970A`
`[    1.016000] PCI: Enabling device 0000:00:12.0 (0000 -> 0003)`
`[    1.020000] PCI: Setting latency timer of device 0000:00:12.0 to 64`
`[    1.020000] pcnet32: PCnet/PCI II 79C970A at 0x1040, 52:54:00:12:34:57 assigned IRQ 10.`
`[    1.020000] eth1: registered as PCnet/PCI II 79C970A`
`[    1.020000] PCI: Enabling device 0000:00:13.0 (0000 -> 0003)`
`[    1.020000] PCI: Setting latency timer of device 0000:00:13.0 to 64`
`[    1.020000] pcnet32: PCnet/PCI II 79C970A at 0x1060, 52:54:00:12:34:58 assigned IRQ 10.`
`[    1.024000] eth2: registered as PCnet/PCI II 79C970A`
`[    1.024000] PCI: Enabling device 0000:00:14.0 (0000 -> 0003)`
`[    1.024000] PCI: Setting latency timer of device 0000:00:14.0 to 64`
`[    1.024000] pcnet32: PCnet/PCI II 79C970A at 0x1080, 52:54:00:12:34:59 assigned IRQ 11.`
`[    1.024000] eth3: registered as PCnet/PCI II 79C970A`
`[    1.028000] pcnet32: 4 cards_found.`
`[    1.028000] PPP generic driver version 2.4.2`
`[    1.028000] PPP Deflate Compression module registered`
`[    1.080000] ata2.01: NODEV after polling detection`
`[    1.080000] ata1.01: NODEV after polling detection`
`[    1.080000] ata1.00: ATA-7: QEMU HARDDISK, 2.5+, max UDMA/100`
`[    1.080000] ata1.00: 2097152 sectors, multi 16: LBA48`
`[    1.084000] ata1.00: configured for UDMA/33`
`[    1.092000] scsi 0:0:0:0: Direct-Access     ATA      QEMU HARDDISK    2.5+ PQ: 0 ANSI: 5`
`[    1.092000] sd 0:0:0:0: [sda] 2097152 512-byte logical blocks: (1.07 GB/1.00 GiB)`
`[    1.096000] sd 0:0:0:0: [sda] Write Protect is off`
`[    1.096000] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00`
`[    1.096000] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA`
`[    1.096000]  sda: sda1`
`[    1.104000] sd 0:0:0:0: [sda] Attached SCSI disk`
`[    1.104000] ata2.00: ATAPI: QEMU DVD-ROM, 2.5+, max UDMA/100`
`[    1.104000] ata2.00: configured for UDMA/33`
`[    1.108000] scsi 1:0:0:0: CD-ROM            QEMU     QEMU DVD-ROM     2.5+ PQ: 0 ANSI: 5`
`[    1.112000] PPP MPPE Compression module registered`
`[    1.112000] NET: Registered protocol family 24`
`[    1.112000] PPPoL2TP kernel driver, V1.0`
`[    1.112000] tun: Universal TUN/TAP device driver, 1.6`
`[    1.112000] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>`
`[    1.112000] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver`
`[    1.112000] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver`
`[    1.112000] uhci_hcd: USB Universal Host Controller Interface driver`
`[    1.112000] PCI: Setting latency timer of device 0000:00:0a.2 to 64`
`[    1.116000] uhci_hcd 0000:00:0a.2: UHCI Host Controller`
`[    1.116000] uhci_hcd 0000:00:0a.2: new USB bus registered, assigned bus number 1`
`[    1.116000] uhci_hcd 0000:00:0a.2: irq 11, io base 0x00001000`
`[    1.120000] usb usb1: configuration #1 chosen from 1 choice`
`[    1.124000] hub 1-0:1.0: USB hub found`
`[    1.124000] hub 1-0:1.0: 2 ports detected`
`[    1.124000] Initializing USB Mass Storage driver...`
`[    1.124000] usbcore: registered new interface driver usb-storage`
`[    1.124000] USB Mass Storage support registered.`
`[    1.128000] serio: i8042 KBD port at 0x60,0x64 irq 1`
`[    1.128000] serio: i8042 AUX port at 0x60,0x64 irq 12`
`[    1.128000] mice: PS/2 mouse device common for all mice`
`[    1.136000] rtc_cmos rtc_cmos: rtc core: registered rtc_cmos as rtc0`
`[    1.136000] rtc0: alarms up to one day, 242 bytes nvram`
`[    1.136000] i2c /dev entries driver`
`[    1.136000] piix4_smbus 0000:00:0a.3: SMBus Host Controller at 0x1100, revision 0`
`[    1.136000] sdhci: Secure Digital Host Controller Interface driver`
`[    1.136000] sdhci: Copyright(c) Pierre Ossman`
[... 17 lines cut ...]
`[    1.408000] Ebtables v2.0 registered`
`[    1.412000] 802.1Q VLAN Support v1.8 Ben Greear <greearb@candelatech.com>`
`[    1.412000] All bugs added by David S. Miller <davem@redhat.com>`
`[    1.412000] lib80211: common routines for IEEE802.11 drivers`
`[    1.412000] lib80211_crypt: registered algorithm 'NULL'`
`[    1.420000] rtc_cmos rtc_cmos: setting system clock to 2016-04-27 14:24:31 UTC (1461767071)`
`[    1.428000] input: AT Raw Set 2 keyboard as /devices/platform/i8042/serio0/input/input0`
`[    1.636000] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input1`
`[    2.076000] UDF-fs: No partition found (1)`
`[    2.076000] List of all partitions:`
`[    2.076000] 1f00            1024 mtdblock0 (driver?)`
`[    2.076000] 1f01            1024 mtdblock1 (driver?)`
`[    2.076000] 1f02            1024 mtdblock2 (driver?)`
`[    2.076000] 1f03            1024 mtdblock3 (driver?)`
`[    2.076000] 1f04            1024 mtdblock4 (driver?)`
`[    2.076000] 1f05            1024 mtdblock5 (driver?)`
[... cut ...]
`[    2.076000] 1f07            1024 mtdblock7 (driver?)`
`[    2.080000] 1f08            1024 mtdblock8 (driver?)`
`[    2.080000] 1f09            1024 mtdblock9 (driver?)`
`[    2.080000] 1f0a          120832 mtdblock10 (driver?)`
`[    2.080000] 0800         1048576 sda driver: sd`
`[    2.080000]   0801         1047552 sda1`
`[    2.080000] No filesystem could mount root, tried:  ext3 ext2 ext4 cramfs squashfs vfat iso9660 romfs udf`
`[    2.080000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(8,1)`

ddcc, 2016-05-02:
The filesystem image is bad. Try deleting the corresponding directory for that firmware from scratch, and restarting at the makeImage.sh step. Does this produce any warnings?

azzeddineazzeddine, 2016-05-02:
Thank's a lot !

closed 2016-05-02 by ddcc

ddcc, 2016-05-04:
I'm not sure what you mean. The extracted firmware is copied into a raw disk image, which is then used by QEMU.

0BuRner, 2016-11-22:
Same problem for me. Except it didn't work event after restarting at the makeImage.sh step.
Here is the log I log when running that script :

> [...]
> ----Creating Filesystem----
> mke2fs 1.43.1 (08-Jun-2016)
> The file /dev/mapper/loop0p1 does not exist and no size was specified

Is it normal behaviour ? What should I do ?

ddcc, 2016-11-23:
Other people have reported problems with Kali. I believe that if you're running off of a live CD, loop0 might be used for the live CD boot. Is there both loop0* and loop1* under /dev/mapper? If so, you should change  `get_device` in `firmadyne.config` to return `/dev/mapper/loop1p1`.

wangray, 2017-12-16:
Hi,

I'm having issues along these lines as well. I am using an ESX vm. I have tried changing `get_device` to `/dev/mapper/loop1p1`, `/dev/mapper/loop2p1`, but in all cases I get `Unable to mount root fs on unknown-block(8,1)`. 

Here's the contents of my `/dev/mapper` directory: 
```
control  laredo--66--vg-root  laredo--66--vg-swap_1  loop10p1  loop11p1  loop2p1  loop8p1  loop9p1
```
and the output of `makeImage.sh`: 
```
Querying database for architecture... Password for user firmadyne:
mipseb
----Running----
----Copying Filesystem Tarball----
----Creating QEMU Image----
Formatting '/home/raywang/firmadyne/scratch//1//image.raw', fmt=raw size=1073741824
----Creating Partition Table----

Welcome to fdisk (util-linux 2.27.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xbfe000f6.

Command (m for help): Created a new DOS disklabel with disk identifier 0x8badcb8f.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-2097151, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-2097151, default 2097151):
Created a new partition 1 of type 'Linux' and of size 1023 MiB.

Command (m for help): The partition table has been altered.
Syncing disks.

----Mounting QEMU Image----
add map loop13p1 (252:8): 0 2095104 linear 7:13 2048
----Creating Filesystem----
mke2fs 1.42.13 (17-May-2015)
/dev/mapper/loop10p1 contains a ext2 file system
	last mounted on /home/raywang/firmadyne/scratch/1/image on Sat Dec 16 11:25:44 2017
Proceed anyway? (y,n) y
Discarding device blocks: done
Creating filesystem with 261888 4k blocks and 65536 inodes
Filesystem UUID: 64d557f6-00f6-486c-b05e-4c02f9254cb0
Superblock backups stored on blocks:
	32768, 98304, 163840, 229376

Allocating group tables: done
Writing inode tables: done
Writing superblocks and filesystem accounting information: done

----Making QEMU Image Mountpoint----
----Mounting QEMU Image Partition 1----
----Extracting Filesystem Tarball----
----Creating FIRMADYNE Directories----
----Patching Filesystem (chroot)----
Warning: Recreating device nodes!
Removing /etc/scripts/sys_resetbutton!
----Setting up FIRMADYNE----
----Unmounting QEMU Image----
loop deleted : /dev/loop13
```