firmadyne/firmadyne #7: inferNetwork.sh script failed on multiple firmwares

ivytin, 2016-04-28:
I run `inferNetwork.sh` with multiple firmwares but all failed to generate `run.sh`.
- - -
1. [D-Link DIR-505L FIRMWARE_1.05](http://support.dlink.com/ProductInfo.aspx?m=DIR-505L)
The script terminated with following error log
```bash
...
Inferring network...
Interfaces: [('br0', '192.168.0.1')]
Traceback (most recent call last):
  File "/home/arvin/firmadyne//scripts//makeNetwork.py", line 367, in <module>
    main()
  File "/home/arvin/firmadyne//scripts//makeNetwork.py", line 364, in main
    process(infile, iid, arch, endianness, makeQemuCmd, outfile)
  File "/home/arvin/firmadyne//scripts//makeNetwork.py", line 282, in process
    for dev in brifs:
TypeError: 'NoneType' object is not iterable
```
- - -
2. [D-Link DIR-815 FIRMWARE_1.02](http://support.dlink.com/ProductInfo.aspx?m=DIR-815)
This time the script run a success with:
```bash
qemu: terminating on signal 2 from pid 2508
Inferring network...
Interfaces: []
Done!
```
But there was still no `run.sh` either, and the `qemu.initial.serial.log` said:
```bash
[    1.536000] 1f00            1024 mtdblock0 (driver?)
[    1.536000] 1f01            1024 mtdblock1 (driver?)
[    1.536000] 1f02            1024 mtdblock2 (driver?)
[    1.536000] 1f03            1024 mtdblock3 (driver?)
[    1.536000] 1f04            1024 mtdblock4 (driver?)
[    1.536000] 1f05            1024 mtdblock5 (driver?)
[    1.536000] 1f06            1024 mtdblock6 (driver?)
[    1.536000] 1f07            1024 mtdblock7 (driver?)
[    1.536000] 1f08            1024 mtdblock8 (driver?)
[    1.536000] 1f09            1024 mtdblock9 (driver?)
[    1.536000] 1f0a          120832 mtdblock10 (driver?)
[    1.536000] 0800         1048576 sda driver: sd
[    1.540000]   0801         1047552 sda1
[    1.540000] No filesystem could mount root, tried:  ext3 ext2 ext4 cramfs squashfs vfat iso9660 romfs udf
[    1.540000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(8,1)
```

ddcc, 2016-04-28:
The first was a bug in the script; that should be fixed now.
The second has a bad filesystem image. Try deleting the corresponding directory for that firmware from scratch, and restarting at the makeImage.sh step. Does this produce any warnings?

ivytin, 2016-04-29:
I run `makeImage.sh` again and no error this time.
Thank you and really nice work!

closed 2016-04-29 by ivytin