firmadyne/firmadyne #8: Kernel panic - not syncing: No init found

mrmacete, 2016-05-08:
This happens running a firmware which may be incomplete (i don't trust the source of the image i have), but it's curious enough:

```
[    0.688000] sd 0:0:0:0: [sda] Write Protect is off
[    0.688000] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    0.688000] scsi 1:0:0:0: CD-ROM            QEMU     QEMU DVD-ROM     2.0. PQ: 0 ANSI: 5
[    0.692000] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    0.692000]  sda: sda1
[    0.700000] sd 0:0:0:0: [sda] Attached SCSI disk
[    0.720000] input: AT Raw Set 2 keyboard as /devices/platform/i8042/serio0/input/input0
[    0.924000] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input1
[    0.940000] VFS: Mounted root (ext2 filesystem) on device 8:1.
[    0.940000] Freeing prom memory: 956k freed
[    0.948000] Freeing unused kernel memory: 220k freed
[    0.960000] firmadyne: do_execve: /firmadyne/console
[    0.960000] OFFSETS: offset of pid: 0x100 offset of comm: 0x1f0
[    0.960000] Kernel panic - not syncing: No init found.  Try passing init= option to kernel.
```

it complains about missing init, also enforcing init=/bin/init in the kernel command line (which i verified it exists and is executable), simply says "cannot execute /bin/init" and goes with the same error.

I believe it's a misconfiguration of the image i'm using as input of all the process, because it has been quite a pain to get it extracted and something may be missing... have you ever seen something similar, something that could point out the right direction to search?

thanks!

ddcc, 2016-05-08:
Does the kernel architecture and endianness match that of the init binary?

tUn4, 2018-06-20:
hi there, i just got the error like this.
"Kernel panic - not syncing: No init found.  Try passing init= option to kernel."

> "Does the kernel architecture and endianness match that of the init binary?"

i do not know what you mean? i've used mipsel, and where can i find init binary?
i just see "rdinit=/firmadyne/preInit.sh" in run.sh file. Did the file what you mean?

ddcc, 2018-06-20:
The init binary is part of the firmware that you are emulating, and it is responsible for initializing the system. That error means that the kernel can't find the init binary on the firmware, so either the filesystem is corrupted, the init binary has a nonstandard name, the architectures don't match, etc.

tUn4, 2018-06-21:
Yes, but i've thought that the firmadyne generate firmware's image and init binary so do i have to download init binary? I just follow by the instruction of firmadyne guide and it raised the above error.
How can i fix this issue?

ddcc, 2018-06-21:
Try deleting the scratch directory corresponding to this firmware image, and restart at makeImage.sh.

viviWangsaccounmt, 2019-08-13:
I meet the same problem . I try to delete the old image file and start again.Still not work.