firmadyne/firmadyne #14: Can't ping host from within emulated system

ackerleytng, 2016-08-18:
I emulated a router, and from that router's management page I see

```
 Pinging 192.168.0.2 with 64 bytes of data:

 Request timed out.
 Request timed out.
 Request timed out.
 Request timed out.

 Ping statistics for 192.168.0.2
  Packets: Sent = 4, Received = 0, Lost = 4 (100% loss)
```

Here's the tcpdump from the host.

```
$ sudo tcpdump -U -X -i tap1 icmp
[sudo] password for user: 
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tap1, link-type EN10MB (Ethernet), capture size 262144 bytes
04:56:56.092439 IP 0.0.0.0 > user-HP-Compaq-Elite-8300-CMT: ICMP echo request, id 411, seq 0, length 72
	0x0000:  4500 005c 0000 4000 4001 79f7 0000 0000  E..\..@.@.y.....
	0x0010:  c0a8 0002 0800 2602 019b 0000 1994 4d65  ......&.......Me
	0x0020:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0030:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0040:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0050:  5050 5050 5050 5050 5050 5050            PPPPPPPPPPPP
04:56:57.700227 IP 0.0.0.0 > user-HP-Compaq-Elite-8300-CMT: ICMP echo request, id 411, seq 1, length 72
	0x0000:  4500 005c 0000 4000 4001 79f7 0000 0000  E..\..@.@.y.....
	0x0010:  c0a8 0002 0800 9d8f 019b 0001 19ac d5be  ................
	0x0020:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0030:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0040:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0050:  5050 5050 5050 5050 5050 5050            PPPPPPPPPPPP
04:56:59.308334 IP 0.0.0.0 > user-HP-Compaq-Elite-8300-CMT: ICMP echo request, id 411, seq 2, length 72
	0x0000:  4500 005c 0000 4000 4001 79f7 0000 0000  E..\..@.@.y.....
	0x0010:  c0a8 0002 0800 13cf 019b 0002 19c5 5f65  .............._e
	0x0020:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0030:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0040:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0050:  5050 5050 5050 5050 5050 5050            PPPPPPPPPPPP
04:57:00.916516 IP 0.0.0.0 > user-HP-Compaq-Elite-8300-CMT: ICMP echo request, id 411, seq 3, length 72
	0x0000:  4500 005c 0000 4000 4001 79f7 0000 0000  E..\..@.@.y.....
	0x0010:  c0a8 0002 0800 89e7 019b 0003 19dd e933  ...............3
	0x0020:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0030:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0040:  5050 5050 5050 5050 5050 5050 5050 5050  PPPPPPPPPPPPPPPP
	0x0050:  5050 5050 5050 5050 5050 5050            PPPPPPPPPPPP
```

The source ip in the tcpdump is `0.0.0.0`, and that probably explains why the host didn't reply. Any ideas on why this is the case?

ddcc, 2016-08-20:
Sorry, the networking can be finicky. I don't have the system set up right
now, you may need to change the routing table on the host or guest.

ackerleytng, 2016-08-26:
Thanks for your quick reply!
I've checked, I think it looks right.

On the host:
```
$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    100    0        0 eno1
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 eno1
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 tap1
192.168.0.1     192.168.0.1     255.255.255.255 UGH   0      0        0 tap1
192.168.1.0     0.0.0.0         255.255.255.0   U     100    0        0 eno1
```

On the guest:
```
# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 br0
```

ackerleytng, 2016-08-26:
Figured it out!

There is an iptables rule that tries to nat packets to 0.0.0.0. 

```
# iptables -t nat -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     udp  --  anywhere             255.255.255.255     udp spt:bootps dpt:bootpc 
ACCEPT     all  --  anywhere             224.0.0.0/4         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  anywhere             anywhere            
SNAT       all  --  192.168.0.0/24       192.168.0.0/24      to:0.0.0.0          <<============== This line
MASQUERADE  all  --  anywhere             anywhere            

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain PREROUTING_DMZ (0 references)
target     prot opt source               destination         

Chain PREROUTING_PT (0 references)
target     prot opt source               destination         

Chain PREROUTING_UPNP (0 references)
target     prot opt source               destination         

Chain PREROUTING_VS (0 references)
target     prot opt source               destination      
```

It's part of the router startup script - the router wants to nat LAN packets to make it seem like it's coming from the WAN ip.

```
iptables -t nat -R POSTROUTING 2  -s $lan_ip/$lan_mask -d $lan_ip/$lan_mask -j SNAT --to-source $wan_ip
```

Since the `$wan_ip` is not set in the emulated router, the command above causes the source IP addresses to appear to come from 0.0.0.0.

closed 2016-08-26 by ackerleytng

ddcc, 2016-08-26:
Glad you figured it out!