firmadyne/firmadyne #15: Stuck in inferNetwork.sh

Puzzor, 2016-08-30:
Hi,

I think the inferNetwork.sh got stuck and the "qemu.initial.serial.log" looks like this(I deleted some usefulness info):

[    1.248000] Freeing prom memory: 956k freed
[    1.264000] Freeing unused kernel memory: 220k freed
nvram_get_buf: console_loglevel
sem_lock: Triggering NVRAM initialization!
nvram_init: Initializing NVRAM...
sem_get: Key: 4101c036
sem_get: Key: 410d002b
nvram_set_default_builtin: Setting built-in default values!
nvram_set: console_loglevel = "7"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: restore_defaults = "1"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: sku_name = ""
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: wla_wlanstate = ""
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: lan_if = "br0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: lan_ipaddr = "192.168.0.50"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: lan_bipaddr = "192.168.0.255"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: lan_netmask = "255.255.255.0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: time_zone = "EST5EDT"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: wan_hwaddr_def = "01:23:45:67:89:ab"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: wan_ifname = "eth0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: lan_ifnames = "eth1 eth2 eth3 eth4"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: ethConver = "1"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: lan_proto = "dhcp"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: wan_ipaddr = "0.0.0.0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: wan_netmask = "255.255.255.0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: wanif = "eth0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: time_zone_x = "0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: rip_multicast = "0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: bs_trustedip_enable = "0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set_default: Loading built-in default values = 1!
nvram_set_default: Checking for symbol "router_defaults"...
nvram_set_default: Checking for symbol "Nvrams"...
nvram_set_default_image: Copying overrides from defaults folder!
sem_get: Key: 410d002b
cp: cannot stat '/firmadyne/libnvram.override/*': No such file or directory
sem_get: Key: 410d002b
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_get_buf: = "7"
nvram_get_buf: boot_board_hw
sem_get: Key: 410d002b
sem_get: Key: 410d002b
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...

It seems it has been always waiting for semaphore initialization. Finally, there is no "run.sh" in scratch, could you please show me some suggestion?

ddcc, 2016-08-30:
Unfortunately, the libnvram library is known to be somewhat flaky and not particularly well-written. I've added a timeout on the semaphore initialization, you could try using this version: https://github.com/firmadyne/libnvram/releases/tag/1.0a .

Puzzor, 2016-08-31:
After using your modified version, it went a litter further, but still fails. The initial log is as follows(I added an ENTRY for boot_board_hw ):

em_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: time_zone_x = "0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: rip_multicast = "0"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set: boot_board_hw = "rt2880"
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_set_default: Loading built-in default values = 1!
nvram_set_default: Checking for symbol "router_defaults"...
nvram_set_default: Checking for symbol "Nvrams"...
nvram_set_default_image: Copying overrides from defaults folder!
sem_get: Key: 410d002b
cp: cannot stat '/firmadyne/libnvram.override/*': No such file or directory
sem_get: Key: 410d002b
sem_get: Key: 410d002b
sem_get: Key: 410d002b
nvram_get_buf: = "7"
nvram_get_buf: boot_board_hw
sem_get: Key: 410d002b
sem_get: Key: 410d002b
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_unlock: Unable to get semaphore!
nvram_get_buf: = "rt2880"
nvram_match: boot_board_hw (rt2880) ?= "rt2880"
nvram_match: true
/dev/gpio: No such file or directory
nvram_get_buf: boot_board_hw
sem_get: Key: 410d002b
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_lock: Unable to get semaphore!
sem_get: Key: 410d002b
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_unlock: Unable to get semaphore!
nvram_get_buf: = "rt2880"
nvram_match: boot_board_hw (rt2880) ?= "rt2880"
nvram_match: true
/dev/gpio: No such file or directory
[... 77 lines cut ...]
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_lock: Unable to get semaphore!
sem_get: Key: 410d002b
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410d002b, Semaphore: 8001)...
sem_unlock: Unable to get semaphore!
/sbin/init: can't resolve symbol 'bootnv_get'
[    2.068000] Kernel panic - not syncing: Attempted to kill init!

ddcc, 2016-08-31:
It looks like you'll also need to figure out the semantics of `bootnv_get` and implement it or add an alias.

Silipwn, 2019-01-28:
Stuck at similar issue, but there's no kernel panic. How do I go about changing the default values in NVRAM?
[Qemu Log](https://ptpb.pw/Tb7j)

ddcc, 2019-01-28:
See https://github.com/firmadyne/firmadyne/issues/57#issuecomment-454669091 and https://github.com/firmadyne/firmadyne/issues/36#issuecomment-271098540