firmadyne/firmadyne #17: What's  the username&password for the demo firmware

fashioncj, 2016-10-10:
When I run the command of `./scripts/run-debug.sh 1` , I meet the screen with `Welcome to SDK ...NETGEAR123456 login`, I try input 'admin' but it fails. What's the right username and password?
----
I also try to use `./scratch/1/run.sh` and run command `nc -U /tmp/qemu.1.S1` in the same time. But I can't see anything output and input seems no effect.

ddcc, 2016-10-10:
The username and password is going to be specific to whichever firmware image you're trying to emulate. You're going to have to look online to see if you can find it, or extract the passwd/shadow file from the image and try to bruteforce the hash.

closed 2016-10-10 by ddcc

fashioncj, 2016-10-11:
The passwd/shadow file shows that the user of root/admin without password. But when I used `run-debug.sh` to run the firmware your provide, I finally see the text of "Welcome to SDK, Have a lot of fun.. netgare123456 login:", but I enter root/admin try to login both of them show login incorrect.

ddcc, 2016-10-11:
I updated the README file. For that firmware, you will need to mount the filesystem, delete the `/etc/securetty` file, and unmount the filesystem. Then, the script should work.