firmadyne/firmadyne #21: FIRMWARE_DIR: unbound variable

tikf124, 2016-11-12:
Hi.

When I run the getArch.sh script, I get the following error:

hacker@hacker-VirtualBox:~/firmadyne$ ./scripts/getArch.sh ./images/1.tar.gz
./firmadyne.config: line 7: FIRMWARE_DIR: unbound variable

--
hacker@hacker-VirtualBox:~/firmadyne$ pwd
/home/hacker/firmadyne
hacker@hacker-VirtualBox:~/firmadyne$ cat firmadyne.config
#!/bin/sh

# uncomment and specify full path to FIRMADYNE repository
#FIRMWARE_DIR=/home/hacker/firmadyne/

# specify full paths to other directories
BINARY_DIR=${FIRMWARE_DIR}/binaries/
TARBALL_DIR=${FIRMWARE_DIR}/images/
SCRATCH_DIR=${FIRMWARE_DIR}/scratch/
SCRIPT_DIR=${FIRMWARE_DIR}/scripts/

# functions to safely compute other paths

check_arch () {
    ARCHS=("armel" "mipseb" "mipsel")

    if [ -z "${1}" ]; then
        return 0
    fi

    match=0
    for i in "${ARCHS[@]}"; do
        if [ "${1}" == "$i" ]; then
            match=1
        fi
    done

    if [ "${match}" -eq 0 ]; then
        return 0
    fi

    return 1
}

check_number () {
    if [ "${1}" -ge 0 ]; then
        return 1
    fi

    return 0
}

check_root () {
    if [ "${EUID}" -eq 0 ]; then
        return 1
    fi

    return 0
}

get_console () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    echo "${BINARY_DIR}/console.${1}"
}

get_fs () {
    if check_number "${1}"; then
        echo "Error: Invalid image number!"
        exit 1
    fi

    echo "`get_scratch "${1}"`/image.raw"
}

get_fs_mount () {
    if check_number "${1}"; then
        echo "Error: Invalid image number!"
        exit 1
    fi

    echo "`get_scratch "${1}"`/image/"
}

get_kernel () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "${BINARY_DIR}/zImage.${1}"
            ;;
        mipseb)
            echo "${BINARY_DIR}/vmlinux.${1}"
            ;;
        mipsel)
            echo "${BINARY_DIR}/vmlinux.${1}"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_nvram () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    echo "${BINARY_DIR}/libnvram.so.${1}"
}

get_qemu () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "qemu-system-arm"
            ;;
        mipseb)
            echo "qemu-system-mips"
            ;;
        mipsel)
            echo "qemu-system-mipsel"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_qemu_disk () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "/dev/vda1"
            ;;
        mipseb)
            echo "/dev/sda1"
            ;;
        mipsel)
            echo "/dev/sda1"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_qemu_machine () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "virt"
            ;;
        mipseb)
            echo "malta"
            ;;
        mipsel)
            echo "malta"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_scratch () {
    if check_number "${1}"; then
        echo "Error: Invalid image number!"
        exit 1
    fi

    echo "${SCRATCH_DIR}/${IID}/"
}

get_device () {
    echo "/dev/mapper/loop0p1"
}
--

What is the problem?

ddcc, 2016-11-12:
You need to uncomment FIRMWARE_DIR by removing the `#`.

tikf124, 2016-11-13:
Hi.

I solved the problem of running "./scripts/getArch.sh" with your help

However, another problem arises.

An error occurs in execution of "./scripts/tar2db.py".

hacker@hacker-VirtualBox:~/firmadyne$ ./scripts/tar2db.py -i 1 -f ./images/1.tar.gz
---
Traceback (most recent call last):
  File "./scripts/tar2db.py", line 100, in <module>
    main()
  File "./scripts/tar2db.py", line 97, in main
    process(iid, infile)
  File "./scripts/tar2db.py", line 77, in process
    insertObjectToImage(iid, file2oid, links, cur)
  File "./scripts/tar2db.py", line 57, in insertObjectToImage
    for x in files2oids])
psycopg2.IntegrityError: duplicate key value violates unique constraint "object_to_image_oid_iid_filename_key"
DETAIL:  Key (oid, iid, filename)=(2, 1, /home/www/clearLog.php) already exists.
----

What is the problem?

ddcc, 2016-11-13:
You should try clearing the database again. You can't run the same script more than once for the same image image ID, because the database doesn't allow for duplicate entries.

tikf124, 2016-11-14:
Hi

I solved the problem of running "./scripts/tar2db.py " with your help

However, another problem arises.

An error occurs in execution of "./scripts/makeImage.sh".

---
root@Karas:~/firmadyne# ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipseb
Running firmware 1: terminating after 60 secs...
qemu: could not load kernel '/root/firmadyne/binaries//vmlinux.mipseb'
---
root@Karas:~/firmadyne# ls -la /root/firmadyne/binaries//vmlinux.mipseb
-rw-r--r-- 1 root root 5117519 Nov 13 18:59 /root/firmadyne/binaries//vmlinux.mipseb

What is the problem?

ddcc, 2016-11-14:
Did you download the binaries? See the installation steps.

tikf124, 2016-11-16:
Hi

I solved the problem of running "./scripts/makeImage.sh" with your help

However, another problem arises.

An error occurs in execution of "./scratch/1/run.sh".

The firmware I want to analyze does not have run.sh.

I will attach the qemu.initial.serial.log log

--
[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:44:57 UTC 2016

[    0.000000] 

[    0.000000] LINUX started...

[    0.000000] bootconsole [early0] enabled

[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)

[    0.000000] FPU revision is: 00739300

[    0.000000] Determined physical RAM map:

[    0.000000]  memory: 00001000 @ 00000000 (reserved)

[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)

[    0.000000]  memory: 00606000 @ 000f0000 (reserved)

[    0.000000]  memory: 0f90a000 @ 006f6000 (usable)

[    0.000000] debug: ignoring loglevel setting.

[    0.000000] Wasting 57024 bytes for tracking 1782 unused pages

[    0.000000] Initrd not found or empty - disabling initrd

[    0.000000] Zone PFN ranges:

[    0.000000]   DMA      0x00000000 -> 0x00001000

[    0.000000]   Normal   0x00001000 -> 0x00010000

[    0.000000] Movable zone start PFN for each node

[    0.000000] early_node_map[1] active PFN ranges

[    0.000000]     0: 0x00000000 -> 0x00010000

[    0.000000] On node 0 totalpages: 65536

[    0.000000] free_area_init_node: node 0, pgdat 806923c0, node_mem_map 81000000

[    0.000000]   DMA zone: 32 pages used for memmap

[    0.000000]   DMA zone: 0 pages reserved

[    0.000000]   DMA zone: 4064 pages, LIFO batch:0

[    0.000000]   Normal zone: 480 pages used for memmap

[    0.000000]   Normal zone: 60960 pages, LIFO batch:15

[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024

[    0.000000] Kernel command line: firmadyne.syscall=1 root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1
[... 5655 lines cut ...]
/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory

/dev/nvram: No such file or directory
/dev/nvram: No such file or directory
/dev/nvram: No such file or directory

--

What is the problem?

ddcc, 2016-11-16:
We don't support emulation of all firmware, because they each behave differently. It looks like there's a problem with the NVRAM emulation, so you'll need to disassemble the init binary to determine the settings that the firmware is looking for, and modify the libnvram library to load the correct default settings.