firmadyne/firmadyne #22: Few firmware doesn't work on Kali linux

arunmagesh, 2016-11-16:
Hello,
I tried to emulate a Dlink firmware in Ubuntu and it worked perfectly, but in kali the Network is up and it is not opening any login screen. 
Nmap shows all 1000 ports are filtered.

ddcc, 2016-11-16:
Unfortunately, we've only tested with Ubuntu 14.04, so I'm not sure about compatibility with Kali. You might want to check if the network tap device is being created successfully, the routing table is updated, and whether there are any restrictive default iptables rules.

arunmagesh, 2016-11-16:
Just checked. 
1. TAP device created 
2. Routing table generated
3.No restrictive iptables.

on Kali Linux 2016.1 Release

ddcc, 2016-11-16:
You could also compare the network traffic on the TAP device with tcpdump/Wireshark, and see if it reaching the emulated device. If you're getting back but timeouts, you'd probably want to compare the QEMU output and kernel log to figure out why the emulated device isn't listening where you expect it. Otherwise, if you get TCP resets, you'd probably want to directly check the kernel log.