firmadyne/firmadyne #23: Firmware login - inferNetwork

0BuRner, 2016-11-24:
Hi,

I successfully ran the firmware example following the readme.
Then I tried to do the same with another firmware.
I got stuck at the inferNetwork.sh step. It wasn't able to find an interface, so no run.sh file created. Then I looked up at the qemu initial logs and it seems that the firmware require a login and probably a password to go further. If I'm right, how can I give one via firmadyne ?

Here are the logs of inferNetwork :
> root@kali:~/firmadyne# ./scripts/inferNetwork.sh 2
> Querying database for architecture... Password for user firmadyne: 
> mipseb
> Running firmware 2: terminating after 60 secs...
> qemu-system-mips: terminating on signal 2 from pid 12420
> Inferring network...
> Interfaces: []
> Done!

Here are my logs from qemu : 
>[    1.916000] Status: 0000a413    USER EXL IE 
>[    1.920000] Cause : 10800008
>[    1.920000] BadVA : 00000038
>[    1.920000] PrId  : 00019300 (MIPS 24Kc)
>
> (none) mips #1 Thu Feb 18 01:39:21 UTC 2016 (none)
>(none) login: 

Here is the link of the firmware :
http://static.tp-link.com/resources/software/TL-WR740N_V4_140520.zip

ddcc, 2016-11-25:
It shouldn't matter whether the firmware requires a password to login or not; pretty much all will have a password prompt on the root shell. The system looks for system calls to certain kernel networking functions that bind IP addresses, in order to infer the correct address. If this isn't happening, typically there is some NVRAM-related problem earlier in the boot process that is causing a crash or abort before the networking comes up. You'll want to see what are the last few NVRAM keys requested before the crash/abort, and whether a default value needs to be set for that key to continue the boot process.

0BuRner, 2016-11-26:
[...]
[    1.256000] VFS: Mounted root (ext2 filesystem) on device 8:1.
[    1.260000] Freeing prom memory: 956k freed
[    1.272000] Freeing unused kernel memory: 220k freed
[    1.332000] firmadyne: sys_reboot[PID: 45 (init)]: magic1:fee1dead, magic2:28121969, cmd:0
[    1.348000] firmadyne: sys_socket[PID: 1 (init)]: family:1, type:1, protocol:0
[    1.352000] firmadyne: sys_socket[PID: 1 (init)]: family:1, type:2, protocol:0
init started:  BusyBox v1.01 (2014.05.20-12:50+0000) multi-call binary
[    1.364000] firmadyne: sys_socket[PID: 47 (init)]: family:1, type:1, protocol:0
[    1.364000] firmadyne: sys_socket[PID: 47 (init)]: family:1, type:2, protocol:0
[    1.380000] firmadyne: do_execve: /firmadyne/console
[    1.380000] OFFSETS: offset of pid: 0x100 offset of comm: 0x1f0
[    1.460000] firmadyne: sys_socket[PID: 53 (ifconfig)]: family:2, type:1, protocol:0
[    1.460000] firmadyne: __inet_insert_ifa[PID: 53 (ifconfig)]: device:lo ifa:0x7f000001
[    1.464000] firmadyne: __inet_insert_ifa[PID: 53 (ifconfig)]: device:lo ifa:0x7f000001
This Board use 2.6.31
insmod: cannot open module `/lib/modules/2.6.31/kernel/iptable_raw.ko': No such file or directory
insmod: cannot open module `/lib/modules/2.6.31/kernel/flashid.ko': No such file or directory
insmod: cannot open module `/lib/modules/2.6.31/kernel/harmony.ko': No such file or directory
[    1.820000] firmadyne: sys_socket[PID: 102 (init)]: family:1, type:1, protocol:0
[    1.820000] firmadyne: sys_socket[PID: 102 (init)]: family:1, type:2, protocol:0
[    1.900000] do_page_fault() #2: sending SIGSEGV to httpd for invalid read access from
[    1.900000] 00000038 (epc == 2ab1d488, ra == 0048e420)
[    1.900000] Cpu 0
[    1.900000] $ 0   : 00000000 1000a400 00000057 00000000
[    1.900000] $ 4   : 00000000 00000001 00000001 00518dce
[    1.900000] $ 8   : 2ab5f1f8 00000000 00000001 fffffff8
[    1.900000] $12   : fffffffe 00000001 00000000 8fb32038
[    1.900000] $16   : 00510000 00550000 00576624 00000038
[    1.900000] $20   : 00550000 00000000 00000000 00000068
[    1.900000] $24   : 00000018 2ab1d460                  
[    1.900000] $28   : 2ab67510 7fefe268 00000066 0048e420
[    1.900000] Hi    : 00000000
[    1.904000] Lo    : 00000056
[    1.904000] epc   : 2ab1d488 0x2ab1d488
[    1.904000]     Not tainted
[    1.904000] ra    : 0048e420 0x48e420
[    1.904000] Status: 0000a413    USER EXL IE 
[    1.904000] Cause : 10800008
[    1.904000] BadVA : 00000038
[    1.904000] PrId  : 00019300 (MIPS 24Kc)
[    1.904000] Modules linked in:
[    1.904000] Process httpd (pid: 99, threadinfo=8fbda000, task=8fb32038, tls=00000000)
[    1.904000] Stack : 00000057 2ab5f1ac ffffffff 2ab1dfc4 2ab67510 004ee898 2aafd4d0 00000000
[    1.904000]         2ab67510 2aadb180 2ab472d4 2ab5f1e8 00000000 00000000 2aafd4d0 00510000
[    1.904000]         00550000 00576624 00000038 0048e420 00588bd8 00576624 00000038 004adcfc
[    1.904000]         00555070 00000038 00550000 00000000 00555070 00576dcc 00588bd8 0047c888
[    1.904000]         00000000 00000001 00000066 2aaac858 00555070 0048fc64 00000000 00000000
[    1.908000]         ...
[    1.908000] Call Trace:
[    1.908000]  (Bad stack address)
[    1.908000] 
[    1.908000] Code: afb10040  afb0003c  afbc0010 <8c930038> 1660000c  00808821  8f9988c0  8f858910  2490003c 
[    1.916000] httpd/99: potentially unexpected fatal signal 11.
[    1.916000] 
[    1.916000] Cpu 0
[    1.916000] $ 0   : 00000000 1000a400 00000057 00000000
[    1.916000] $ 4   : 00000000 00000001 00000001 00518dce
[    1.916000] $ 8   : 2ab5f1f8 00000000 00000001 fffffff8
[    1.916000] $12   : fffffffe 00000001 00000000 8fb32038
[    1.916000] $16   : 00510000 00550000 00576624 00000038
[    1.916000] $20   : 00550000 00000000 00000000 00000068
[    1.916000] $24   : 00000018 2ab1d460                  
[    1.916000] $28   : 2ab67510 7fefe268 00000066 0048e420
[    1.916000] Hi    : 00000000
[    1.916000] Lo    : 00000056
[    1.916000] epc   : 2ab1d488 0x2ab1d488
[    1.916000]     Not tainted
[    1.916000] ra    : 0048e420 0x48e420
[    1.916000] Status: 0000a413    USER EXL IE 
[    1.920000] Cause : 10800008
[    1.920000] BadVA : 00000038
[    1.920000] PrId  : 00019300 (MIPS 24Kc)

 (none) mips #1 Thu Feb 18 01:39:21 UTC 2016 (none)
(none) login:

ddcc, 2016-11-27:
Looking at that log, I don't see the firmware try to assign IP addresses to any network adapters. I suspect that it's aborting parts of the boot process as soon as it encounters a failure, specifically inserting the kernel module. See if you can find a configuration file or init script on the filesystem, and modify it to avoid inserting any kernel modules. You may find the mount.sh and umount.sh scripts useful for doing this.

0BuRner, 2016-11-27:
Thx again for your help but I'm still stuck :(

I managed to modify configuration to avoid loading kernel modules and it worked (I didn't see insmod error anymore). But still the same result. So I also disabled the start of "httpd" because it seemed to throw an error too. But doesn't matter, the boot process still doesn't go further...

I found a (non-emulated I guess) bootlog of other TP-Link router, dunno if it can help : https://wiki.openwrt.org/toh/tp-link/tl-mr11u/bootlog

ddcc, 2016-11-28:
Hmm, looking at that bootlog, it seems that the kernel module failures are normal. At this point, I'd guess that the firmware is trying to access NVRAM or flash via a mechanism that we don't emulate in libnvram. If you have access to IDA Pro or another disassembler, I'd suggest looking to see where the string "Now flash open!" is being printed out from to get an idea of where this is occurring. In the simplest case, you'll just need to add an alias for the library function to libnvram, but in a more complex case, if the system is accessing e.g. /dev/nvram directly, you'd need to add some IOCTL or read/write emulation to the firmadyne kernel module.

0BuRner, 2016-11-30:
Hi, I don't remember where I got that but I have the source code of the firmware for that router model.
Here is where I found the "Now flash open!" string : 

- pb92\linux\kernels\mips-linux-2.6.31\drivers\mtd\devices\ar7240_flash_ioctl.c
- ap143-2.0\linux\kernels\mips-linux-2.6.31\drivers\mtd\devices\ath_flash_ioctl.c

Where should I look if I didn't had the source code? Which executable should I disassemble ? I have really no idea what to do exactly now... :s Should I edit libnvram/alias.c to add something for Atheros driver?

ddcc, 2016-12-01:
Generally, you would start looking at the `init` process, which is essentially the first process called by the kernel after the system has booted. From there, you would look at the rules/configuration for whatever handles the boot process, generally SysV-style init. In this case, modifying `libnvram` probably isn't useful, because something on the system is trying to access the flash device directly via an IOCTL.

You should try to figure out what is the name of the device that is being accessed (e.g. `/dev/nvram`); this should be in the source code you have. Then, figure out what usermode process is trying to access it, and if you can get it to bypass the hardware device. Otherwise, you'll need to add a stub for the device in e.g. https://github.com/firmadyne/kernel-v4.1/blob/firmadyne-v4.1.17/drivers/firmadyne/devfs_stubs.c (depending on kernel version). Unfortunately, you'll probably also need an actual dump of the NVRAM or the actual hardware, to make the stub pretend to read/write the correct data.

HacKeD0x90, 2017-05-04:
Hello,
I'm facing the same issue, 
Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
mke2fs 1.42.12 (29-Aug-2014)
The file /dev/mapper/loop0p1 does not exist and no size was specified.
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 7336
Querying database for architecture... mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Interfaces: []
Done!

Running the firmware finally : 

can any one inform me how to trouble shoot or solve this issue?
the firmware I'm trying to emulate is Dlink_firmware.bin