firmadyne/firmadyne #24: Firmadyne - Object has no attribute databese

irini90, 2016-11-28:
I keep getting this error and I am not sure what the issue is. I read that on Kali the Dlink firmware cannot be emulated so I tried a different one, I the error is still the same. Can someone help please?
```

root@wimo:~/tools/firmadyne# ./fat.py

	Welcome to the Firmware Analysis Toolkit - v0.1
	Offensive IoT Exploitation Training  - http://offensiveiotexploitation.com
	By Attify - https://attify.com  | @attifyme
	
Enter the name or absolute path of the firmware you want to analyse : Dlink_firmware.bin
Enter the brand of the firmware : Dlink
Dlink_firmware.bin
Now going to extract the firmware. Hold on..
/root/tools/firmadyne/sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "Dlink_firmware.bin" images 
Traceback (most recent call last):
  File "/root/tools/firmadyne/sources/extractor/extractor.py", line 730, in <module>
    main()
  File "/root/tools/firmadyne/sources/extractor/extractor.py", line 727, in main
    extract.extract()
  File "/root/tools/firmadyne/sources/extractor/extractor.py", line 189, in extract
    self._extract_item(item)
  File "/root/tools/firmadyne/sources/extractor/extractor.py", line 197, in _extract_item
    ExtractionItem(self, path, 0).extract()
  File "/root/tools/firmadyne/sources/extractor/extractor.py", line 227, in __init__
    host=self.extractor.database)
  File "/usr/lib/python2.7/dist-packages/psycopg2/__init__.py", line 164, in connect
    conn = _connect(dsn, connection_factory=connection_factory, async=async)
psycopg2.OperationalError: FATAL:  password authentication failed for user "firmadyne"
FATAL:  password authentication failed for user "firmadyne"

Exception AttributeError: "'ExtractionItem' object has no attribute 'database'" in <bound method ExtractionItem.__del__ of <__main__.ExtractionItem object at 0x7f55bea2f590>> ignored
Traceback (most recent call last):
  File "./fat.py", line 113, in <module>
    main()
  File "./fat.py", line 100, in main
    extractor(firm_name,firm_brand)
  File "./fat.py", line 45, in extractor
    output = subprocess.check_output(extractor_command, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '/root/tools/firmadyne/sources/extractor/extractor.py -b Dlink -sql 127.0.0.1 -np -nk "Dlink_firmware.bin" images ' returned non-zero exit status 1
```

adi0x90, 2016-11-28:
@irini90  : Did it ask you for the password during the process or it didn't prompt for any?

irini90, 2016-11-28:
@adi0x90 No it didn't!

adi0x90, 2016-11-28:
I think maybe putting the password in the script itself could work. Let me try this and let you know. 
@ddcc : Have you ever encountered this?

ddcc, 2016-11-28:
I have no idea what this "Firmware Analysis Toolkit" is, and what changes have been made to the source code. You might want to ask them for help.

By default, many of the scripts assume the username `firmadyne` and password `firmadyne` to access the database. If the actual database username/password is different, you'll want to update the scripts.

adi0x90, 2016-11-28:
It's a script to automate the steps to be done manually in Firmadyne. 
@irini90 : Can you try these steps one after the other and see if it works :

```
./reset.sh
./sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk "WNAP320 Firmware Version 2.0.3.zip" images
./scripts/getArch.sh ./images/1.tar.gz
./scripts/tar2db.py -i 1 -f ./images/1.tar.gz
sudo ./scripts/makeImage.sh 1
./scripts/inferNetwork.sh 1
./scratch/1/run.sh
```

irini90, 2016-11-29:
@adi0x90 It worked. I now get the IP address of the interface however when I try to access it in the browser it doesn't load. I have pinged the IP address but its taking a very long time to get a response (still haven't had a response)

ddcc, 2016-11-30:
What is the IP address of the interface? Can you post your routing table (`route -nv`)?

irini90, 2016-11-30:
Here is the IP address it gives me for the interface - 192.168.0.1:

```
Creating TAP device tap1...
Set 'tap1' persistent and owned by uid 0
Bringing up TAP device...
Adding route to 192.168.0.1...
Starting emulation of firmware... Done!
The emulated firmware may not be accessible while booting.
Press any key to destroy the network and shutdown emulation.
```

And my routing table is:

```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.72.0.1       0.0.0.0         UG    100    0        0 eth0
10.72.0.0       0.0.0.0         255.255.248.0   U     100    0        0 eth0
10.239.40.2     10.72.0.1       255.255.255.255 UGH   100    0        0 eth0
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 tap1
192.168.0.1     192.168.0.1     255.255.255.255 UGH   0      0        0 tap1
```

ddcc, 2016-11-30:
That's strange, it looks like it should be reachable from the network. You could try scanning it using `nmap -P0 -O -sV` to see if there are any other listening services. Otherwise, there may be some emulation issue that is preventing daemons from loading, you'd have to check the kernel log to see.

mongoosymongoose, 2024-01-04:
> It's a script to automate the steps to be done manually in Firmadyne. @irini90 : Can you try these steps one after the other and see if it works :
>
> ```
> ./reset.sh
> ./sources/extractor/extractor.py -b [...]

I am having a similar issue but can't find `reset.sh`. Is that an older file that is no longer in the repo?