firmadyne/firmadyne #25: /libnvram.defaults not created

trentn, 2016-12-09:
I've come across a bug that seems to break `makeImage.sh`. The error occurs when makeImage.sh is executing `fixImage.sh`. The output is below. It seems like the `/libnvram.defaults` directory is never created anywhere. Where is it supposed to be created?

I did attempt to have `fixImage.sh` create the directory if it was not already, but that causes other issues later when attempting to infer the network. (`qemu.initial.serial.log` will end with an "unable to mount" error as described in the troubleshooting section for not generating `run.sh`)

```
Querying database for architecture... Password for user firmadyne: 
armel
----Running----
----Copying Filesystem Tarball----
----Creating QEMU Image----
Formatting '/root/firmadyne//scratch//5//image.raw', fmt=raw size=1073741824
----Creating Partition Table----

Welcome to fdisk (util-linux 2.28.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xdaf8d020.

Command (m for help): Created a new DOS disklabel with disk identifier 0xb4cfdf9f.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-2097151, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-2097151, default 2097151): 
Created a new partition 1 of type 'Linux' and of size 1023 MiB.

Command (m for help): The partition table has been altered.
Syncing disks.

----Mounting QEMU Image----
add map loop0p1 (252:0): 0 2095104 linear 7:0 2048
----Creating Filesystem----
mke2fs 1.42.13 (17-May-2015)
Discarding device blocks: done                            
Creating filesystem with 261888 4k blocks and 65536 inodes
Filesystem UUID: f6508bb1-f56f-4654-81fa-aa5f55c4bdd6
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376

Allocating group tables: done                            
Writing inode tables: done                            
Writing superblocks and filesystem accounting information: done

----Making QEMU Image Mountpoint----
----Mounting QEMU Image Partition 1----
----Extracting Filesystem Tarball----
----Creating FIRMADYNE Directories----
----Patching Filesystem (chroot)----
Creating /etc/TZ!
Creating /etc/passwd!
Warning: Recreating device nodes!
Removing /etc/scripts/sys_resetbutton!
Creating default rip_multicast!
/fixImage.sh: line 158: can't create /libnvram.defaults/rip_multicast: nonexistent directory
Creating default bs_trustedip_enable!
/fixImage.sh: line 163: can't create /libnvram.defaults/bs_trustedip_enable: nonexistent directory
Creating default filter_rule_tbl!
/fixImage.sh: line 168: can't create /libnvram.defaults/filter_rule_tbl: nonexistent directory
Creating default rip_enable!
/fixImage.sh: line 173: can't create /libnvram.defaults/rip_enable: nonexistent directory
```

ddcc, 2016-12-09:
Thanks for catching that, at some point I renamed the directory from `libnvram.defaults` to `libnvram.override`, but forgot to update it in that script.

closed 2016-12-09 by ddcc

wdormann, 2016-12-12:
I'm not sure that this is the fix.  With current version:

Creating default rip_multicast!
/fixImage.sh: line 158: can't create /libnvram.override/rip_multicast: nonexistent directory
Creating default bs_trustedip_enable!
/fixImage.sh: line 163: can't create /libnvram.override/bs_trustedip_enable: nonexistent directory
Creating default filter_rule_tbl!
/fixImage.sh: line 168: can't create /libnvram.override/filter_rule_tbl: nonexistent directory
Creating default rip_enable!
/fixImage.sh: line 173: can't create /libnvram.override/rip_enable: nonexistent directory