firmadyne/firmadyne #26: Run.sh executed successfully, but could not connect to the firmware.

kkkpoh, 2016-12-10:
I downloaded the Ubuntu 14.04.5 LTS 32 bit to setup firmadyne from scratch. I have everything setup as per README.md, and run.sh was generated successfully for the firmware I am testing. However, upon running the run.sh (the console shows the firmware running successfully), I could not ping the emulated on the IP indicated in the run.sh, nor I could reach the web interface of the firmware (the firmware runs fine on AttifyOS thou).

I have check that the route was added successfully (tap1 created successfully) as the IP address is the same as per run.sh. I traced through qemu.final or qemu.inital log but I could not tell what went wrong.

```
#route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.153.2   0.0.0.0         UG    0      0        0 eth0
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 tap1
192.168.0.1     192.168.0.1     255.255.255.255 UGH   0      0        0 tap1
192.168.153.0   0.0.0.0         255.255.255.0   U     1      0        0 eth0

```
Below is the stripped version of the qemu.initial.serial.log：

#grep -iv firmadyne qemu.final.serial.log
```[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:44:57 UTC 2016
[    0.000000] 
[    0.000000] LINUX started...
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 00606000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f90a000 @ 006f6000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 57024 bytes for tracking 1782 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x00010000
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x00010000
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 806923c0, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60960 pages, LIFO batch:15
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252524k/255016k available (4164k kernel code, 2252k reserved, 1550k data, 220k init, 0k highmem)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:256
[    0.000000] CPU frequency 200.00 MHz
[    0.000000] Console: colour dummy device 80x25
[    0.000000] Calibrating delay loop... 1314.81 BogoMIPS (lpj=2629632)
[    0.104000] Mount-cache hash table entries: 512
[    0.112000] NET: Registered protocol family 16
[    0.120000] bio: create slab <bio-0> at 0
[    0.124000] vgaarb: loaded
[    0.124000] SCSI subsystem initialized
[    0.124000] libata version 3.00 loaded.
[    0.124000] usbcore: registered new interface driver usbfs
[    0.124000] usbcore: registered new interface driver hub
[    0.128000] usbcore: registered new device driver usb
[    0.128000] pci 0000:00:00.0: reg 14 32bit mmio pref: [0x1000000-0x1ffffff]
[    0.128000] pci 0000:00:0a.1: reg 20 io port: [0x00-0x0f]
[    0.128000] pci 0000:00:0a.2: reg 20 io port: [0x00-0x1f]
[... 569 lines cut ...]
/dev/gpio: No such file or directory
factory reset button pressed for 11 secs
IGMP proxy is not enabled!
No such file or directory
PBC pressed for 11 secs
Change the STATUS LED...
[/etc/init.d/S10system.sh] done!
[/etc/init.d/S30final.sh]
Finally, free pagecache, dentries and inodes ...
/dev/gpio: No such file or directory
[/etc/init.d/S30final.sh] done!
rcS done!
Starting pid 954, console /dev/tts/0: '/bin/sh'

BusyBox v1.00 (2010.11.23-11:52+0000) Built-in shell (msh)
Enter 'help' for a list of built-in commands.

[   16.620000] eth2: no IPv6 routers present
[   16.748000] eth2.2: no IPv6 routers present
[   17.396000] eth2.0: no IPv6 routers present
[/var/run/upnp_alive.sh] ...

```

Appreciate if anyone could provide some pointers to this. Thank you.

ddcc, 2016-12-12:
The factory reset functionality may be doing something weird; try adding `touch /dev/gpio` to `fixImage.sh`, delete the `1` subdirectory under `scratch`, and restart from `makeImage.sh`.

Otherwise, it seems like the network is coming up, so I'm not sure what's going on. Since the firmware spawns a shell on the default console, you should be able to poke around with `iptables` and `ifconfig` to see if there's something strange there. I made a change to enable the VNC server by default, so if you update the local git repository and restart from the `inferNetwork.sh` step, you should be able to access the console using step 10 from the README.

kkkpoh, 2016-12-14:
Just a quick update, adding 'touch dev/gpio' gave the same result.

I am poking through the console and hope to find something in there. Will update if I could identify the root cause.

Thank you.

liyansong2018, 2021-02-23:
Bridge setting error. You should reset the bridge in the qemu virtual machine

For example
```shell
brctl addif br0 eth0
```