firmadyne/firmadyne #28: /libnvram.override not created - error in makeImage.sh

wdormann, 2016-12-12:
Here's a problem I ran into, while attempting to create an image with the latest git version of firmadyne:

$ sudo ./scripts/makeImage.sh 1
Querying database for architecture... Password for user firmadyne: 
armel
----Running----
----Copying Filesystem Tarball----
----Creating QEMU Image----
Formatting '/home/fuzz/in/firmadyne//scratch//1//image.raw', fmt=raw size=1073741824
----Creating Partition Table----

Welcome to fdisk (util-linux 2.27.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xfaec4e5b.

Command (m for help): Created a new DOS disklabel with disk identifier 0xbf863f73.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-2097151, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (20482097151, default 2097151): 
Created a new partition 1 of type 'Linux' and of size 1023 MiB.

Command (m for help): The partition table has been altered.
Syncing disks.

----Mounting QEMU Image----
add map loop0p1 (252:2): 0 2095104 linear 7:0 2048
----Creating Filesystem----
mke2fs 1.42.13 (17-May-2015)
Discarding device blocks: done                            
Creating filesystem with 261888 4k blocks and 65536 inodes
Filesystem UUID: 6472c364-3013-475b-b668-95a76aa17fc4
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376

Allocating group tables: done                            
Writing inode tables: done                            
Writing superblocks and filesystem accounting information: done

----Making QEMU Image Mountpoint----
----Mounting QEMU Image Partition 1----
----Extracting Filesystem Tarball----
----Creating FIRMADYNE Directories----
----Patching Filesystem (chroot)----
Creating /etc/TZ!
Creating /etc/hosts!
Creating /etc/passwd!
Warning: Recreating device nodes!
Removing /etc/scripts/sys_resetbutton!
Creating default rip_multicast!
/fixImage.sh: line 158: can't create /libnvram.override/rip_multicast: nonexistent directory
Creating default bs_trustedip_enable!
/fixImage.sh: line 163: can't create /libnvram.override/bs_trustedip_enable: nonexistent directory
Creating default filter_rule_tbl!
/fixImage.sh: line 168: can't create /libnvram.override/filter_rule_tbl: nonexistent directory
Creating default rip_enable!
/fixImage.sh: line 173: can't create /libnvram.override/rip_enable: nonexistent directory

The problem is that makeImage.sh creates "/firmware/nvram.override", while fixImage.sh looks for "/nvram.override".   (note the missing "/firmadyne/").    It seems that removing "/firmadyne" from makeImage.sh doesn't fix everything, nor does adding "/firmadyne" to fixImage.sh.   The error from the latter technique is:

nvram_set_default_image: Copying overrides from defaults folder!
sem_get: Key: 410c0019
cp: cannot remove '/libnvram/bs_trustedip_enable': No such file or
directory
cp: cannot remove '/libnvram/filter_rule_tbl': No such file or directory
cp: cannot remove '/libnvram/rip_enable': No such file or directory
cp: cannot remove '/libnvram/rip_multicast': No such file or directory
sem_get: Key: 410c0019
sem_get: Key: 410c0019
sem_get: Key: 410c0019
nvram_get_buf: = "0"
[    3.005122] Kernel panic - not syncing: Attempted to kill init!
exitcode=0x00001600
[    3.005122]
nvram_set_default_image: Copying overrides from defaults folder!
sem_get: Key: 410c0019
cp: cannot remove '/libnvram/bs_trustedip_enable': No such file or directory
cp: cannot remove '/libnvram/filter_rule_tbl': No such file or directory
cp: cannot remove '/libnvram/rip_enable': No such file or directory
cp: cannot remove '/libnvram/rip_multicast': No such file or directory
sem_get: Key: 410c0019
sem_get: Key: 410c0019
sem_get: Key: 410c0019
nvram_get_buf: = "0"
[    3.005122] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00001600
[    3.005122]
[    3.005463] CPU: 0 PID: 1 Comm: init Tainted: G        W       4.1.17+ #1
[    3.005652] Hardware name: Generic DT based system
[    3.005855] [<c001c99c>] (unwind_backtrace) from [<c0019d30>] (show_stack+0x10/0x14)
[    3.006095] [<c0019d30>] (show_stack) from [<c0410b78>] (panic+0x70/0x1c8)
[    3.006328] [<c0410b78>] (panic) from [<c0025714>] (do_exit+0x3c0/0x774)
[    3.006540] [<c0025714>] (do_exit) from [<c0026718>] (do_group_exit+0x0/0xa8)
[    3.006780] [<c0026718>] (do_group_exit) from [<c0016d00>] (ret_fast_syscall+0x0/0x3c)
[    3.007207] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00001600
[    3.007207]

ddcc, 2016-12-12:
Ok, let me take a look. Is this with the example firmware image in the README, or a different one?

wdormann, 2016-12-12:
It could be that the image is unsupported (due to missing nvram value?)

http://www.downloads.netgear.com/files/GDC/R8000/R8000-V1.0.3.4_1.1.2.zip

If it is a missing nvram issue, it might be of value to document how one could determine what value is missing.

wdormann, 2016-12-12:
I think there may be two things going on here:

1) The git version of firmadyne still fails with images that use "libnvram.override" due to the discrepancies with how the directory are referenced.  (See the above link to the R8000 image for an example).

2) If I patch the scripts to use the proper directories for libnvram, init dies, resulting in a kernel panic.   This is perhaps due to the R8000 (and similar) images looking for a non-existant nvram value?   Grepping through the qemeu log file, it does look like the "libnvram.override" values are being set in my hacky-patched version.  But it still crashes on boot.

ddcc, 2016-12-12:
There was a bug with the libnvram path, which should be fixed now. You'll need to rerun the download.sh script because the binaries have been updated.

However, that bug doesn't affect the kernel panic, which is an issue with the NVRAM emulation. This is mentioned briefly in the [FAQ](https://github.com/firmadyne/firmadyne#faq), but I agree the documentation could be better. Generally, the last requested value tends to be the problematic one, but it isn't always the case, and is difficult to verify without disassembling the `init` binary. In this case, the only requested key is `time_zone`, because it triggers the libnvram initialization, then returns the value `0`, followed by the crash. I'm not sure if it's the problem, though, because even with an override to  `EST5EDT`, the crash still occurs.

You'll probably need to attach a debugger during execution, but this can be tricky to set up: 
1. With full-system emulation, attach to QEMU's built-in gdb stub. However, the debugger won't automatically know where kernel and user space is in memory, so you may need to manually do `add-symbol-file` with gdb.
2. With user-mode emulation, set up a Linux container or VM, chroot into the firmware image, and run the `init` binary with the firmadyne libnvram in the LD_LIBRARY_PATH (or just replace the preexisting binary entirely). This is easier to debug, because you can attach directly to the `init` process, but the system state won't be exactly the same, since the kernel is different.

wdormann, 2016-12-12:
Any tips on how to get gdb working with qemu as firmadyne uses it?  Best guess I had was to edit run-debug.sh and add the "-s" commandline argument with qemu (to start the built-in gdb server on 1234/tcp).  But when I run run-debug.sh it just starts up the target machine without waiting.

ddcc, 2016-12-12:
You probably want `-S`, which will enable the stub and wait for a debugger to attach. `-s` doesn't wait.

wdormann, 2016-12-12:
Got it.  I needed to add "-s -S" to the qemu commandline in run.armel-debug.sh.  This now allows me to attach gdb remotely.  But while gdb is connected and I can continue to start up the kernel, it continues through the exception without stopping or giving me any info.  Probably just an artifact of not being familiar with kernel-level gdb stuff.  I'm used to userspace.   I guess I'll have to dig a bit deeper.

However, if you think you may have a setup that could find where the problem is, that'd be great!

ddcc, 2016-12-12:
Yeah, kernel debugging can be tricky, especially since there may not be any debugging symbols included with the default kernel binary (I don't recall). It's been a while, so I'd suggest trying to break around `try_to_run_init_process()` in `main.c`, and go from there.

I think (2) would be probably be easier, using QEMU's usermode gdb stub. Chaining the correct sequence of chroot / qemu-user commands may be the only tricky part, though binfmt_misc would probably make it easier. I don't exactly recall, but it may even be possible to skip the chroot by starting in the root directory and adjusting the `-L` emulation parameter.

xiaohuajiao, 2018-03-08:
any update for "Kernel panic - not syncing: Attempted to kill init!" ?

ddcc, 2018-03-08:
@xiaohuajiao You should make a separate issue for unrelated errors. That error is not the problem; the kernel always panics when the `init` process dies. The problem is why the `init` process dies, but since that binary is firmware specific, there is no generic solution. Usually it involves fixing up some missing NVRAM key.

xiaohuajiao, 2018-03-16:
for kernel panic `Kernel panic - not syncing: Attempted to kill init`. I met one firmware that also have this problem. i solve(or bypass) it by modify a bit of kernel code. 

My case is that: 
> `/sbin/init` in that firmware must accept at least one argument, else it will return non-zero, then kernel panic. we can found code in kernel, which try to run "/sbin/init", without any more argument. So we just need to reverse the main function in init, then modify the kernel code and give a valid argument to init, then it will return zero...

Admittedly, it's not a nice solution or a right solution. just a way to bypass this...