firmadyne/firmadyne #29: firmadyne.config not found error in inferNetwork.sh

txjoe, 2016-12-13:
I keep getting this error

> `Error: Could not find 'firmadyne.config'!`

This is weird because firmadyne.config is present in the parent directory (the directory which contains the scripts directory).

So I commented out the part where it checks for the config file. 
Then it ran partially and I got this error

```

> Running firmware 1: terminating after 60 secs...
> Error: Could not find 'firmadyne.config'!

```

ddcc, 2016-12-13:
Did you set `FIRMWARE_DIR` to the correct path? See the [README](https://github.com/firmadyne/firmadyne#usage).

txjoe, 2016-12-13:
I did

```
#if [ -e ./firmadyne.config ]; then
#    source ./firmadyne.config
#elif [ -e ../firmadyne.config ]; then
#source ../firmadyne.config
#else
#    echo "Error: Could not find 'firmadyne.config'!"
#    exit 1
#fi

But this part of the script isn't using the FIRMWARE_DIR right?
It's just checking the current & the parent directories for the firmadyne.config file.
And the file is present in the parent directory.

Thanks for your quick response!
```

ddcc, 2016-12-13:
No, you shouldn't be editing that file. Open `firmadyne.config`, uncomment and set the `FIRMWARE_DIR` variable.

txjoe, 2016-12-13:
The FIRMWARE_DIR variable was set all along. I re-checked the path just to be sure, that doesn't seem to be the problem

```
if [ -e ./firmadyne.config ]; then
    source ./firmadyne.config
elif [ -e ../firmadyne.config ]; then
source home/mydir/firmadyne/firmadyne.config
else
    echo "Error: Could not find 'firmadyne.config'!"
    exit 1
fi
```

^^ Using the absolute path there worked for me.
I got a new error now though
`
qemu-system-mips: -drive if=ide,format=raw,file=/home/txjoe/code/FAT/firmadyne/scratch//1//image.raw: Could not open '/home/txjoe/code/FAT/firmadyne/scratch//1//image.raw': No such file or directory`

Still trying to get firmadyne to work. Any ideas?

ddcc, 2016-12-13:
Did you set `FIRMWARE_DIR` with an absolute path? I recommend you don't modify that snippet of code, because it's in almost every script file, so you'd have to update all of them. Did you follow the instructions and run makeImage?

txjoe, 2016-12-13:
Hey, thanks for the help, the error was related to makeImage, I missed the error output from makeImage! Thanks for your quick responses!

closed 2016-12-23 by ddcc