firmadyne/firmadyne #30: ProgrammingError: relation "image" does not exist

xhopkins, 2016-12-23:
Hi,
I have a problem when executing this script:

_./sources/extractor/extractor.py -b Netgear -sql 127.0.0.1 -np -nk "WNAP320.zip" images_

I obtained the following output:

_File "./sources/extractor/extractor.py", line 345, in update_database_
_ProgrammingError: relation "image" does not exist_
_LINE 1: UPDATE image SET rootfs_extracted='True' WHERE id='WNAP320.z..._

There is a problem when creating and populating the Database

Any idea?

xhopkins, 2016-12-23:
An update...
Immediately after executing the scripts this is the first output:

> Traceback (most recent call last):
>   File "./sources/extractor/extractor.py", line 274, in generate_tag
>      cur.execute("SELECT id FROM brand WHERE name=%s", (brand, ))
> ProgrammingError: relation "brand" does not exist
> LINE 1: SELECT id FROM brand WHERE name='Netgear'

Regards,

ddcc, 2016-12-23:
Did you load the database schema from the [instructions](https://github.com/firmadyne/firmadyne#database)?

xhopkins, 2016-12-23:
Thank you, I've just resolved my problem... 
I forget to install some python libraries.

Please close this Issue

closed 2016-12-23 by ddcc