firmadyne/firmadyne #32: Firmware Login - inferNetwork

decidedlygray, 2017-01-03:
(title intentionally similar to #23, but i think this is a different problem..)

Everything worked fine for the latest version of the example firmware works as expected. However, when I try another firmware:
```
wget http://www.downloads.netgear.com/files/GDC/WNDR3400V3/WNDR3400v3-V1.0.1.4_1.0.52.zip
```

I can't connect to the network interface. Running the inferNetwork script gives me:
```
blackstone@blackstone-vm:/opt/firmadyne$ ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipsel
Running firmware 1: terminating after 60 secs...
Bad SWSTYLE=0x04
Inferring network...
Interfaces: [('br0', '192.168.1.1')]
Done!
```

Run.sh is generated. I'm seeing a lot of "unable to open key" errors, like this when I run the run.sh:
```
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_get: Waiting for semaphore initialization (Key: 410e00f9, Semaphore: 8001)...
sem_unlock: Unable to get semaphore!
nvram_get_buf: Unable to open key: /firmadyne/libnvram/wl0_wep!
```

I was wondering if you're able to recreate this, and if this is a bug, or if this is specific to my setup?

Thanks!

ddcc, 2017-01-03:
It sounds like there may be a problem with the NVRAM emulation with that firmware. If the firmware requires a NVRAM value for a specific key during the process, and it's not a key that a default value is available for, then firmadyne returns an error for the NVRAM API call.

Unfortunately, resolving this is difficult, because it usually involves disassembling or analyzing the firmware to determine what entries are required, and whether there is a source of default NVRAM entries available, then updating and recompiling the NVRAM library to account for this.

decidedlygray, 2017-01-03:
Interesting. I have a few followup questions if you don't mind, regarding this issue. It sounds like it might be something worth looking into, and I'm sure if I do I'll learn some stuff along the way :) To help get me started:

1. Is this something that could be accomplished using a tool like radare2? Or is there another (free/cheap) tool you'd recommend that makes things easier?
2. Would you happen to have any links to online resources that could point me in the right direction, specifically on how I would start going about trying to find/enumerate NVRAM values/keys/entries?
3. In your opinion, would this be something that could eventually be automated via a Python script, if someone were to put in the work up front to research a bit on methods to enumerate NVRAM details?

Thank you for your time

ddcc, 2017-01-03:
1. I've been using IDA, but radare2 or any other disassembler should work fine. I'd recommend starting with the log file to see how far the emulation gets, and then following up in the disassembly, starting with the `init` binary. It may also be useful to review the comments in issue #28.
2. I'd recommend looking through the libnvram repository, the relevant parts of the paper that mention libnvram, and the FAQ in the README for this repository. It may also be useful to read through https://shadow-file.blogspot.com/2015_01_01_archive.html, which discusses `nvram-faker`, a different library but with the same operating principle.
3. It's definitely possible to implement a more automatic NVRAM emulation, potentially using program analysis techniques, since the API is fairly fixed. Attempting to automatically infer the NVRAM API would be more tricky, but likely still doable. It's one of things that I plan to do look at when I get more time.

decidedlygray, 2017-01-04:
Much appreciated, thank you for the information!

closed 2017-01-04 by decidedlygray