firmadyne/firmadyne #33: netgear123456 login

bennyjiang, 2017-01-05:
when I ran ./scratch/1/run.sh,
netgear123456 login: [   38.020000] brtrunk: no IPv6 routers present
[   38.360000] eth0: no IPv6 routers present
Welcome to SDK.
Have a lot of fun...
netgear123456 login: 

I want to know the default  netgear username and password

bennyjiang, 2017-01-05:
delete all files ?(/etc/security/)

bennyjiang, 2017-01-05:
delete all files?(/etc/security/) @ddcc

ddcc, 2017-01-05:
No, the file `/etc/securetty`

bharat089, 2018-01-30:
Even after removing /etc/securetty, its not working. It stucks in netgear123456 login.

ddcc, 2018-02-01:
I just tried it again, and it works for me.

```
ddcc@virtual:/home/ddcc/firmadyne$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.3 LTS
Release:        16.04
Codename:       xenial

ddcc@virtual:/home/ddcc/firmadyne$ wget http://www.downloads.netgear.com/files/GDC/WNAP320/WNAP320%20Firmware%20Version%
202.0.3.zip
--2018-02-01 13:53:28--  http://www.downloads.netgear.com/files/GDC/WNAP320/WNAP320%20Firmware%20Version%202.0.3.zip
Resolving www.downloads.netgear.com (www.downloads.netgear.com)... 23.194.181.234
Connecting to www.downloads.netgear.com (www.downloads.netgear.com)|23.194.181.234|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5362552 (5.1M) [application/zip]
Saving to: ‘WNAP320 Firmware Version 2.0.3.zip’

WNAP320 Firmware Version 2.0.3.z 100%[==========================================================>]   5.11M  9.96MB/s    in 0.5s

2018-02-01 13:53:29 (9.96 MB/s) - ‘WNAP320 Firmware Version 2.0.3.zip’ saved [5362552/5362552]

ddcc@virtual:/home/ddcc/firmadyne$ python2 ./sources/extractor/extractor.py -sql 127.0.0.1 -b Netgear -np -nk "WNAP320 F
ir.0.3.zip" images
>> Database Image ID: 1

/home/ddcc/firmadyne/WNAP320 Firmware Version 2.0.3.zip
>> MD5: 51eddc7046d77a752ca4b39fbda50aff
>> Tag: 1
>> Temp: /tmp/tmpPZe5Lg
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>>>> Zip archive data, at least v2.0 to extract, compressed size: 1197, uncompressed size: 2667, name: ReleaseNotes_WNAP320_fw_2.0.3
.HTML
>> Recursing into archive ...

/tmp/tmpPZe5Lg/_WNAP320 Firmware Version 2.0.3.zip.extracted/WNAP320_V2.0.3_firmware.tar
        >> MD5: 6b66d0c845ea6f086e0424158d8e5f26
        >> Tag: 1
        >> Temp: /tmp/tmpRe12LD
        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
        >>>> POSIX tar archive (GNU), owner user name: "gz.uImage"
        >> Recursing into archive ...

/tmp/tmpRe12LD/_WNAP320_V2.0.3_firmware.tar.extracted/kernel.md5
                >> MD5: 0e15e5398024c854756d3e5f7bc78877
                >> Skipping: text/plain...

/tmp/tmpRe12LD/_WNAP320_V2.0.3_firmware.tar.extracted/root_fs.md5
                >> MD5: b43dc86ce23660652d37d97651ba1c77
                >> Skipping: text/plain...

/tmp/tmpRe12LD/_WNAP320_V2.0.3_firmware.tar.extracted/rootfs.squashfs
                >> MD5: 7ce95b252346d2486d55866a1a9782be
                >> Tag: 1
                >> Temp: /tmp/tmpDkWtqE
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >>>> Squashfs filesystem, big endian, lzma signature, version 3.1, size: 4433988 bytes, 1247 inodes, blocksize: 6553
6 bytes, created: 2011-06-23 10:46:19
                >>>> Found Linux filesystem in /tmp/tmpDkWtqE/_rootfs.squashfs.extracted/squashfs-root!
                >> Skipping: completed!
                >> Cleaning up /tmp/tmpDkWtqE...
        >> Skipping: completed!
        >> Cleaning up /tmp/tmpRe12LD...
>> Skipping: completed!
>> Cleaning up /tmp/tmpPZe5Lg...

ddcc@virtual:/home/ddcc/firmadyne$ ./scripts/getArch.sh ./images/1.tar.gz
./bin/busybox: mipseb
Password for user firmadyne:
[... 72 lines cut ...]

ddcc@virtual:/home/ddcc/firmadyne$ sudo ./scripts/uount.sh 1
----Running----
----Unmounting----
----Disconnecting Device File----
loop deleted : /dev/loop0

ddcc@virtual:/home/ddcc/firmadyne$ ./scratch/1/run.sh
Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 1000
Bringing up TAP device...
Adding route to 192.168.0.100...
Starting firmware emulation... use Ctrl-a + x to exit

....

Welcome to SDK.

Have a lot of fun...

netgear123456 login: root
Password:

[root@netgear123456 /root]# 
```

bharat089, 2018-02-02:
I have tried the above mentioned steps one by one. I am using 14.04 on virtual box. It again stucks on netgear login. rest all the steps are working fine and giving the same output as yours, except last step of running run.sh. By the way , what was your password for netgear123456?

ddcc, 2018-02-02:
It's the same as the instructions; the password is just `password`. After logging in, you can do whatever you want to the emulated firmware. Note that I've omitted a bunch of the kernel messages and startup text from the boot. Are you sure it's not logging in; e.g. if you hit enter a few times, it doesn't do anything?

bharat089, 2018-02-06:
Thanks so much. It worked ....!!!!

bharat089, 2018-02-07:
Could you please give me a example of few commands which I can try after logging into the Netgear123456?

Any help is appreciated...!!!!

ddcc, 2018-02-07:
I'm not sure what you want to do. As stated in the README and paper, this is a tool for emulating embedded firmware. You get the equivalent of having attached a physical console to the device, but without needing the actual hardware. So you could look through the binaries on the system, figure out what services are running and listening on the network, etc.