firmadyne/firmadyne #36: Setting NVRAM default values without recompilation?

dfbravo, 2017-01-06:
In #32 @ddcc mentioned that adding new NVRAM values requires adding them to `config.h` and recompiling the library. Can default values be added through the `/firmadyne/libnvram.override/` folder instead? 

The `nvram_init` function calls `nvram_set_default` which starts the process. `nvram_set_default_builtin` is called which uses the `NVRAM_DEFAULTS` macro to set the default values hard coded in `nvram.h`. Back in `nvram_set_default` the `NVRAM_DEFAULTS_PATH` is used which seems to load default values from files within the rootfs. Finally `nvram_set_default_image` is called which runs `system("/bin/cp "OVERRIDE_POINT"* "MOUNT_POINT);`

Is it possible to use `nvram_set_default_image` instead of  `nvram_set_default_builtin` to load default values? Another folder can be used to store the default values and are copied over to the tmpfs used for NVRAM.

I'm want to remove the recompilation step when changing default values in NVRAM because I'm interested in scripting the process of finding missing default values and adding them.

ddcc, 2017-01-07:
Yes, they can be put there as well instead of directly into the code. See https://github.com/firmadyne/firmadyne/blob/master/scripts/fixImage.sh for examples of where this is used.

However, generally there's not a single key-value pair that is necessary for the system to boot, but tens if not hundreds of key-value pairs. Thus, the easiest way to get a new firmware working is to figure out where the manufacturer stores the default values (e.g. after resetting the system), and add it to `NVRAM_DEFAULTS_PATH ` in https://github.com/firmadyne/libnvram/blob/master/config.h .

closed 2017-01-07 by ddcc

tulipszs, 2019-02-19:
@ddcc Can you provide some examples of where manufacturer stores default nvram values? Or how you would go about finding out? Thanks

ddcc, 2019-02-19:
Unfortunately, it's a somewhat manual process of figuring out the NVRAM is being handled, and then looking to see where the values are coming from. If you look at the linked `config.h` file, sometimes the default values are stored in a file on the filesystem (`PATH`), or sometimes they are key-value pairs stored as a table in a shared symbol (`TABLE`). You probably want to search inside some of the binaries / libraries, either using something basic like `strings <foo> | grep <bla>`.

Alternatively, you can work backwards by figuring out where the boot process / program of interest is crashing, examine the last couple of NVRAM key/value pairs that were requested, overriding them with different values and seeing if you get farther. Additionally, disassembling / decompiling them and looking for anything interesting makes this process a lot easier.