firmadyne/firmadyne #37: VLAN problem

xOmenus, 2017-01-08:
I can not fix this. What could be the problem?

```
Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 0
Initializing VLAN...
./run.sh: line 34: VLANID: unbound variable

```

closed 2017-01-08 by ddcc

0roman, 2017-02-13:
Firstly, great work with firmadyne. Secondly, the same problem still exists for a number of TP link router firmwares ( e.g. http://www.tp-link.de/download/Archer-C2_V1.html all firmware releases of version 1 ). 

Runing run.sh displays the following:

    Creating TAP device tap3_0...
    Set 'tap3_0' persistent and owned by uid 1000
    Initializing VLAN...
    ./scratch/3/run.sh: line 35: TAPDEV: unbound variable

Any suggestions? Here is my log file : [qemu.initial.serial.log.tar.gz](https://github.com/firmadyne/firmadyne/files/771417/qemu.initial.serial.log.tar.gz)

Best regards,

Roman

ddcc, 2017-02-13:
Just to check, you updated your local copy of the repository to the latest, and reran the inferNetwork.sh script? If so, could you upload a copy of the run.sh file?

ddcc, 2017-02-13:
Nevermind, found the problem. I think it should be fixed now, let me know if it isn't. Thanks!

commit 027c1d2c (2017-01-08): Fix #37
commit 7445b5b8 (2017-02-13): Further fix #37