firmadyne/firmadyne #38: wnap320 new Firmware wont emulate

jan85f, 2017-01-09:
Hi,

the old WNAP320 Firmware Version 2.0.3 could be emulated without a problem.

the new version (http://www.downloads.netgear.com/files/GDC/WNAP320/WNAP320_V3.5.5.0.zip)
does not create the run.sh because it cant find an interface.
it is the same FS as in the old, even the same  default interface-ip (192.168.0.100) is set in the firmware. 

Maybe you can give it a try =)

Kind Regards and awesome work what you all did!

ddcc, 2017-01-09:
I don't have time to look into the root cause, but for now you can just copy the `run.sh` file generated for the previous version, and it should work for the new version.

ddcc, 2017-01-09:
Unfortunately, that firmware image is stalling somewhere in the boot process, and doesn't attempt to even bring up the network. You might want to compare the filesystem between the two versions, and see what changes have been made to the boot process.

jan85f, 2017-01-10:
Hi Dominic,

thanks, I will go for the diff, thanks for the advice =)