firmadyne/firmadyne #43: 'statSetsockopt : Protocol not available' Error

0roman, 2017-02-21:
Running **run.sh** displays the following error when emulating several tp link router firmware from the Archer C2 and C20i. In the following we used [Archer_C2(EU)_V1_160512.zip](https://github.com/firmadyne/firmadyne/files/789701/Archer_C2.EU._V1_160512.zip).
 

    sh: can't create /proc/tplink/eth_forward: nonexistent directory
    [ openSwitchForwarding ] 204:  get eth_forward status error

    statSetsockopt
    : Protocol not available
    [ stat_getsockopt ] 219:  257 opt error

    statSetsockopt
    : Protocol not available
    [ stat_getsockopt ] 219:  257 opt error

    ...

Any suggestions?

Our run.sh script an log files: 

[run.sh.tar.gz](https://github.com/firmadyne/firmadyne/files/789753/run.sh.tar.gz)

[qemu.initial.serial.log.tar.gz](https://github.com/firmadyne/firmadyne/files/789754/qemu.initial.serial.log.tar.gz)

[qemu.final.serial.log.tar.gz](https://github.com/firmadyne/firmadyne/files/789755/qemu.final.serial.log.tar.gz)

ddcc, 2017-02-21:
Looks to be some sort of networking-related failure. I'd guess that there's a program that's opening some sort of socket socket or protocol family with unsupported type. In the simplest case, it'd just be flipping a kernel configuration to build the request socket type or protocol family. You're looking for a return value of `ENOPROTOOPT` from one of the requested socket types or protocol families listed in `sys/socket.h`.

tUn4, 2018-06-27:
hello,
i just got the same issue with the tplink firmware. is there any way to fix this.
"statSetsockopt
: Protocol not available
[ stat_getsockopt ] 219:  257 opt error"

ddcc, 2018-06-27:
The situation hasn't changed. You'd need to figure out the source of the problem, as mentioned in my last comment, and correct it.