firmadyne/firmadyne #46: Debugging firmware images that aren't successfully emulated

prashast, 2017-04-28:
Hey @ddcc , I had a question regarding the debugging framework for binaries that aren't successfully emulated. I wanted to remotely debug a web server binary that was running as a part of the emulation but I was having trouble connecting to the gdb stub that I was running in QEMU. Do you have any pointers on as to how you go about debugging these binaries?

ddcc, 2017-04-28:
Unfortunately, debugging system-mode QEMU is a pain, so I try to avoid it, and substitute with workarounds when possible. There's a discussion of this in the comments for issue #28 : https://github.com/firmadyne/firmadyne/issues/28#issuecomment-266573357 , and in the next few comments.

Aside from using QEMU's built-in stubs for system/user-mode emulation, another approach is to use system-mode emulation, build a statically-linked GDB stub for the target, and run it inside the emulator attached to the binary of interest. Of course, you'll need a cross-compile toolchain, which can also be difficult to get ahold of; you can either build it from scratch using e.g. buildroot, or attempt to find GPL sources and look for a toolchain in there. Alternatively, if the platform is popular enough, you can usually find pre-compiled binaries online. Also, if you have access to IDA Pro, it comes with its own pre-compiled debug stubs (not GDB-compatible) in the install directory.

prashast, 2017-04-28:
Yeah, I currently had a statically linked gdbserver binary which I had attached to a webserver binary but was having trouble connecting to the stub from the host using the tap/bridge interface. Either way, thanks for the prompt reply.

prashast, 2017-05-05:
Hi, so I've been trying for a while to setup the debugging framework so that I can figure out where exactly is the web server binary crashing when I'm trying to emulate it and for that I had chosen to go the statically linked gdbserver route as I had mentioned before. However, I am having trouble connecting my host GDB to the gdb stub running on the emulated debian mipsel machine. 

I had setup the bridge and the tap interface like so:
```
sudo brctl addbr br0
# ip addr add 192.168.179.1/24 broadcast 192.168.179.255 dev br0
# ip link set br0 up
# ip tuntap add dev tap0 mode tap user $(whoami)
# ip link set tap0 up promisc on
 # ip link set tap0 master br0
 # dnsmasq --interface=br0 --bind-interfaces --dhcp-range=192.168.179.2,192.168.179.254
```

After which I ran the qemu using the tap interface that I had setup. My guest gets an ip address(192.168.179.89) in the range set by the dnsmasq and I am able to ping the guest from the host and vice versa and even able to ssh into the guest from the host. However when I setup a gdb session `./gdbserver --multi 192.168.179.89:2345` and I try to connect to it from the host it just gives me a timeout response on the host side. I ran tcpdump on the tap interface that was setup and I can see a bunch of SYN-ACK packets being exchanged. I'm pretty sure I have to set some Iptable rules in order to make this work and was wondering if you had any insight into it regarding this or the kind of firewall rules I need to setup in order to make this work or things I can do to debug this.

ddcc, 2017-05-06:
Are you manually configuring the network instead of using the autogenerated run.sh script? Instead of binding gdbserver to a specific IP, I think you can just do `gdbserver :12345 foo`. Sometimes the networking can be a bit finicky where one host can't seem to talk to the other reliably, in which case you can try reversing the client/server using netcat, socat, or ssh (e.g. dropbear) using port forwarding.

prashast, 2017-05-08:
Thanks for being so receptive, I truly appreciate it. I tried a couple of the alternatives but I keep stumbling across this error that whenever I tried to connect to the statically linked gdbserver from the host whether it be a tcp connection or through ssh port forwarding.
```
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
```
My connection between the host and the guest seems to be setup correctly because I am able to ping the guest from the host and vice versa.  I was using gdb-7.10 and was wondering if that could be the issue since I am using a 2.6.32 kernel. Would using a deprecated version of GDB work or is there something else that might be going wrong. The gdbserver binary itself seemed to have no issues when I tested it on the host but it only has issues when I'm trying to connect to it from the host. Again, thanks for answering all these queries.

ddcc, 2017-05-08:
If you're on Debian/Ubuntu, make sure you've installed `gdb-multiarch`. Depending on the mode that you launched gdbserver with, to connect to it in gdb you'll need to use either `target remote` or `target extended-remote`.

prashast, 2017-05-08:
I tried connecting with both `gdb-mutliarch` and the `mipsel-unknown-linux-uclibc-gdb` and was using the appropriate modes on both sides, --multi with target extended-remote or target remote if my gdbserver was attached to a program but still the same error was cropping up.

ddcc, 2017-05-08:
Is this over a serial link? It seems like some other people encountered this when they didn't configure the serial port and baudrate correctly. Otherwise, I can only suggest looking at it with wireshark and seeing if that's useful.

prashast, 2017-05-12:
I finally managed to connect to the remote gdbserver when I used a newer kernel. For some reason the gdbserver wasn't working on the 2.6 kernel. However, now I'm facing this issue where whenever I chroot ,attach the webserver binary to the gdbserver and then when I try to connect `gdb-multiarch` using the commands: 
```
set arch mips
set endian little
set sysroot squashfs-root/  #my firmware root directory
target remote 192.168.179.89:2345
```
It gives me this error:

> Error while reading shared library symbols for squashfs-root/lib/ld-uClibc.so.0:
> Remote connection closed
> /build/gdb-cXfXJ3/gdb-7.11.1/gdb/thread.c:89: internal-error: inferior_thread: Assertion `tp' failed.

I even tried the gdb which comes with the cross compiler with which I had statically compiled my gdbserver with and I get the same issue. The `ld-uClibc.so.0` is a symbolic link to `ld-uClibc-0.9.28.so` though I doubt that would be an issue. I've set up a bind mount of the /proc inside the chroot because gdb complained that it wasn't able to read the proc entry of the pid that was created for the server.

> Reading symbols from target:/lib/ld-uClibc.so.0...(no debugging symbols found)...done.
> Error while reading shared library symbols for target:/lib/ld-uClibc.so.0:
> Remote connection closed

I tried to connect IDA to the gdbserver stub which works fine and attaches to the process successfully as long as I don't set a breakpoint. Without setting a breakpoint if I keep on continuing the process remotely it runs to it's completion, however as soon as i set a breakpoint, it breaks at that point but as soon as i resume it exits and the gdbserver segfaults with the error.

> linux_wait_1: Assertion `target_options & TARGET_WNOHANG' failed. 

I was wondering if there is anything that I'm doing wrong over here that you could point it out to me or a way that I can proceed to debug this issue.

ddcc, 2017-05-12:
I'm a little confused about your setup. If you're using the remote gdbserver, then presumably the whole firmware is running in system-mode QEMU, so why is there a local chroot involved? You do need an extracted local copy of the firmware filesystem for GDB, but I believe setting the sysroot in GDB should take care of redirecting absolute paths into that. Are you sure the extracted ld-uClibc.so.0 file is valid ELF file and matches the one inside the QEMU image? I don't think I've seen either error before, I can only suggest trying again with the same version of gdb and gdbserver. You could also turn on verbose/debugging for both gdb and gdbserver to see if there's anything useful.

prashast, 2017-05-12:
Yeah, I managed to get it to run using IDA, thanks for all the help!

closed 2017-05-12 by prashast

wjaxx, 2019-04-25:
> I finally managed to connect to the remote gdbserver when I used a newer kernel. For some reason the gdbserver wasn't working on the 2.6 kernel. However, now I'm facing this issue where whenever I chroot ,attach the webserver binary to the gdbserver and then when I try to connect `gdb-multiarch` using the commands:
> 
> ```
> set arch mips
> set endian little
> set sysroot squashfs-root/  #my firmware root directory
> target remote 192.168.179.89:2345
> ```
> It gives me this error:
> 
> > Error while reading shared library symbols for squashfs-root/lib/ld-uClibc.so.0:
> > Remote connection closed
> > /build/gdb-cXfXJ3/gdb-7.11.1/gdb/thread.c:89: internal-error: inferior_thread: Assertion `tp' failed.
> 
> I even tried the gdb which comes with the cross compiler with which I had statically compiled my gdbserver with and I get the same issue. The `ld-uClibc.so.0` is a symbolic link to `ld-uClibc-0.9.28.so` though I doubt that would be an issue. I've set up a bind mount of the /proc inside the chroot because gdb complained that it wasn't able to read the proc entry of the pid that was created for the server.
> 
> > Reading symbols from target:/lib/ld-uClibc.so.0...(no debugging symbols found)...done.
> > Error while reading shared library symbols for target:/lib/ld-uClibc.so.0:
> > Remote connection closed
> 
> I tried to connect IDA to the gdbserver stub which works fine and attaches to the process successfully as long as I don't set a breakpoint. Without setting a breakpoint if I keep on continuing the process remotely it runs to it's completion, however as soon as i set a breakpoint, it breaks at that point but as soon as i resume it exits and the gdbserver segfaults with the error.
> 
> > linux_wait_1: Assertion `target_options & TARGET_WNOHANG' failed.
> 
> I was wondering if there is anything that I'm doing wrong over here that you could point it out to me or a way that I can proceed to debug this issue. I have no idea why it would have an issue reading symbols from the shared library, I even tried to set the sysroot to the target using `set sysroot target:/` so that I can download the shared library and read the symbols from there but still the same issue kept coming up. The binary which I'm debugging is a mips lsb 32 bit executable.

I have the same issues that gdbserver cannot work in 2.6.x kernel.But I don't know how to change kernel in firmadye.Could you please give me a solution to get new kernel which firmadyne could support.
Thanks in advance.

ddcc, 2019-04-26:
I don't think there's a straightforward solution. You'll need to rebase the firmadyne patches from e.g. the 2.6.x kernel onto e.g. a new Linux stable kernel, update the kernel configuration, and build the new kernel.

sssyyynnngithub, 2019-10-24:
@ddcc prashast I want to use the gdbserver on the target ,so  I use the kernel vmlinux-3.2.0-4-4kc-malta instead. But  firmadyne can not infer network ip :
root@kali:~/firmadyne# ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipseb
Running firmware 1: terminating after 60 secs...
qemu-system-mips: -net nic,vlan=0: 'vlan' is deprecated. Please use 'netdev' instead.
qemu-system-mips: terminating on signal 2 from pid 1945 (timeout)
Inferring network...
Interfaces: []
Done!
and can not start network :
qemu-system-mips: -net nic,vlan=0: 'vlan' is deprecated. Please use 'netdev' instead.
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.2.0-4-4kc-malta (debian-kernel@lists.debian.org) (gcc version 4.6.3 (Debian 4.6.3-14) ) #1 Debian 3.2.51-1
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0067b000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f894000 @ 0076b000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 60768 bytes for tracking 1899 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x0000ffff
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x0000ffff
[    0.000000] On node 0 totalpages: 65535
[    0.000000] free_area_init_node: node 0, pgdat 806bfd60, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60959 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65023
[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252252k/254544k available (4614k kernel code, 2292k reserved, 1283k data, 220k init, 0k highmem)
[    0.000000] NR_IRQS:256
[    0.000000] CPU frequency 200.00 MHz
[    0.000000] Console: colour dummy device 80x25
[    0.000000] Calibrating delay loop... 1087.48 BogoMIPS (lpj=2174976)
[    0.020000] pid_max: default: 32768 minimum: 301
[    0.020000] Security Framework initialized
[    0.024000] AppArmor: AppArmor disabled by boot time parameter
[    0.024000] Mount-cache hash table entries: 512
[    0.028000] Initializing cgroup subsys cpuacct
[    0.028000] Initializing cgroup subsys memory
[    0.028000] Initializing cgroup subsys devices
[    0.028000] Initializing cgroup subsys freezer
[    0.028000] Initializing cgroup subsys net_cls
[    0.028000] Initializing cgroup subsys blkio
[    0.032000] Initializing cgroup subsys perf_event
[    0.040000] devtmpfs: initialized
[    0.048000] print_constraints: dummy: 
[    0.048000] NET: Registered protocol family 16
[    0.056000] bio: create slab <bio-0> at 0
[... 298 lines cut ...]

Starting Translator...      sh: cannot create /proc/sys/net/bridge/bridge-http-redirect-flush-mac: nonexistent directory
sh: cannot create /proc/sys/net/bridge/bridge-http-redirect-enabled: nonexistent directory
[http_redirect_tr]

Starting Translator...      [dhcp]

Starting Translator...      kill: cannot kill pid 611: No such process
[ntp]

Starting Translator...      [timezone]

Starting Translator...      kill: cannot kill pid 624: No such process
[sc_radio]
Error in opening the device.
: No such device

System initilization is ..  [DONE...]

Welcome to SDK.

Have a lot of fun...

netgear123456 login: 
Could you help me ?Thank you !

sssyyynnngithub, 2019-11-23:
@ddcc Sir ,Could you give me the instructions on how to rebase the firmadyne patches from 2.6.x kernel noto a new linux stable kernel , update the kernel configuration, and build the new kernel . Thank you very much !

sssyyynnngithub, 2019-11-23:
@prashast . Sir , do you solve the problem of using IDA to debug on firmadyne? Could you please share how do you do it ? Thank you very much !