firmadyne/firmadyne #47: Firmware not taking IP

HacKeD0x90, 2017-05-04:
I was trying to emulate the following firmware, and after a long night to install the required tools. 
i ran in to a strange thing, it didn't get any ip and no interface was created.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
mke2fs 1.42.12 (29-Aug-2014)
The file /dev/mapper/loop0p1 does not exist and no size was specified.
Please check the makeImage function
Everything is done for the image id 3
Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 8852
Querying database for architecture... mipseb
Running firmware 3: terminating after 60 secs...
Inferring network...
Interfaces: []
Done!

Running the firmware finally : 

appreciate your assist on where i can start troubleshooting

ddcc, 2017-05-04:
I removed the post from the other issue, because it's not the same problem. I don't recognize some of the output that you posted. Did you install firmadyne and follow the instructions from the readme?

HacKeD0x90, 2017-05-04:
Yes i installed the dependencies like sasquach, binwalk.
Setup the postgress sql with firmadyne,firmadyne as UN and password and set up the database firmware with the scheme.
And when i runs successfuly except it doesn't take any ip address and i can't find any interface created. 
But it doesn't give any compilation error so i don't think it's a dependencies issue.
But I'm confused where to troubleshoot?

ddcc, 2017-05-04:
Normally, running firmadyne doesn't involve compiling any code, so there shouldn't be any compilation errors. It doesn't look like the disk image for the firmware was created. What is the output of the makeImage step from the readme?

HacKeD0x90, 2017-05-04:
sudo ./scripts/makeImage.sh 1
[sudo] password for sakr: 
Sorry, try again.
[sudo] password for sakr: 
Querying database for architecture... Password for user firmadyne: 
mipsel
----Running----
----Copying Filesystem Tarball----
----Creating QEMU Image----
Formatting '/home/sakr/IOT/firmadyne/scratch//1//image.raw', fmt=raw size=1073741824 
----Creating Partition Table----
Device contains neither a valid DOS partition table, nor Sun, SGI or OSF disklabel
Building a new DOS disklabel with disk identifier 0x7d4885ce.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)

Command (m for help): Building a new DOS disklabel with disk identifier 0xf2d02e12.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)

Command (m for help): Partition type:
   p   primary (0 primary, 0 extended, 4 free)
   e   extended
Select (default p): Partition number (1-4, default 1): First sector (2048-2097151, default 2048): Using default value 2048
Last sector, +sectors or +size{K,M,G} (2048-2097151, default 2097151): Using default value 2097151

Command (m for help): The partition table has been altered!

Syncing disks.
----Mounting QEMU Image----
add map loop0p1 (252:0): 0 2095104 linear /dev/loop0 2048
----Creating Filesystem----
mke2fs 1.42 (29-Nov-2011)
Discarding device blocks: done                            
Filesystem label=
OS type: Linux
Block size=4096 (log=2)
Fragment size=4096 (log=2)
Stride=0 blocks, Stripe width=0 blocks
65536 inodes, 261888 blocks
13094 blocks (5.00%) reserved for the super user
First data block=0
Maximum filesystem blocks=268435456
8 block groups
32768 blocks per group, 32768 fragments per group
8192 inodes per group
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376

Allocating group tables: done                            
Writing inode tables: done                            
Writing superblocks and filesystem accounting information: done

----Making QEMU Image Mountpoint----
----Mounting QEMU Image Partition 1----
----Extracting Filesystem Tarball----
----Creating FIRMADYNE Directories----
----Patching Filesystem (chroot)----
Creating /etc/TZ!
Creating /etc/hosts!
Creating /etc/passwd!
Removing /etc/scripts/sys_resetbutton!
----Setting up FIRMADYNE----
----Unmounting QEMU Image----
loop deleted : /dev/loop0

ddcc, 2017-05-04:
That looks fine. What about inferNetwork.sh?

HacKeD0x90, 2017-05-04:
sakr@vulMachine:~/IOT/firmadyne$ ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipsel
Running firmware 1: terminating after 60 secs...
Inferring network...
Traceback (most recent call last):
  File "/home/sakr/IOT/firmadyne/scripts//makeNetwork.py", line 400, in <module>
    main()
  File "/home/sakr/IOT/firmadyne/scripts//makeNetwork.py", line 397, in main
    process(infile, iid, arch, endianness, makeQemuCmd, outfile)
  File "/home/sakr/IOT/firmadyne/scripts//makeNetwork.py", line 294, in process
    data = open(infile).read()
IOError: [Errno 2] No such file or directory: '/home/sakr/IOT/firmadyne/scratch//1/qemu.initial.serial.log'
sakr@vulMachine:~/IOT/firmadyne$ vim scripts/inferNetwork.sh 
sakr@vulMachine:~/IOT/firmadyne$ ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipsel
Running firmware 1: terminating after 60 secs...
timeout: unrecognized option '--preserve-status'
Try `timeout --help' for more information.
sakr@vulMachine:~/IOT/firmadyne$ vim scripts/inferNetwork.sh 
sakr@vulMachine:~/IOT/firmadyne$ ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipsel
Running firmware 1: terminating after 60 secs...
timeout: unrecognized option '--preserve-status'
Try `timeout --help' for more information.

i did comment to this part in the code but maybe this is the problem?

ddcc, 2017-05-04:
No, that part shouldn't be commented out. Part of the problem is that the `timeout` binary on your system doesn't seem to support the `preserve-status` argument, so it's terminating instead of emulating the firmware. Do you have the latest version of the `coreutils` package? What is the output of `timeout --version`? For now, you can try removing the `--preserve-status` argument from `inferNetwork.sh`.

HacKeD0x90, 2017-05-04:
so i upgraded my timeout version and thankfully it took IP as below, However after running run.sh, i still couldn't connect to it through network
sakr@vulMachine:~/IOT/firmadyne$     ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipsel
Running firmware 1: terminating after 60 secs...
qemu: terminating on signal 2 from pid 4156
Inferring network...
Interfaces: [('br0', '192.168.0.1')]
Done!

output of wget
sakr@vulMachine:~$ wget http://192.168.0.1
--2017-05-04 22:20:54--  http://192.168.0.1/
Connecting to 192.168.0.1:80...

HacKeD0x90, 2017-05-04:
Please ignore the previous problem solved Many Thanks :)
i was connecting to a wrong IP

HacKeD0x90, 2017-05-04:
I should be connecting to this right?
Interfaces: [('br0', '192.168.0.1')]
in this case it didn't work yet it's not connecting
i connected to my tun created on my machine and thought it worked sorry.
any hint on what maybe the issue
i got an interface created
tap1_0    Link encap:Ethernet  HWaddr 2e:2d:48:9d:b5:05  
          inet addr:192.168.0.2  Bcast:0.0.0.0  Mask:255.255.255.0
          inet6 addr: fe80::2c2d:48ff:fe9d:b505/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:184 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2213 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:41152 (41.1 KB)  TX bytes:136826 (136.8 KB)

HacKeD0x90, 2017-05-04:
here is the output of arpping
sakr@vulMachine:~$ arping -I tap1_0 192.168.0.1
WARNING: interface is ignored: Operation not permitted
ARPING 192.168.0.1 from 192.168.0.2 tap1_0
Unicast reply from 192.168.0.1 [00:DE:FA:19:C0:02]  1.715ms
Unicast reply from 192.168.0.1 [00:DE:FA:19:C0:02]  0.673ms
Unicast reply from 192.168.0.1 [00:DE:FA:19:C0:02]  0.713ms
Unicast reply from 192.168.0.1 [00:DE:FA:19:C0:02]  0.734ms
^CSent 4 probes (1 broadcast(s))
Received 4 response(s)

However still can't connect to the firmware which is Dlink
sakr@vulMachine:~$ wget 192.168.0.1
--2017-05-04 22:41:08--  http://192.168.0.1/
Connecting to 192.168.0.1:80... failed: Connection timed out.
Retrying.

ddcc, 2017-05-04:
It seems like the emulation is at least partially working, and showing up on the network. You might need to look through the emulation log or access a console on the emulated firmware to see if there's a web server running, what the firewall policies are, etc.

HacKeD0x90, 2017-05-04:
Can you please give me a helo on how to connect as console?

ddcc, 2017-05-04:
Are you using run.sh? It should be connected automatically.

HacKeD0x90, 2017-05-04:
Ah sorry
the logs were blocking the view, first time notice it. 
so i will troubleshoot it as it seems it's a network issue
many thanks

HacKeD0x90, 2017-05-04:
Can i disable these logs when running run.sh

ddcc, 2017-05-04:
If they're kernel logs, you can try changing the kernel boot parameters and loglevel. If they're from the firmware, you can't without modifying the firmware image.

kevins1022, 2018-04-20:
@ddcc Firmware not taking IP too.

```
[?] Enter the name or absolute path of the firmware you want to analyse : wnap320.zip
[?] Enter the brand of the firmware : a
[+] Now going to extract the firmware. Hold on..
[+] Firmware : wnap320.zip
[+] Brand : a
[+] Database image ID : 1
[+] Identifying architecture
[+] Architecture : mipseb
[+] Storing filesystem in database
[!] Filesystem already exists
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : []
[+] Running the firmware finally
```

and my timeout.
```
root@ubuntu:~/firmware-analysis-toolkit/firmadyne# timeout --version?
timeout: unrecognized option '--version?'
Try 'timeout --help' for more information.
root@ubuntu:~/firmware-analysis-toolkit/firmadyne# timeout --version
timeout (GNU coreutils) 8.21
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Padraig Brady.
root@ubuntu:~/firmware-analysis-toolkit/firmadyne# apt install coreutils
Reading package lists... Done
Building dependency tree
Reading state information... Done
coreutils is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 329 not upgraded.
```

ddcc, 2018-04-21:
You should create a new issue instead of posting to old issues. If it works after installing kpartx, which is required, I'm not sure what the problem is.

enamiirf, 2024-03-28:
[image]

enamiirf, 2024-03-28:
I don't know why there is no IP address

enamiirf, 2024-03-28:
$ sudo ./scripts/inferNetwork.sh 1
Querying database for architecture... Password for user firmadyne: 
mipseb
Running firmware 1: terminating after 60 secs...
qemu-system-mips: terminating on signal 2 from pid 14228 (timeout)
Inferring network...
Interfaces: []
Done!

gaochen61, 2025-05-28:
> I don't know why there is no IP address

also the problem, have you solved it?