firmadyne/firmadyne #48: netgear123456 login failure

ykddd, 2017-05-11:
After deleting the the file /etc/securetty,I still can't log in with root ang password.

Welcome to SDK.

Have a lot of fun...

netgear123456 login: root
Login incorrect

why?

ddcc, 2017-05-11:
Did you unmount the filesystem after mounting it? If you remount the filesystem, is the file `/etc/securetty` gone? Can you try deleting the QEMU disk and restarting from make image?

ykddd, 2017-05-12:
I make a mistake that forgeting to mount the filesystem ,I directly delete the /etc/securetty in my computer not the /etc/securetty in the filesystem.......I am so stupid....

closed 2017-05-12 by ddcc

richprice36, 2023-01-26:
So I have tryed everyting I can find to figure out how to login as root to the demo WNAP320.  I have it running and can login to the AP website etc..  When as for user I type root but the prompt never asks for a password.  Not sure what to do here..