firmadyne/firmadyne #50: arch='armel' can't emulate network successfully

MikimotoH, 2017-05-15:
I tested 4250 firmware images, including ASUS, D-Link, Netgear. I found in the database there is no any arch='armel' network emulation successfully. 
Can you give a successful example that arch='armel' ?

ddcc, 2017-05-15:
I don't have access to the database right now, but you can grab a copy of the old database from https://github.com/firmadyne/firmadyne#database-1 , and look for ARM images that were extracted successfully, though that doesn't show if the emulation and network detection worked correctly. Alternatively, if you look through the case studies in the paper, I recall at least a couple of them should be ARM.

MikimotoH, 2017-05-15:
Because I can't fix the Netgear R7000, which has kernel panic (not syncing) when QEMU is running. I don't know why don't you choose the same Linux version 2.6 as you do it for mipsel and mipseb, but choose a mismatched Linux version 4.1 for armel.

ddcc, 2017-05-15:
You can read the details behind that decision in #44

mikelhpdatke, 2017-07-06:
I have read the file infernetwork.py.
I tried to find some strings which find ifaces, addr, vlan_id,.v.v ("__inet_insert_ifa", "register_vlan_dev", "br_add_if", "br_dev_ioctl") in three kernels: vmlinux.mipsel, vmlinux.mipseb, zImage.armel

And I realize that the strings only appear in vmlinux.mipsel and vmlinux.mipsel. zImg.armel is not

So I cant understand why firmadyne could emulate firmware which has ARCH=armel?

ddcc, 2017-07-06:
They appear in all the ARM kernels: https://github.com/firmadyne/kernel-v4.1/blob/firmadyne-v4.1.17/drivers/firmadyne/hooks.c and https://github.com/firmadyne/kernel-v3.10/blob/firmadyne-v3.10.96/drivers/firmadyne/hooks.c

mikelhpdatke, 2017-07-06:
I tried to emulate somes firmwares ARM and when I view `qemu.initial.serial.log`, some strings such as  "register_vlan_dev", "br_add_if", "br_dev_ioctl,.. didn't appear in this file -> I guess this is the result why Firmadyne did not run inferNetwork.py successfully.

So I confuse that the reason is where come from the bug of kernel or some errors in the process of initial filesystem, or wrong parameter while setting up QEMU, or the process of initial in directory /etc/*,..?

dfbravo, 2017-07-06:
@mikelhpdatke I believe those strings will only show up in `qemu.initial.serial.log` if the system calls associated with those hooks are executed. This could happen due to many reasons. The firmware doesn't boot up properly is most likely the case. As to the reason why it fails to boot up is a mystery to me and changes case by case. You'd have to debug those errors yourself.

Please correct me if I am wrong.

ddcc, 2017-07-06:
Yeah, @dfbravo pretty much sums it up.