firmadyne/firmadyne #53: No IP for interface when emulating netgear r6250 firmware

j3rg, 2017-05-24:
The title says it all I don't get any IP address when inferring network for a netgear r6250 firmware (specific version of v1.0.1.84_1.0.78). Binwalk unpacks the firmware successfully the issue seems just be that the no ip for interface br0. I notice you also mention that we should report this issue because it might be a bug hence I'm doing so. If you need anymore information just let me know and I can assist.

Thanks,
j3rg

ddcc, 2017-05-24:
Thanks for the report. In general, we can't infer the network for all firmware, so fixing support for specific firmware is a manual process.

j3rg, 2017-05-24:
Oh ok understood I hope to contribute to this project soon. Currently learning python.

ving2, 2019-09-22:
@ddcc 
The problem occurs on many firmwares different vendors. What you need to do to fix the problem? Can you write this information in wiki (or readme)?

ddcc, 2019-09-22:
It's a known problem of the emulation approach, which would require additional developer effort to find and test additional effort. For more details, see the paper. But, as it is a research project, I am no longer actively developing this project.