firmadyne/firmadyne #55: Printing debug output from libnvram

prashast, 2017-06-12:
So I was using firmadyne to emulate a D-Link image, it emulates perfectly though I was interested in seeing the debug output emitted from libnvram. Looking at the code, it should print it if the DEBUG is defined which is the case in `config.h`. I even tried printing to stdout instead of stderr but still couldn't see any debug output being printed. I know the nvram functions are being triggered because I can see the default config file being copied over to the firmadyne library. Is there any other modification that I need to make in order to make the libnvram print it's debug output?

ddcc, 2017-06-12:
It's been a while since I've touched the code, but nothing comes to mind. The only thing is that printing of stderr to the default console could depend on the firmware image that you're emulating, but you say that you've tried with stdout instead. Just to make sure it's not a problem with the library, you could try building the test.c file and running that, which should trigger the printing out.

prashast, 2017-06-13:
Yeah, just checked, seems like it's an issue with the firmware image itself. Thanks again!

closed 2017-06-13 by prashast