firmadyne/firmadyne #56: AttributeError: 'module' object has no attribute 'scan'

luffynami27, 2017-06-24:
I'm getting this error whenever i try to emulate any firmware using firmadyne,can someone please help?

Enter the name or absolute path of the firmware you want to analyse : DVRF_v03.bin
Enter the brand of the firmware : Linksys
DVRF_v03.bin
Now going to extract the firmware. Hold on..
/root/Desktop/firmadyne/sources/extractor/extractor.py -b Linksys -sql 127.0.0.1 -np -nk "DVRF_v03.bin" images 
Traceback (most recent call last):
  File "/root/Desktop/firmadyne/sources/extractor/extractor.py", line 425, in extract
    if analysis():
  File "/root/Desktop/firmadyne/sources/extractor/extractor.py", line 472, in _check_archive
    return self._check_recursive("archive")
  File "/root/Desktop/firmadyne/sources/extractor/extractor.py", line 632, in _check_recursive
    for module in binwalk.scan(self.item, "-e", "-r", "-y", fmt,signature=True, quiet=True):
AttributeError: 'module' object has no attribute 'scan'
test
The database ID is 1
Getting image type
tar: ./images/1.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
Traceback (most recent call last):
  File "fat.py", line 113, in <module>
    main()
  File "fat.py", line 105, in main
    get_image_type(image_id)
  File "fat.py", line 58, in get_image_type
    output = subprocess.check_output(get_image_type_command, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 544, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '/root/Desktop/firmadyne/scripts/getArch.sh ./images/1.tar.gz' returned non-zero exit status 1

ddcc, 2017-06-24:
Did you install binwalk according to the directions? https://github.com/firmadyne/firmadyne#extractor

luffynami27, 2017-07-02:
Thanks,that fixed it 
this time i'm not able to get an ip, i'm getting the following error while running infernetwork.sh.Please help
-------------------------------------------------------------------------------------------------	
Enter the name or absolute path of the firmware you want to analyse : DVRF_v03.bin
Enter the brand of the firmware : Linksys
DVRF_v03.bin
Now going to extract the firmware. Hold on..
/root/Desktop/firmadyne/sources/extractor/extractor.py -b Linksys -sql 127.0.0.1 -np -nk "DVRF_v03.bin" images 

WARNING: Extractor.execute failed to run external extractor 'unsquashfs -d 'squashfs-root' '%e'': [Errno 2] No such file or directory, 'unsquashfs -d 'squashfs-root' '%e'' might not be installed correctly
test
The database ID is 1
Getting image type
Password for user firmadyne: 
Found image type of  mipsel
Putting information to database
Tar2DB
Creating Image
Executing command 

sudo /root/Desktop/firmadyne/scripts/makeImage.sh 1
Password for user firmadyne: 
Device contains neither a valid DOS partition table, nor Sun, SGI or OSF disklabel
Building a new DOS disklabel with disk identifier 0xb1543144.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
Building a new DOS disklabel with disk identifier 0xc151997d.
Changes will remain in memory only, until you decide to write them.
After that, of course, the previous content won't be recoverable.

Warning: invalid flag 0x0000 of partition table 4 will be corrected by w(rite)
mke2fs 1.42.5 (29-Jul-2012)
Please check the makeImage function
Everything is done for the image id 1
Setting up the network connection
Password for user firmadyne: 
qemu: terminating on signal 2 from pid 6970
Traceback (most recent call last):
  File "fat.py", line 113, in <module>
    main()
  File "fat.py", line 108, in main
    network_setup(image_id)
  File "fat.py", line 83, in network_setup
    output = subprocess.check_output(network_cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 544, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command 'sudo /root/Desktop/firmadyne/scripts/inferNetwork.sh 1' returned non-zero exit status 124
---------------------------------------------------------------------------
qemu.initial.serial.logs
-------------------------------------------------------------------------------------------------------------------
[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:44:57 UTC 2016
[    0.000000] 
[    0.000000] LINUX started...
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 00606000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f90a000 @ 006f6000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 57024 bytes for tracking 1782 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x00010000
[    0.000000] Movable zone start PFN for each node
[... 3910 lines cut ...]
nvram_set: device_view_type = "0"
sem_get: Unable to get semaphore key!
sem_lock: Unable to get semaphore!
sem_get: Unable to get semaphore key!
sem_unlock: Unable to get semaphore!
nvram_set: tunnel_status = "disconnected"
sem_get: Unable to get semaphore key!
sem_lock: Unable to get semaphore!
sem_get: Unable to get semaphore key!
sem_unlock: Unable to get semaphore!
nvram_set_default: Loading from native built-in table: router_defaults (0x2ac9fcb0) = 1!
nvram_set_default: Checking for symbol "Nvrams"...
nvram_set_default_image: Copying overrides from defaults folder!
sem_get: Unable to get semaphore key!
sem_lock: Unable to get semaphore!
cp: cannot stat '/firmadyne/libnvram.override/*': No such file or directory
sem_get: Unable to get semaphore key!
sem_unlock: Unable to get semaphore!
sem_get: Unable to get semaphore key!
sem_lock: Unable to get semaphore!
sem_get: Unable to get semaphore key!
sem_unlock: Unable to get semaphore!
nvram_get_buf: = "-08 1 1"
[    2.980000] Kernel panic - not syncing: Attempted to kill init!
---------------------------------------------------------------

ddcc, 2017-07-02:
Are you sure you installed binwalk and its dependencies? It still looks like you're missing some software that binwalk uses.