firmadyne/firmadyne #57: Problems when initiating libnvram with ARCH=mipsel

mikelhpdatke, 2017-07-04:
I tested many router's firmwares, including Dlink, Linksys, Netgear. Its OK.

But with Camera IP: `Tenvis` (get link and download in your database), It got errors with libnvram and init video. And I cant access the console command in firmware when run  `./scratch/1/run.sh`

> cp: cannot stat '/firmadyne/libnvram.override/*': No such file or directory
sem_get: Key: 412301e0
nvram_set: NULL key or value!
nvram_close: Closing NVRAM...
p_NetLinkRouteTop IS 1 LnkStu 1 
/dev/gpio: No such file or directory
/dev/gpio: No such file or directory
nvram_get_buf: mvdet_step
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/mvdet_step!
nvram_get_buf: vf_width
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/vf_width!
nvram_get_buf: vf_height
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/vf_height!
nvram_get_buf: vf_fps
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/vf_fps!
nvram_get_buf: vf_bright
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/vf_bright!
nvram_get_buf: vf_contrast
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/vf_contrast!
nvram_get_buf: vf_flip
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/vf_flip!
nvram_get_buf: ptz_cfg
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/ptz_cfg!
nvram_get_buf: ptz_speed
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/ptz_speed!
/dev/gpio: No such file or directory
nvram_get_buf: pwr_freq
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/pwr_freq!
nvram_get_buf: pwr_led
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/pwr_led!
nvram_get_buf: ir_led
sem_get: Key: 412301e0
sem_get: Key: 412301e0
nvram_get_buf: Unable to open key: /firmadyne/libnvram/ir_led!
/dev/gpio: No such file or directory
/dev/gpio: No such file or directory
ERROR opening V4L interface 
: No such file or directory
 Init Video Process failed !! exit fatal 
**** INIT_Video failed  *****
adstrm_init 
i2scmd:open i2s driver failed (-1)...exit
nvram_get_buf: NULL input key, buffer: tmpfs!
[    1.776000] do_page_fault() #2: sending SIGSEGV to goahead for invalid read access from
[    1.776000] 00000000 (epc == 2ab81740, ra == 0045c7ec)
[    1.776000] Cpu 0
[    1.776000] $ 0   : 00000000 1000a400 00000000 2abd81a0
[    1.776000] $ 4   : 00000000 00000000 0000000a 00000001
[    1.776000] $ 8   : 00000000 00000002 00000001 8fb86280
[    1.776000] $12   : 8fb86280 00000200 00000100 00000400
[    1.776000] $16   : 0053d02c 004e0000 00000000 ffffffff
[    1.776000] $20   : 2abd96b0 7fa90f44 00411ad8 00000001
[    1.776000] $24   : 00000008 2ab81700                  
[    1.780000] $28   : 2abe0510 7fa90cc0 0047b498 0045c7ec
[    1.780000] Hi    : 00000011
[    1.780000] Lo    : 0010a281
[    1.780000] epc   : 2ab81740 0x2ab81740
[    1.780000]     Not tainted
[    1.780000] ra    : 0045c7ec 0x45c7ec
[    1.780000] Status: 0000a413    USER EXL IE 
[    1.780000] Cause : 10800008
[    1.780000] BadVA : 00000000
[    1.780000] PrId  : 00019300 (MIPS 24Kc)
[    1.780000] Modules linked in:
[    1.780000] Process goahead (pid: 138, threadinfo=8fb38000, task=8f967258, tls=00000000)
[    1.780000] Stack : 00000000 ffffffff 2aac0000 004e0000 2abe0510 ffffffff 0053d02c 004e0000
[    1.780000]         00000000 ffffffff 2abd96b0 7fa90f44 00411ad8 00000001 0047b498 0045c7ec
[    1.780000]         00000002 0053d0c4 0053b1e0 00000000 0053b1e0 004e0000 00000000 ffffffff
[    1.780000]         ffffffff 0047b7dc 00000000 00000000 ffffffff 00000000 0000000c 00000017
[    1.780000]         0053b1e0 00000000 00000000 00000000 00000000 00000000 2aaae690 00000000
[    1.780000]         ...
[    1.780000] Call Trace:
[    1.780000] 
[    1.780000] 
[    1.780000] Code: afb00018  afbc0010  8f8387fc <80820000> 8c630000  00021040  00431021  94420000  00808821 
[    1.780000] goahead/138: potentially unexpected fatal signal 11.
[    1.784000] 
[    1.784000] Cpu 0
[    1.784000] $ 0   : 00000000 1000a400 00000000 2abd81a0
[    1.784000] $ 4   : 00000000 00000000 0000000a 00000001
[    1.784000] $ 8   : 00000000 00000002 00000001 8fb86280
[    1.784000] $12   : 8fb86280 00000200 00000100 00000400
[    1.784000] $16   : 0053d02c 004e0000 00000000 ffffffff
[    1.784000] $20   : 2abd96b0 7fa90f44 00411ad8 00000001
[    1.784000] $24   : 00000008 2ab81700                  
[    1.784000] $28   : 2abe0510 7fa90cc0 0047b498 0045c7ec
[    1.784000] Hi    : 00000011
[    1.784000] Lo    : 0010a281
[    1.784000] epc   : 2ab81740 0x2ab81740
[    1.784000]     Not tainted
[    1.784000] ra    : 0045c7ec 0x45c7ec
[    1.784000] Status: 0000a413    USER EXL IE 
[    1.784000] Cause : 10800008
[    1.784000] BadVA : 00000000
[    1.784000] PrId  : 00019300 (MIPS 24Kc)
[   11.468000] eth2: no IPv6 routers present
ls
abc123
top     
?
awpdl

ddcc, 2017-07-04:
Yeah, it looks like the web server is crashing with a NULL pointer dereference. You can try manually adding NVRAM entries for the values that it is looking for; I've updated the FAQ to address this in https://github.com/firmadyne/firmadyne/blob/master/README.md#a-process-crashed-eg-do_page_fault-2-sending-sigsegv-for-invalid-read-access-from-00000000 .

r00tb3, 2018-04-03:
@ddcc I'm facing the same problem for dlink firmware I didn't understand what to do exactly?

Should I clone the repo and build NVRAM again or what?

rk700, 2018-08-02:
@ddcc I came into the same issue, and it seems to be a mistake in `nvram_get`:
https://github.com/firmadyne/libnvram/pull/4/files

DerZc, 2019-01-15:
@ddcc  I came into the same issue, can you tell me how to figure out what NVRAM key-value pairs the firmware is expecting.

ddcc, 2019-01-16:
@DerZc The libnvram library prints out all of the unrecognized keys that are being requested by the firmware. To add an override, you can either modify the source code of the library and rebuild it, or mount the filesystem of the firmware, and place a file in /firmadyne/libnvram, with the key as the filename, and the value as the contents of the file.

To figure out what key/value pairs the firmware is expecting, you'll need to disassemble the program doing the NVRAM operations in the firmware, or look for some other configuration file in the firmware with a list of default or expected pairs.

jollytrivedi2019, 2023-07-05:
How can i create the file with key as filename and what should be the format and location of the same. It would be great if I can get some insight on this

megele, 2023-07-05:
Let's say your firmware asks for an NVRAM entry whose key is `foo` and the value is `bar`. Then, you have to create a file called foo (i.e., the key for the entry) in /firmadyne/libnvram and store the value bar inside that file.
`$ echo bar > /firmadyne/libnvram/foo`
should do the trick. Note that this must be stored in the image's file system not your local file system.

jollytrivedi2019, 2023-07-06:
Thank you! 
I created the entries in fixImage.sh file like $ echo -n "0" > /firmadyne/libnvram/buffalo_id.  The error "Unable to open key: /firmadyne/libnvram/buffalo_id" is resolved, But now getting new error - "Unable to get semaphore key". Am i missing something here?