firmadyne/firmadyne #59: Can't access the shell

mikelhpdatke, 2017-07-27:
When I emulated Dahua' firmwares, after running ` ./scratch/1/run.sh`, I can't access to shell in qemu
This is my short log:

> [mcat: can't open '/proc/dahua/selfchecked': No such file or directory
> /etc/init.d/S99dahua: line 55: telnetd: not found
> dvr_upnp_tv_ctrlpt_main 247 UPNP.PRO::CtrlPointStart
> dvr_upnp_tv_ctrlpt_main 253 UPNP.PRO::eth num:4
> pppd: applet not found
> ==============> 3gnet version: 1.00 - Complie time Apr  1 2017 15:19:42 SvnVersion = 4548<=============
> [mkernel version = 4.1.17
> [m[0;32;31m[LIBRES] ERROR  (/home/jenkins/jk_slave/workspace/NVR_DH3.X0839_NVR4x/tmp_build_dir/libdvr/src/libres/build/../src/resource_manager.c|ResIoctl|67): open /dev/dh_resource failed
> [m[0;32;31m[LIBDVR] ERROR  (/home/jenkins/jk_slave/workspace/NVR_DH3.X0839_NVR4x/tmp_build_dir/libdvr/src/libsysbase/build/../src/usb/usbinfo.c|init_usbif_lib|965): get usb info from resource failed
> [m[net3g]Fail to init libdvr!--netcard.cpp(2744)dvr_net3g_main 2017-07-27 09:22:27
> 
> udhcpd: /etc/udhcpd.conf: No such file or directory
> udhcpd: can't open /etc/udhcpd.conf
> 
> ifconfig: SIOCGIFFLAGS: No such device
> 
> 7z ANSI-C Decoder 9.20 : Igor Pavlov : Public domain : 2010-11-18
> 
> CtrlPointRefresh 467 UPNP.PRO::CtrlPointSearch
> Src/upnp/ssdp_ctrlpt.c:624, SendAdvertisement success
> dvr_upnp_tv_ctrlpt_main 260 UPNP.PRO::success CtrlPointStart
> [    6.379902] cfg80211: Calling CRDA to update world regulatory domain
> Extracting Challenge
> 
> Everything is Ok
> |       libres version:    15532 - Apr  1 2017 15:10:05
> [0;32;31m[LIBRES] ERROR  (/home/jenkins/jk_slave/workspace/NVR_DH3.X0839_NVR4x/tmp_build_dir/libdvr/src/libres/build/../src/resource_manager.c|ResIoctl|67): open /dev/dh_resource failed
> [m[0;32;31m[LIBRES] ERROR  (/home/jenkins/jk_slave/workspace/NVR_DH3.X0839_NVR4x/tmp_build_dir/libdvr/src/libres/build/../src/resource_manager.c|ResInit|361): resource init faild,exit the process
> [m[    9.530830] cfg80211: Calling CRDA to update world regulatory domain
> [   12.680165] cfg80211: Calling CRDA to update world regulatory domain
> [   15.830100] cfg80211: Calling CRDA to update world regulatory domain
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 195) exited. Scheduling for restart.
> [   18.980489] cfg80211: Calling CRDA to update world regulatory domain
> [   22.130613] cfg80211: Calling CRDA to update world regulatory domain
> [   25.280265] cfg80211: Calling CRDA to update world regulatory domain
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 201) exited. Scheduling for restart.
> [   28.430118] cfg80211: Calling CRDA to update world regulatory domain
> [   31.581145] cfg80211: Calling CRDA to update world regulatory domain
> [   34.730230] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
> dvr_upnp_tv_ctrlpt_main 278 UPNP router is not on line
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 207) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 212) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 215) exited. Scheduling for restart.
> dvr_upnp_tv_ctrlpt_main 278 UPNP router is not on line
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 218) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 221) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 224) exited. Scheduling for restart.
> dvr_upnp_tv_ctrlpt_main 278 UPNP router is not on line
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 227) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 230) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 233) exited. Scheduling for restart.
> dvr_upnp_tv_ctrlpt_main 278 UPNP router is not on line
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 236) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 239) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 242) exited. Scheduling for restart.
> dvr_upnp_tv_ctrlpt_main 278 UPNP router is not on line
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 245) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 248) exited. Scheduling for restart.
> 
> process '/sbin/getty -L ttyS0 115200 vt100' (pid 251) exited. Scheduling for restart.
> dvr_upnp_tv_ctrlpt_main 278 UPNP router is not on line

ddcc, 2017-07-27:
It looks like the process handling the default terminal keeps crashing for some reason. Unfortunately the cause of this tends to be firmware-specific, and probably involves manual debugging, e.g. https://github.com/firmadyne/firmadyne#how-do-i-debug-the-emulated-firmware

FrankSx, 2017-07-28:
i am currently working on a scripted debugger