firmadyne/firmadyne #61: Can't start web server

jinyu00, 2017-10-15:
when i use this firmware :
`
ftp://ftp2.dlink.com/PRODUCTS/DIR-890L/REVA/DIR-890L_REVA_FIRMWARE_1.03.B07.ZIP 
`
i could't connect to the qemu vm.

```
ubuntu:~/firmadyne$ ./scripts/inferNetwork.sh 3
Querying database for architecture... Password for user firmadyne: 
armel
Running firmware 3: terminating after 60 secs...
qemu-system-arm: terminating on signal 2 from pid 21817
Inferring network...
Interfaces: [('br0', '192.168.0.1'), ('br1', '192.168.7.1')]
Done!
```

when i start the vm, i can't connect to it

```
ubuntu:~$ ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
From 192.168.0.2 icmp_seq=1 Destination Host Unreachable
From 192.168.0.2 icmp_seq=2 Destination Host Unreachable
From 192.168.0.2 icmp_seq=3 Destination Host Unreachable
```

ddcc, 2017-10-15:
Did you try `br1`?

jinyu00, 2017-10-15:
yeah, when i ping br1, it can successfully exec,  but it also has not listen 80, or 22....

ddcc, 2017-10-15:
Have you inspected the network configuration of the guest in QEMU? Can you use the console to ping the host, check the firewall rules, and see if any daemons are listening?

ddcc, 2017-10-15:
Can you try to inspect the networking configuration of the guest inside QEMU? Using the console, you could try pinging the host, checking the firewall configuration, and seeing if any daemons are listening.