firmadyne/firmadyne #62: makeNetwork script error

HacKeD0x90, 2017-10-17:
I get this error when trying to run the script on a huawei firmware device
Either arch or endianness not found try mipsel/mipseb/armel/armeb

below is output of binwalk on the binary
67019 0x105CB Squashfs filesystem, little endian, version 4.0, compression:xz, size: 5979560 bytes, 1259 inodes, blocksize: 1048576 bytes, created: 2017-08-08 04:10:50
6047179 0x5C45CB uImage header, header size: 64 bytes, header CRC: 0xD3292F09, created: 2017-08-08 04:10:13, image size: 1438876 bytes, Data Address: 0x81208000, Entry Point: 0x81208000, data CRC: 0xFC9BD3F6, OS: Linux, CPU: ARM, image type: OS Kernel Image, compression type: lzma, image name: "Linux-2.6.30"
6047243 0x5C460B LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 3857792 bytes

ddcc, 2017-10-17:
It doesn't look like the firmware is assigning IP addresses to any interfaces (except 127.0.0.1 to lo), so it can't infer the networking configuration.

HacKeD0x90, 2017-10-17:
I'm sorry but i don't understand what is the issue?
can it be solved?
I'm doing a pentest on a firmware of a router, so i was planning to emulate it on firmadyne

ddcc, 2017-10-17:
The issue is that firmadyne is a research tool, and can't support emulating all possible firmware. This is one of those that appears to not be supported.

HacKeD0x90, 2017-10-17:
Do you know if there is a work around or another emulator for that?

ddcc, 2017-10-17:
There's no simple workaround, you'd need to manually debug the firmware to figure out why it can't be emulated and how to fix it. I don't know of a off-the-shelf emulator that just works for any embedded firmware.