firmadyne/firmadyne #63: ssh port is not open after emulating firmware using firmadyne

suruchi1701, 2017-10-25:
I used to emulate firmware using firmadyne, but now ssh port is not open after emulating the same firmware. Earlier it used to work for the same firmware. 

Firmware - kkeps.bin

ddcc, 2017-10-26:
I'm not sure why that would be the case. Have you tried restarting from makeImage?

suruchi1701, 2017-10-26:
Actually, i am using https://github.com/attify/firmware-analysis-toolkit which provides python program (fat.py) and it uses Firmadyne for emulating the firmware.

ddcc, 2017-10-26:
I'm not familiar with that project, perhaps you could contact them for support?

suruchi1701, 2017-10-27:
Yes, I am doing that. Meanwhile, i am trying just with Firmadyne also.

suruchi1701, 2017-10-27:
[image]

Still same with Firmadyne also. Earlier it used open ssh port for the same firmware. Now it is not.

ddcc, 2017-10-27:
Is the upper-right window the output of run.sh? It's possible that the SSH daemon is crashing after startup, have you logged through the kernel log?

suruchi1701, 2017-10-30:
Yes, It is run.sh output. 
Where I can check kernel log ?
Please find below complete log while running run.sh:
 
root@kali:~/Desktop/firmadyne# ./scratch/2/run.sh
Creating TAP device tap2_0...
Set 'tap2_0' persistent and owned by uid 0
Bringing up TAP device...
Adding route to 192.168.10.253...
Creating TAP device tap2_1...
Set 'tap2_1' persistent and owned by uid 0
Bringing up TAP device...
Adding route to 192.168.1.1...
Starting firmware emulation... use Ctrl-a + x to exit
[    0.000000] Linux version 2.6.32.70 (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:39:21 UTC 2016
[    0.000000] 
[    0.000000] LINUX started...
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU revision is: 00019300 (MIPS 24Kc)
[    0.000000] FPU revision is: 00739300
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00001000 @ 00000000 (reserved)
[    0.000000]  memory: 000ef000 @ 00001000 (ROM data)
[    0.000000]  memory: 0061e000 @ 000f0000 (reserved)
[    0.000000]  memory: 0f8f1000 @ 0070e000 (usable)
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Wasting 57792 bytes for tracking 1806 unused pages
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   DMA      0x00000000 -> 0x00001000
[    0.000000]   Normal   0x00001000 -> 0x0000ffff
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x0000ffff
[    0.000000] On node 0 totalpages: 65535
[    0.000000] free_area_init_node: node 0, pgdat 806aa3c0, node_mem_map 81000000
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 4064 pages, LIFO batch:0
[    0.000000]   Normal zone: 480 pages used for memmap
[    0.000000]   Normal zone: 60959 pages, LIFO batch:15
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65023
[    0.000000] Kernel command line: root=/dev/sda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Primary instruction cache 2kB, VIPT, 2-way, linesize 16 bytes.
[    0.000000] Primary data cache 2kB, 2-way, VIPT, no aliases, linesize 16 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 252428k/254916k available (4260k kernel code, 2252k reserved, 1549k data, 220k init, 0k highmem)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:256
[    0.000000] CPU frequency 200.00 MHz
[    0.000000] Console: colour dummy device 80x25
[    0.004000] Calibrating delay loop... 1130.49 BogoMIPS (lpj=2260992)
[    0.104000] Mount-cache hash table entries: 512
[    0.116000] NET: Registered protocol family 16
[    0.128000] bio: create slab <bio-0> at 0
[    0.132000] vgaarb: loaded
[    0.132000] SCSI subsystem initialized
[    0.136000] libata version 3.00 loaded.
[    0.136000] usbcore: registered new interface driver usbfs
[    0.140000] usbcore: registered new interface driver hub
[    0.140000] usbcore: registered new device driver usb
[    0.144000] pci 0000:00:00.0: reg 14 32bit mmio pref: [0x1000000-0x1ffffff]
[    0.144000] pci 0000:00:0a.1: reg 20 io port: [0x00-0x0f]
[    0.148000] pci 0000:00:0a.2: reg 20 io port: [0x00-0x1f]
[    0.148000] pci 0000:00:0a.3: BAR 8: address space collision on of bridge [0x1100-0x110f]
[    0.152000] pci 0000:00:0a.3: quirk: region 1100-110f claimed by PIIX4 SMB
[... 194 lines cut ...]
[    7.664000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32849
[    7.664000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32850
[    8.348000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32850
[    8.360000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32849
[    8.372000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32850
[    8.376000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32849
[    9.428000] eth0: link up
[    9.440000] device eth0 entered promiscuous mode
[    9.452000] br-lan: port 1(eth0) entering learning state
[    9.496000] eth1: link up
[   10.296000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32848
[   10.312000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32848
[   10.640000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32850
[   10.648000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32849
[   10.648000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32850
[   10.656000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32849
[   10.688000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32849
[   10.688000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32850
[   10.872000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32850
[   10.880000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32849
[   11.452000] br-lan: port 1(eth0) entering forwarding state
[   13.900000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32848
[   14.644000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32848
[   14.672000] EXT2-fs error (device sda1): ext2_lookup: deleted inode referenced: 32848
procd: - init complete -

FrankSx, 2017-10-30:
> [ 0.128000] bio: create slab at 0

Did you give my version of fat a try:
https://github.com/FrankSx/firmware-analysis-toolkit , if you need further
help email me at : fixes.it.frank@gmail.com

ddcc, 2017-10-30:
The output from the run.sh script includes the kernel log. It looks like there's something wrong with the root filesystem. Have you retried from the makeImage stage? Also, you could try with a different Linux distribution, other people have reported problems with Kali in the past.

FrankSx, 2017-10-30:
I agree, the filesystem maybe the problem as I advised in the email change
the filesystem in the makeimage script and also try attifyOS in a virtual
machine