firmadyne/firmadyne #65: Where to find the results of vulnerabilities?

yuanchang200, 2017-11-21:
Hi, after run these commands:

./analyses/snmpwalk.sh 192.168.0.100
./analyses/webAccess.py 1 192.168.0.100 log.txt
mkdir exploits; ./analyses/runExploits.py -t 192.168.0.100 -o exploits/exploit -e x (requires Metasploit Framework)
sudo nmap -O -sV 192.168.0.100

Where can I see the vulnerabilities that firmadyne find about this firmware?

ddcc, 2017-11-21:
Most of the scripts print where the output is written to, you can look at the help for command-line options, or skim the source code. We had some scripts to load the results into a SQL database.

yuanchang200, 2017-11-21:
Can you tell me which command is the one that prints the results of vulnerabilities?

ddcc, 2017-11-21:
As I said, this is not a "point and click" tool to find/exploit vulnerabilities, but a platform for running dynamic analyses to find vulnerabilities.

`snmpwalk.sh` explicitly states that the results of the SNMP enumeration are dumped to `snmp.public.txt` and `snmp.private.txt`. These need to be manually interpreted to determine which OIDs are sensitive. 

`webAccess.py` queries the list of possible webpages on the filesystem for a given firmware from the SQL server, attempts to access each using the specified IP address, and logs the accessible ones to disk. Again, the log needs to be manually interpreted to determine which files are sensitive.

`nmap.sh` runs a basic TCP port scan for a firmware and dumps the results out to disk. These are listening services, which may have potential vulnerabilities, that should be manually reviewed.

`runExploits.sh` runs the specified exploit against a firmware, and logs each result to disk. If you look at the source code, it explicitly shows what CVE numbers or Metasploit exploits correspond to each number.