firmadyne/firmadyne #67: Cannot make filesystem

zhuanpan, 2017-12-06:
When inputting sudo ./scripts/makeImage.sh, the output is /dev/mapper/loop0p1 is mounted; will not make a filesystem here!. I have do it again using your FAQ,but it has no effect. Looking forward to your answer

ddcc, 2017-12-07:
That's strange. Are you using a USB boot or live image? People have mentioned an incompatibility in the past with Kali.

wangray, 2017-12-16:
Hi,

I'm having issues along these lines as well. I am using an ESX vm. I have tried changing `get_device` to `/dev/mapper/loop1p1`, `/dev/mapper/loop2p1`, but in all cases I get `Unable to mount root fs on unknown-block(8,1)`. 

Here's the contents of my `/dev/mapper` directory: 
```
control  laredo--66--vg-root  laredo--66--vg-swap_1  loop10p1  loop11p1  loop2p1  loop8p1  loop9p1
```
and the output of `makeImage.sh`: 
```
Querying database for architecture... Password for user firmadyne:
mipseb
----Running----
----Copying Filesystem Tarball----
----Creating QEMU Image----
Formatting '/home/raywang/firmadyne/scratch//1//image.raw', fmt=raw size=1073741824
----Creating Partition Table----

Welcome to fdisk (util-linux 2.27.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xbfe000f6.

Command (m for help): Created a new DOS disklabel with disk identifier 0x8badcb8f.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-2097151, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-2097151, default 2097151):
Created a new partition 1 of type 'Linux' and of size 1023 MiB.

Command (m for help): The partition table has been altered.
Syncing disks.

----Mounting QEMU Image----
add map loop13p1 (252:8): 0 2095104 linear 7:13 2048
----Creating Filesystem----
mke2fs 1.42.13 (17-May-2015)
/dev/mapper/loop10p1 contains a ext2 file system
	last mounted on /home/raywang/firmadyne/scratch/1/image on Sat Dec 16 11:25:44 2017
Proceed anyway? (y,n) y
Discarding device blocks: done
Creating filesystem with 261888 4k blocks and 65536 inodes
Filesystem UUID: 64d557f6-00f6-486c-b05e-4c02f9254cb0
Superblock backups stored on blocks:
	32768, 98304, 163840, 229376

Allocating group tables: done
Writing inode tables: done
Writing superblocks and filesystem accounting information: done

----Making QEMU Image Mountpoint----
----Mounting QEMU Image Partition 1----
----Extracting Filesystem Tarball----
----Creating FIRMADYNE Directories----
----Patching Filesystem (chroot)----
Warning: Recreating device nodes!
Removing /etc/scripts/sys_resetbutton!
----Setting up FIRMADYNE----
----Unmounting QEMU Image----
loop deleted : /dev/loop13
```

ddcc, 2017-12-16:
What's the output of `sudo dmsetup ls` and `losetup -a`? Are you running Kali or a live boot image in the VM?

V1ncent7, 2018-04-25:
I‘ve met the same question with U.
And there is not any interface when input sudo  ./scripts/interNetwork.sh

Querying database for architecture... Password for user firmadyne: 
mipsel
Running firmware 5: terminating after 60 secs...
main-loop: WARNING: I/O thread spun for 1000 iterations
qemu-system-mipsel: terminating on signal 2 from pid 78553
Inferring network...
Interfaces: []
Done!

I've checked the qemu.initial.serial.log and there isnt any '__inet_insert_ifa' as well.

ddcc, 2018-04-25:
You should create a separate issue. Have you checked https://github.com/firmadyne/firmadyne#runsh-is-not-generated ?

V1ncent7, 2018-04-27:
thank you,i've solved this problem.
But a new error happened.When i run ./run.sh, there is a Kernel panic --not syncing: No working init found.  Try passing init= option to kernel.
And i've found the file in bin and sbin are just text/plain(unexecutable)rather than the symbloic link to busybox just like the example given by github. So i've just change them to symbolic link munaully.

Below is the output of run.sh

Starting firmware emulation... use Ctrl-a + x to exit
tee: /home/vince/firmadyne//scratch//9//qemu.final.serial.log: Permission denied
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.17+ (vagrant@vagrant-ubuntu-trusty-64) (gcc version 5.3.0 (GCC) ) #1 Thu Feb 18 01:05:21 UTC 2016
[    0.000000] CPU: ARMv7 Processor [412fc0f1] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine model: linux,dummy-virt
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c061dfe8, node_mem_map cfdf9000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.2 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: root=/dev/vda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 253340K/262144K available (4297K kernel code, 170K rwdata, 1584K rodata, 180K init, 148K bss, 8804K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc05c67bc   (5882 kB)
[    0.000000]       .init : 0xc05c7000 - 0xc05f4000   ( 180 kB)
[    0.000000]       .data : 0xc05f4000 - 0xc061e840   ( 171 kB)
[    0.000000]        .bss : 0xc0621000 - 0xc06462d4   ( 149 kB)
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 62.50MHz (virt).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x1cd42e208c, max_idle_ns: 881590405314 ns
[    0.000084] sched_clock: 56 bits at 62MHz, resolution 16ns, wraps every 4398046511096ns
[    0.000217] Switching to timer-based delay loop, resolution 16ns
[    0.002050] Console: colour dummy device 80x30
[    0.002319] Calibrating delay loop (skipped), value calculated using timer frequency.. 125.00 BogoMIPS (lpj=625000)
[    0.002389] pid_max: default: 32768 minimum: 301
[    0.002968] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002991] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.007405] CPU: Testing write buffer coherency: ok
[    0.012607] Setting up static identity map for 0x40008240 - 0x40008298
[    0.023245] VFP support v0.3: implementor 41 architecture 4 part 30 variant f rev 0
[    0.029200] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.036586] NET: Registered protocol family 16
[    0.038472] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.041189] cpuidle: using governor ladder
[    0.041334] cpuidle: using governor menu
[    0.055025] Serial: AMBA PL011 UART driver
[    0.056403] 9000000.pl011: ttyS0 at MMIO 0x9000000 (irq = 53, base_baud = 0) is a PL011 rev1
[    0.082732] console [ttyS0] enabled
[    0.105413] vgaarb: loaded
[    0.106815] SCSI subsystem initialized
[    0.107883] usbcore: registered new interface driver usbfs
[    0.108906] usbcore: registered new interface driver hub
[    0.109830] usbcore: registered new device driver usb
[    0.117006] cfg80211: Calling CRDA to update world regulatory domain
[    0.119848] Switched to clocksource arch_sys_counter
[... 143 lines cut ...]
[    0.524399] Initializing XFRM netlink socket
[    0.524723] NET: Registered protocol family 10
[    0.529220] ip6_tables: (C) 2000-2006 Netfilter Core Team
[    0.530286] sit: IPv6 over IPv4 tunneling driver
[    0.533480] NET: Registered protocol family 17
[    0.534046] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    0.534573] Bridge firewalling registered
[    0.534766] Ebtables v2.0 registered
[    0.536050] 8021q: 802.1Q VLAN Support v1.8
[    0.536734] Registering SWP/SWPB emulation handler
[... cut ...]
[    0.591872] Hardware name: Generic DT based system
[    0.592039] [<c001c99c>] (unwind_backtrace) from [<c0019d30>] (show_stack+0x10/0x14)
[    0.592165] [<c0019d30>] (show_stack) from [<c0410b78>] (panic+0x70/0x1c8)
[    0.592242] [<c0410b78>] (panic) from [<c040f33c>] (kernel_init+0xb8/0xe4)
[    0.592401] [<c040f33c>] (kernel_init) from [<c0016da8>] (ret_from_fork+0x14/0x2c)
[    0.592626] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.

ddcc, 2018-04-29:
Somehow, the init binaries is missing or not executable. You'll need to check /sbin/init, and if it doesn't exist, find the proper name of the init binary for this firmware and move it there.