firmadyne/firmadyne #71: gdbserver is not working on mipseb kernel (WNAP320 Firmware)

jishoon, 2018-02-09:
Hi, I'm trying to debug remote process using statically compiled gdbserver.

To do this, I've just followed `Usage` in README.md. 
Everything is fine. I can emulate full-system and network is working properly. 

In order to compile `gdbserver`, I 've downloaded precompiled musl-toolchain which provided by this repository(through box.com). 
I compiled gdbserver and the output is following.

```bash
$ file ./gdbserver
./gdbserver: ELF 32-bit MSB executable, MIPS, MIPS-I version 1 (SYSV), statically linked, with debug_info, not stripped
```
I copied `gdbserver` into WNAP320's filesystem.  And, finally, I could run gdbserver on emulated firmware.

```bash
[root@netgear123456 /bin]# ./gdbserver --version
GNU gdbserver (GDB) 7.8
Copyright (C) 2014 Free Software Foundation, Inc.
gdbserver is free software, covered by the GNU General Public License.
This gdbserver was configured as "mipseb-linux-musl"
```

When I execute `gdbserver` with program, It is going to stuck like this.

```bash
[root@netgear123456 /bin]# ./gdbserver --debug :31337 ./busybox 
new_argv[0] = "./busybox"
Process ./busybox created; pid = 632
sigchld_handler
>>>> entering linux_wait_1
linux_wait_1: [Process 632]
my_waitpid (-1, 0x40000001)
my_waitpid (-1, 0x1): status(57f), 632
LWFE: waitpid(-1, ...) returned 632, ERRNO-OK
LLW: waitpid 632 received Trace/breakpoint trap (stopped)
linux_low_filter_event: pc is 0x2aaa8a20
pc is 0x2aaa8a20
stop pc is 0x2aaa8a20
my_waitpid (633, 0x0)
my_waitpid (633, 0x0): status(177f), 633
my_waitpid (633, 0x0)
```

I can establish TCP session between GDB and gdbserver. But there are no data from gdbserver. 

```bash
gef➤  gef-remote '192.168.0.100:31337'
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
[!] Failed to connect to 192.168.0.100:31337: Remote replied unexpectedly to 'vMustReplyEmpty': timeout
```

I repeat this experiment using gdbserver 7.12 and 8.01. The result is same.

## Result on debian-mipseb

I performed this experiment on debian-mipseb. It works very well.

```bash
user@mips-deb6:~$ uname -a
Linux mips-deb6 2.6.32-5-4kc-malta #1 Sat Dec 11 19:41:21 UTC 2010 mips GNU/Linux
user@mips-deb6:~$ ./gdbserver --debug :31337 /bin/ls
new_argv[0] = "/bin/ls"
Process /bin/ls created; pid = 884
>>>> entering linux_wait_1
linux_wait_1: [Process 884]
my_waitpid (-1, 0x40000001)
my_waitpid (-1, 0x80000001): status(0), -1
LWFE: waitpid(-1, ...) returned -1, No child process
leader_pid=884, leader_lp!=NULL=1, num_lwps=1, zombie=0
sigsuspend'ing
sigchld_handler
my_waitpid (-1, 0x40000001)
my_waitpid (-1, 0x1): status(57f), 884
LWFE: waitpid(-1, ...) returned 884, ERRNO-OK
LLW: waitpid 884 received Trace/breakpoint trap (stopped)
linux_low_filter_event: pc is 0x2aaa8880
pc is 0x2aaa8880
stop pc is 0x2aaa8880
my_waitpid (885, 0x0)
my_waitpid (885, 0x0): status(177f), 885
my_waitpid (885, 0x0)
my_waitpid (885, 0x0): status(1057f), 885
my_waitpid (886, 0x0)
my_waitpid (886, 0x0): status(177f), 886
my_waitpid (886, 0x0)
my_waitpid (886, 0x0): status(9), 886
my_waitpid (885, 0x0)
my_waitpid (885, 0x0): status(0), 885
sigchld_handler
Hit a non-gdbserver trap event.
>>>> entering stop_all_lwps
stop_all_lwps (stop, except=none)
wait_for_sigstop: pulling events
my_waitpid (-1, 0x40000001)
my_waitpid (-1, 0x80000001): status(0), -1
LWFE: waitpid(-1, ...) returned -1, No child process
leader_pid=884, leader_lp!=NULL=1, num_lwps=1, zombie=0
LLW: exit (no unwaited-for LWP)
stop_all_lwps done, setting stopping_threads back to !stopping
<<<< exiting stop_all_lwps
Checking whether LWP 884 needs to move out of the jump pad...no
linux_wait_1 ret = LWP 884.884, 1, 5
<<<< exiting linux_wait_1
Listening on port 31337
```

I've attached the [gdbserver](http://150.95.132.43/gdbserver) which used for experiment.

ddcc, 2018-02-09:
I recall running into some issues with `gdbserver` network connectivity as well, but it's been long enough that I don't remember how to fix it. You might to take a look at #46, someone else ran into a similar problem and was able to fix it by changing kernels.

patrickrbc, 2018-09-06:
I am having the same problem as @jishoon, but with a **Linux debian-mipsel 2.6.32-5-4kc-malta**. I was able to make gdbserver work using kernel 3.2, but then everything else breaks.

@jishoon, did you find any gdbserver static build that works in this scenario?

cq674350529, 2018-10-31:
I come across the similar problem. I have tried some old gdbserver of different versions (6.5, 6.6, 6.7 ,6.8, 7.0). None could work well. Do you make it work? @jishoon  @patrickrbc 
Thanks in advance.

ddcc, 2019-01-04:
Ah, that's unfortunate, and firmadyne is not actively developed either.

As a workaround, you could modify the QEMU command-line to forward an additional serial port between the guest/host, and bind the guest gdbserver to that serial port, instead of using TCP/IP.

sssyyynnngithub, 2019-10-24:
@h0rac , First thank you very much for the tips. Could you please me some more detailed steps ? For example , how can I bind rsync rootfs to some directory, bind proc/sys/dev, create networking as necessary and manually start rcS ? Thank you very much!

sssyyynnngithub, 2019-10-24:
@ddcc @jishoon @patrickrbc @cq674350529 Did everybody solve the problem ? I changed a new kernel , but started the network failed .  Could you give me some more detailed problem ? Thank you very much !

cq674350529, 2019-10-25:
@sssyyynnngithub Oops, finally I resorted to use a real device for debugging purpose. It was a hard work for me to make the gdbserver work on some emulated devices.
By the way, maybe you can find a similar firmware which can be emulated well. The most important part is the gdbserver works on that emulated device. Then you can "transfer" part of your target firmware to that similar firmware. Just an advice.
Of course, a real device is always the better choice.

sssyyynnngithub, 2019-11-23:
@cq674350529 Thank you very much!

sssyyynnngithub, 2019-11-23:
@h0rac Sir, you use Malta 3.2 kernel for mips to solve the problem ?  You said that firmadyne is not included in this process. So is it a pure linux 3.2 kernel without patch of firmadyne?Could you please show me the exaction instructions to build thsi Malta3.2 kernel ?Thank you very very much !

ddcc, 2020-09-03:
Fixed by d1d1ffb94ead98eb471dfdf4511b514c08a3099a

closed 2020-09-03 by ddcc