firmadyne/firmadyne #72: getArch return ''

xiaohuajiao, 2018-03-09:
In scripts/getArch.sh
```bash
FILES="$(tar -tf $INFILE | grep -e "/busybox\$") "
FILES+="$(tar -tf $INFILE | grep -E "/sbin/[[:alpha:]]+")"
```
1. When this script try to get file list from IID.tar.gz, the grep may return 1, then this script will stop immediately. To avoid this problem, we can add `set +e` first, then disable it again? like this.

```bash
set +e
FILES="$(tar -tf $INFILE | grep -e "/busybox\$") "
FILES+="$(tar -tf $INFILE | grep -E "/sbin/[[:alpha:]]+")"
set -e
```

2. I met a firmware that it didn't contain any file met above path pattern. Can you add more directories for searching? like `/bin/*`

ddcc, 2018-03-12:
Sure, can you create a pull request for those changes?

xiaohuajiao, 2018-03-15:
created

closed 2018-03-16 by xiaohuajiao