firmadyne/firmadyne #74: How to find the tarballs of extracted filesystems firmware images?

swtkiwi, 2018-03-18:
Hello,I want to use you firmware images,but I cannot find it,can you help me?

ddcc, 2018-03-18:
We don't redistribute any firmware images. You can either run the [scraper](https://github.com/firmadyne/scraper) to gather your own dataset, or grab the URLs from our [database](https://cmu.box.com/s/hnpvf1n72uccnhyfe307rc2nb9rfxmjp) and download them yourself (note that many URLs may be out of date).

AbhishekBhatP, 2019-03-14:
when i run this command sudo ./scripts/makeImage.sh I am getting 
----Running----
----Copying Filesystem Tarball----"
Error: Cannot find tarball of root filesystem for ${IID}!
Plz give the solution

cq674350529, 2019-03-14:
@AbhishekBhatP I can run these scripts normally. The `./scripts/makeImage.sh` will find the image tarballs in the `images` directory. Maybe you should check if there are any image tarballs. One possible reason is your firmware can't be extracted by `binwalk`.

AbhishekBhatP, 2019-03-14:
@cq674350529 I have only 1.tar.gz file in images.

cq674350529, 2019-03-14:
@AbhishekBhatP I'm wondering why the error msg contains `${IID}`? Maybe you can use `echo` statements to output the `"${TARBALL_DIR}/${IID}.tar.gz"`, to see if the output is `<images path>/1.tar.gz`.
https://github.com/firmadyne/firmadyne/blob/f212ed1c09f89049fdb6c945a5c2617efe86aa50/scripts/makeImage.sh#L57-L64

AbhishekBhatP, 2019-03-14:
#!/bin/bash

set -e
set -u

if [ -e ./firmadyne.config ]; then
    source ./firmadyne.config
elif [ -e ../firmadyne.config ]; then
    source ../firmadyne.config
else
    echo "Error: Could not find 'firmadyne.config'!"
    exit 1
fi

if check_number ${1}; then
    echo "Usage: makeImage.sh <image ID> [<architecture]"
    exit 1
fi
IID=${1}

if check_root; then
    echo "Error: This script requires root privileges!"
    exit 1
fi

if [ $# -gt 1 ]; then
    if check_arch "${2}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    ARCH=${2}
else
    echo -n "Querying database for architecture... "
    ARCH=$(psql -d firmware -U firmadyne -h 127.0.0.1 -t -q -c "SELECT arch from image WHERE id=${1};")
    ARCH="${ARCH#"${ARCH%%[![:space:]]*}"}"
    echo "${ARCH}"
    if [ -z "${ARCH}" ]; then
        echo "Error: Unable to lookup architecture. Please specify {armel,mipseb,mipsel} as the second argument!"
        exit 1
    fi
fi

echo "----Running----"
WORK_DIR=`get_scratch ${IID}`
IMAGE=`get_fs ${IID}`
IMAGE_DIR=`get_fs_mount ${IID}`
CONSOLE=`get_console ${ARCH}`
LIBNVRAM=`get_nvram ${ARCH}`
TARBALL_DIR='${FIRMWARE_DIR}/images'

echo "----Copying Filesystem Tarball----"
mkdir -p "${WORK_DIR}"
chmod a+rwx "${WORK_DIR}"
chown -R "${USER}" "${WORK_DIR}"
chgrp -R "${USER}" "${WORK_DIR}"

if [ ! -e "${WORK_DIR}/${IID}.tar.gz" ]; then
echo "found"   
 if [ ! -e "${TARBALL_DIR}/1.tar.gz" ]; then
        echo "Error: Cannot find tarball of root filesystem for ${IID}!"
        exit 1
    else
        cp "${TARBALL_DIR}/${IID}.tar.gz" "${WORK_DIR}/${IID}.tar.gz"
    fi
fi

echo "----Creating QEMU Image----"
qemu-img create -f raw "${IMAGE}" 1G
chmod a+rw "${IMAGE}"

echo "----Creating Partition Table----"
echo -e "o\nn\np\n1\n\n\nw" | /sbin/fdisk "${IMAGE}"

echo "----Mounting QEMU Image----"
DEVICE=$(get_device "$(kpartx -a -s -v "${IMAGE}")")
sleep 1

echo "----Creating Filesystem----"
mkfs.ext2 "${DEVICE}"
sync

echo "----Making QEMU Image Mountpoint----"
if [ ! -e "${IMAGE_DIR}" ]; then
    mkdir "${IMAGE_DIR}"
    chown "${USER}" "${IMAGE_DIR}"
fi

echo "----Mounting QEMU Image Partition 1----"
mount "${DEVICE}" "${IMAGE_DIR}"

echo "----Extracting Filesystem Tarball----"
tar -xf "${WORK_DIR}/$IID.tar.gz" -C "${IMAGE_DIR}"
rm "${WORK_DIR}/${IID}.tar.gz"

echo "----Creating FIRMADYNE Directories----"
mkdir "${IMAGE_DIR}/firmadyne/"
mkdir "${IMAGE_DIR}/firmadyne/libnvram/"
mkdir "${IMAGE_DIR}/firmadyne/libnvram.override/"

echo "----Patching Filesystem (chroot)----"
cp $(which busybox) "${IMAGE_DIR}"
cp "${SCRIPT_DIR}/fixImage.sh" "${IMAGE_DIR}"
chroot "${IMAGE_DIR}" /busybox ash /fixImage.sh
rm "${IMAGE_DIR}/fixImage.sh"
rm "${IMAGE_DIR}/busybox"

echo "----Setting up FIRMADYNE----"
cp "${CONSOLE}" "${IMAGE_DIR}/firmadyne/console"
chmod a+x "${IMAGE_DIR}/firmadyne/console"
mknod -m 666 "${IMAGE_DIR}/firmadyne/ttyS1" c 4 65

cp "${LIBNVRAM}" "${IMAGE_DIR}/firmadyne/libnvram.so"
chmod a+x "${IMAGE_DIR}/firmadyne/libnvram.so"

cp "${SCRIPT_DIR}/preInit.sh" "${IMAGE_DIR}/firmadyne/preInit.sh"
chmod a+x "${IMAGE_DIR}/firmadyne/preInit.sh"

echo "----Unmounting QEMU Image----"
sync
umount "${DEVICE}"
kpartx -d "${IMAGE}"
losetup -d "${DEVICE}" &>/dev/null
dmsetup remove $(basename "$DEVICE") &>/dev/null

@cq674350529 This is my makeImage.sh file.

cq674350529, 2019-03-14:
@AbhishekBhatP Add an line like this in the `makeImage.sh`,  rerun it and check the output.
```shell

 echo "${TARBALL_DIR}/${IID}.tar.gz"

 if [ ! -e "${WORK_DIR}/${IID}.tar.gz" ]; then
     if [ ! -e "${TARBALL_DIR}/${IID}.tar.gz" ]; then 
         echo "Error: Cannot find tarball of root filesystem for ${IID}!" 
         exit 1 
     else 
         cp "${TARBALL_DIR}/${IID}.tar.gz" "${WORK_DIR}/${IID}.tar.gz" 
     fi 
 fi 
```

AbhishekBhatP, 2019-03-14:
I ran the code as you have suggested. But still it is giving the same error.

 root@abhishek-VirtualBox:~/firmadyne#     sudo ./scripts/makeImage.sh 1
Querying database for architecture... Password for user firmadyne: 
mipseb
----Running----
----Copying Filesystem Tarball----
${FIRMWARE_DIR}/images/1.tar.gz
Error: Cannot find tarball of root filesystem for 1!

cq674350529, 2019-03-14:
It seems you did not configure the global env FIRMWARE_DIR. You should config them first in the firmadyne.config file.

AbhishekBhatP, 2019-03-14:
#!/bin/sh

# uncomment and specify full path to FIRMADYNE repository
FIRMWARE_DIR=/home/firmadyne

# specify full paths to other directories
BINARY_DIR=${FIRMWARE_DIR}/binaries/
TARBALL_DIR=${FIRMWARE_DIR}/images/
SCRATCH_DIR=${FIRMWARE_DIR}/scratch/
SCRIPT_DIR=${FIRMWARE_DIR}/scripts/

# functions to safely compute other paths

check_arch () {
    ARCHS=("armel" "mipseb" "mipsel")

    if [ -z "${1}" ]; then
        return 0
    fi

    match=0
    for i in "${ARCHS[@]}"; do
        if [ "${1}" == "$i" ]; then
            match=1
        fi
    done

    if [ "${match}" -eq 0 ]; then
        return 0
    fi

    return 1
}

check_number () {
    if [ "${1}" -ge 0 ]; then
        return 1
    fi

    return 0
}

check_root () {
    if [ "${EUID}" -eq 0 ]; then
        return 1
    fi

    return 0
}

get_console () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    echo "${BINARY_DIR}/console.${1}"
}

get_fs () {
    if check_number "${1}"; then
        echo "Error: Invalid image number!"
        exit 1
    fi

    echo "`get_scratch "${1}"`/image.raw"
}

get_fs_mount () {
    if check_number "${1}"; then
        echo "Error: Invalid image number!"
        exit 1
    fi

    echo "`get_scratch "${1}"`/image/"
}

get_kernel () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "${BINARY_DIR}/zImage.${1}"
            ;;
        mipseb)
            echo "${BINARY_DIR}/vmlinux.${1}"
            ;;
        mipsel)
            echo "${BINARY_DIR}/vmlinux.${1}"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_nvram () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    echo "${BINARY_DIR}/libnvram.so.${1}"
}

get_qemu () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "qemu-system-arm"
            ;;
        mipseb)
            echo "qemu-system-mips"
            ;;
        mipsel)
            echo "qemu-system-mipsel"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_qemu_disk () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "/dev/vda1"
            ;;
        mipseb)
            echo "/dev/sda1"
            ;;
        mipsel)
            echo "/dev/sda1"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_qemu_machine () {
    if check_arch "${1}"; then
        echo "Error: Invalid architecture!"
        exit 1
    fi

    case "${1}" in
        armel)
            echo "virt"
            ;;
        mipseb)
            echo "malta"
            ;;
        mipsel)
            echo "malta"
            ;;
        *)
            echo "Error: Invalid architecture!"
            exit 1
    esac
}

get_scratch () {
    if check_number "${1}"; then
        echo "Error: Invalid image number!"
        exit 1
    fi

    echo "${SCRATCH_DIR}/${IID}/"
}

get_device () {
    # Parses output from kpartx
    echo "/dev/mapper/$(echo $1 | cut -d ' ' -f 3)"
}

@cq674350529 This is my firmadyne.config.sh file

cq674350529, 2019-03-14:
@AbhishekBhatP The output `${FIRMWARE_DIR}/images/1.tar.gz` is wierd. In my machine, I can get the right path of 1.tar.gz, such as `home/firmadyne/images/1.tar.gz`.  
In addition, the first error message is `Error: Cannot find tarball of root filesystem for ${IID}!`, but the second is `Error: Cannot find tarball of root filesystem for 1!`. 
What's your environment? Which shell do you use?

AbhishekBhatP, 2019-03-15:
I am using bash

cq674350529, 2019-03-15:
@AbhishekBhatP According the error msg you provided, the var `${FIRMWARE_DIR}` is not interpreted correctly by the shell. You should figure it out. If interpreted correctly, the `${FIRMWARE_DIR}` will be replaced with the path you configed, like "/home/firmadyne".

> mipseb
----Running----
----Copying Filesystem Tarball----
${FIRMWARE_DIR}/images/1.tar.gz
Error: Cannot find tarball of root filesystem for 1!

AbhishekBhatP, 2019-03-15:
ok. thank you so much bro

Ainevsia, 2021-03-18:
> @AbhishekBhatP According the error msg you provided, the var `${FIRMWARE_DIR}` is not interpreted correctly by the shell. You should figure it out. If interpreted correctly, the `${FIRMWARE_DIR}` [...]

This solves my problem. I mis-configured the ${FIRMWARE_DIR} variable.