firmadyne/firmadyne #75: Network problem. Run some firmware images

VBukharov, 2018-03-25:
Is it possible to run 2 or more images on one machine? 
I run one firmware with ip: 192.168.0.1 and tried to run another (192.168.0.100).
I have problem with network:
[image]

I think problem is with tap\tun interfaces.

ddcc, 2018-03-25:
It's possible, but it'd be easiest to modify the `run.sh` script so that they are assigned separate TAP interfaces on separate subnets. I haven't tried having sharing the same interface and subnet between multiple images, even with non-conflicting IP address assignments.

VBukharov, 2018-03-25:
Thanks for a quick reply) 
As I see, TAP interfaces are different in run.sh: `TAPDEV_0=tap${IID}_0` 
I have an image with ip 192.168.1.1 (after inferNetwork.sh script) and an image with ip 192.168.0.100 and problem is the same.
They are in separate subnets with different TAP interfaces, am I right?

ddcc, 2018-03-25:
Yes, they should be entirely separate. The error from your screenshot above seems like it's coming from qemu, so you may want to check that each instance is binding to a separate interface. You probably also want to check the routing table to make sure that the proper rules have been added for each interface. Is it possible that one of those interfaces (tap*) is conflicting with your actual ethernet device (e.g. eth0)?

VBukharov, 2018-03-26:
I checked what you have said. I tried this case on remote machine to exclude conflicting interfaces.
So, first firmware with ip 192.168.0.100 and second with 192.168.1.1
`ip route` output before run firmwares
`default via 88.151.112.1 dev eth0 onlink 
88.151.112.0/24 dev eth0  proto kernel  scope link  src 88.151.112.106 
169.254.0.0/16 dev eth0  scope link  metric 1000 
`
I run first firmware. `ip route` output after that:
`default via 88.151.112.1 dev eth0 onlink 
88.151.112.0/24 dev eth0  proto kernel  scope link  src 88.151.112.106 
169.254.0.0/16 dev eth0  scope link  metric 1000 
192.168.0.0/24 dev tap1_0  proto kernel  scope link  src 192.168.0.99 
192.168.0.100 via 192.168.0.100 dev tap1_0 
`
Then i run next firmware with 60s delay after creating tap device. `ip route` output:
`default via 88.151.112.1 dev eth0 onlink 
88.151.112.0/24 dev eth0  proto kernel  scope link  src 88.151.112.106 
169.254.0.0/16 dev eth0  scope link  metric 1000 
192.168.0.0/24 dev tap1_0  proto kernel  scope link  src 192.168.0.99 
192.168.0.100 via 192.168.0.100 dev tap1_0 
192.168.1.0/24 dev tap74_0  proto kernel  scope link  src 192.168.1.2 linkdown 
192.168.1.1 via 192.168.1.1 dev tap74_0 linkdown 
`
and when qemu tries to start i have the same error:
`bind: Address already in use
qemu-system-mips: -net socket,vlan=1,listen=:2001: Device 'socket' could not be initialized
`
May be something wrong with `linkdown` in output?

ddcc, 2018-03-27:
Can you post the two `-net` arguments being passed to qemu? Are the two sockets unique? You could also try using strace to look for the specific invocations of the `bind` syscall that are failing.

VBukharov, 2018-04-21:
Problem solved. I just remove 
`    -net nic,vlan=1 
    -net socket,vlan=1,listen=:2001 -net nic,vlan=2 
    -net socket,vlan=2,listen=:2002 -net nic,vlan=3 
    -net socket,vlan=3,listen=:2003
`
in each run.sh.
Thanks for help!

closed 2018-04-21 by VBukharov