firmadyne/firmadyne #76: Firmware "WNAP320_V3.7.9.0_firmware",infernetwork error.

ftk-sostupid, 2018-04-16:
When I emulate the "WNAP320_V3.7.9.0_firmware"(Download link http://support.netgear.cn/Upfilepath/WNAP320_V3.7.9.0_firmware.tar), I got this error:
[image]
If I ignored this and run the ./scratch/2/run.sh, it shows like:
[image]
And then it just stopped, the emulation failed either.
Before this I emulated the example(WNAP320 Firmware Version 2.0.3.zip) soon after.
This caused by the firmware's file format or some other errors? How can I resolve this?
Sorry for my bad English.

ftk-sostupid, 2018-04-17:
I've resolved this problem. The firmware format should be tar or zip, and inside should be like:
[image]
I unziped the former firmware, then the right firmware file comes to me.

closed 2018-04-17 by ftk-sostupid

ddcc, 2018-04-17:
Glad you figured it out. The extractor script should take care of storing the firmware in the correct format.