firmadyne/firmadyne #77: Failed to create the vlan

shellcong, 2018-04-25:
For each run.sh script, it always fails if it need to create the vlan.  Do I need to config a specific network interface name (like "eth0") in my host? Any ideas to solve this problem? Thank you.

The error output is : 

Set 'tap3_0' persistent and owned by uid 0
Initializing VLAN...
**RTNETLINK answers: Network is down**

The run.sh script is:

TAPDEV_0=tap${IID}_0
HOSTNETDEV_0=${TAPDEV_0}
echo "Creating TAP device ${TAPDEV_0}..."
sudo tunctl -t ${TAPDEV_0} -u ${USER}

echo "Initializing VLAN..."
HOSTNETDEV_0=${TAPDEV_0}.3
sudo ip link add link ${TAPDEV_0} name ${HOSTNETDEV_0} type vlan id 3
sudo ip link set ${HOSTNETDEV_0} up

My network config:

ens33     Link encap:Ethernet  HWaddr 00:0c:29:d7:22:dc
          inet addr:172.16.135.130  Bcast:172.16.135.255  Mask:255.255.255.0
          inet6 addr: fe80::4737:97e8:726b:ed15/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:326074 errors:0 dropped:0 overruns:0 frame:0
          TX packets:130090 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:404147408 (404.1 MB)  TX bytes:11166345 (11.1 MB)

ddcc, 2018-04-27:
Hmm, it looks like VLAN support has actually been broken for a while. Try modifying the relevant part of the run.sh script as follows, then rerun it. Does that fix the bug?

```
echo "Initializing VLAN..."
sudo ip link add link ${TAPDEV_0} name ${HOSTNETDEV_0} type vlan id 3
HOSTNETDEV_0=${TAPDEV_0}.3
sudo ip link set ${HOSTNETDEV_0} up
```

hmj520, 2019-05-08:
hello，I used your sample, but the result is as follows. May I ask what is the reason?
root@ubuntu:/home/hmj/firmadyne# ./scratch/1/run.sh
Creating TAP device tap1...
Set 'tap1' persistent and owned by uid 0
Bringing up TAP device...
Adding route to 192.168.0.100...
Starting emulation of firmware... Done!
The emulated firmware may not be accessible while booting.
Press any key to destroy the network and shutdown emulation.
fDeleting route...
Bringing down TAP device...
Deleting TAP device tap1... Set 'tap1' nonpersistent
Done!

joeasds, 2019-12-07:
hello, I try the way to modify the run.sh, but there is a new question, the error output is:
Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 1000
Initializing VLAN...
RTNETLINK answers: File exists
Cannot find device "tap1_0.1"
Bringing up TAP device...
Cannot find device "tap1_0.1"
Cannot find device "tap1_0.1"
Adding route to 192.168.0.1...
Cannot find device "tap1_0.1"

ddcc, 2019-12-07:
For your case, you'd need to replace 3 with 1 in the snippet above. Does that work?

joeasds, 2019-12-07:
It doesn't work that I replace 3 with 1 in the snippet above.It's the resualt in above.Before I modify the run.sh, the error output is :
Creating TAP device tap1_0...
Set 'tap1_0' persistent and owned by uid 1000
Initializing VLAN...
RTNETLINK answers: Network is down
Bringing up TAP device...
RTNETLINK answers: Network is down
Adding route to 192.168.0.1...
RTNETLINK answers: Network is unreachable
Starting firmware emulation...

ddcc, 2019-12-08:
Sorry, looks like it's still not working then.

joeasds, 2019-12-09:
Thanks for your reply, may I try to use the #115 way? Can you tell me exactly how to do it, I don't understand how to modify IP route or other options. Thanks a lot.

ddcc, 2019-12-09:
Can you post your run.sh script?

joeasds, 2019-12-09:
This is the run.sh generated:
#!/bin/bash

set -u

ARCHEND=mipsel
IID=1

if [ -e ./firmadyne.config ]; then
    source ./firmadyne.config
elif [ -e ../firmadyne.config ]; then
    source ../firmadyne.config
elif [ -e ../../firmadyne.config ]; then
    source ../../firmadyne.config
else
    echo "Error: Could not find 'firmadyne.config'!"
    exit 1
fi

IMAGE=`get_fs ${IID}`
KERNEL=`get_kernel ${ARCHEND}`
QEMU=`get_qemu ${ARCHEND}`
QEMU_MACHINE=`get_qemu_machine ${ARCHEND}`
QEMU_ROOTFS=`get_qemu_disk ${ARCHEND}`
WORK_DIR=`get_scratch ${IID}`

TAPDEV_0=tap${IID}_0
HOSTNETDEV_0=${TAPDEV_0}
echo "Creating TAP device ${TAPDEV_0}..."
sudo tunctl -t ${TAPDEV_0} -u ${USER}

echo "Initializing VLAN..."
HOSTNETDEV_0=${TAPDEV_0}.1
sudo ip link add link ${TAPDEV_0} name ${HOSTNETDEV_0} type vlan id 1
sudo ip link set ${HOSTNETDEV_0} up

echo "Bringing up TAP device..."
sudo ip link set ${HOSTNETDEV_0} up
sudo ip addr add 192.168.0.2/24 dev ${HOSTNETDEV_0}

echo "Adding route to 192.168.0.1..."
sudo ip route add 192.168.0.1 via 192.168.0.1 dev ${HOSTNETDEV_0}

function cleanup {
    pkill -P $$
    
echo "Deleting route..."
sudo ip route flush dev ${HOSTNETDEV_0}

echo "Bringing down TAP device..."
sudo ip link set ${TAPDEV_0} down

echo "Removing VLAN..."
sudo ip link delete ${HOSTNETDEV_0}

echo "Deleting TAP device ${TAPDEV_0}..."
sudo tunctl -d ${TAPDEV_0}
}

trap cleanup EXIT

echo "Starting firmware emulation... use Ctrl-a + x to exit"
sleep 1s

 ${QEMU} -m 256 -M ${QEMU_MACHINE} -kernel ${KERNEL} \
    -drive if=ide,format=raw,file=${IMAGE} -append "root=${QEMU_ROOTFS} console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 rdinit=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 user_debug=31 firmadyne.syscall=0" \
    -nographic \
    -net nic,vlan=0 -net socket,vlan=0,listen=:2000 -net nic,vlan=1 -net socket,vlan=1,listen=:2001 -net nic,vlan=1 -net tap,vlan=1,id=net0,ifname=${TAPDEV_0},script=no -net nic,vlan=3 -net socket,vlan=3,listen=:2003 | tee ${WORK_DIR}/qemu.final.serial.log

And I try to change the ip route “sudo ip route add 192.168.0.1 via 192.168.0.1 dev ${HOSTNETDEV_0}”, replace 192.168.0.1 with 192.168.0.2, but it doesn't work...

ddcc, 2019-12-09:
You need to move `HOSTNETDEV_0=${TAPDEV_0}.1` after the first `sudo ip link ...` (switch the order with the next line). Make sure you run the script all the way to bring down the devices as well (don't kill it in the middle). If that still doesn't work, you can try changing the via part to `via 192.168.0.2`.

joeasds, 2019-12-10:
I meet some problem, if I switch the order with the next line, it will show:
RTNETLINK answers: File exists, and the tap1_0 will display "BROADCAST MULTICAST", there is not inet addr.
if I don't change it. it will show: RTNETLINK answers: Network is down, but it will display the ip addr of tap1_0 in the ifconfig. I try changing the via, but it doesn't work. May I change other of the run.sh,  I use the .bin of Dlink dir-645.

ddcc, 2019-12-10:
I don't understand your last sentence. But try this:
```
HOSTNETDEV_0=${TAPDEV_0}.1
sudo ip link add link ${TAPDEV_0} name ${HOSTNETDEV_0} type vlan id 1
sudo ip link set ${TAPDEV_0} up
```

joeasds, 2019-12-11:
Thank you for your reply! it does work.

ddcc, 2019-12-11:
Fixed in 7224391ce5416f9ac1032b773dccfcaada04cb01 .

closed 2019-12-11 by ddcc