firmadyne/firmadyne #78: Netgear firmware unable to simulate in the firmadyne

susantaroy2002, 2018-06-07:
Unable to simulate Netgear firmware, not if there any thing wrong on .chk  file extension. 

Here I am sharing some details about my blocker, 
1. Tried on 2 netgear firmware version - "D8500-V1.0.3.39_1.0.1.chk" and "R7000-V1.0.9.28_10.2.32.chk"
2. attaching the error screenshot. 

Please let me know if you need any further details to investigate this issue.

[image]
[image]
[image]

susantaroy2002, 2018-07-03:
Yeah, It work directly with firmadyne, however tested with dlink it works. But not sure about the NetGear. Any SME that you can redirect me.

ddcc, 2018-07-03:
I don't know what an SME is.

santhoshn16, 2020-01-29:
try dropping the db, user, create them again and follow the steps you will get it.