firmadyne/firmadyne #80: How to using instrumentation of kernel

nghiphu, 2018-06-13:
It's said that Linux kernel which has modified in firmadyne could instrumen system call. So i want to use this kernel for collecting system call of all processes executed in firmware. It will help us detect many problems. But i could not find any document talk how to use instrument ability of kernel, how to set an get system call log file when run firmware.

ddcc, 2018-06-13:
Have you seen e.g. https://github.com/firmadyne/kernel-v4.1/blob/firmadyne-v4.1.17/README.md ?

nghiphu, 2018-06-14:
I had known that. It’s said that setting system call arguments to 256 when excute qemu in run.sh file to get system call instrumentation info, but I don’t see any thing, even in log file of scratch folder. You can show me how to get system call log of all process with firmadyne

ddcc, 2018-06-14:
The mechanism should definitely be working, because it's being used to infer the networking configuration.

The run.sh script that you're modifying should be the generated one; e.g. `./scratch/1/run.sh`. It will contain the string `firmadyne.syscall=0`; just change that to the appropriate bitmask.

nghiphu, 2018-06-14:
Yes, i had tried. I tried to change some value 1,2,... and 256 for all but i didn’t see any change in log file in scratch folder. Where does firmadyne store system call log for process?

ddcc, 2018-06-14:
The system call is simply printed to the kernel log, which is mirrored to `qemu.final.serial.log`. For example, if set the bitmask to 1, you should see calls to `sys_reboot`, `do_mount`, `vfs_ioctl`, etc.

nghiphu, 2018-06-14:
it's done! thanks sm!

LeHoangHY, 2019-03-01:
can you explant bitmask of firmadyne.syscall value? what information I can get when set it is 1,2,3....?

ddcc, 2019-03-01:
The bitmasks are implemented here: https://github.com/firmadyne/kernel-v4.1/blob/firmadyne-v4.1.17/drivers/firmadyne/hooks.c#L21

Essentially, if you set the corresponding bit for one of these groups of syscalls, firmadyne will enable debug logging of associated syscalls' arguments to the system log each time one is called. It's a bitmask because you pass the bitwise-or of all the bits you want set in the firmadyne.syscall kernel parameter at startup.