firmadyne/firmadyne #82: Compiling from Source

nghiphu, 2018-07-09:
I'm trying to compiling kernel from source follow tuto in https://github.com/firmadyne/firmadyne, but it occured error: 
"checking for the correct version of gmp.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
ftp://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
+ die Failed to build gcc-5.3.0
+ echo Failed to build gcc-5.3.0
Failed to build gcc-5.3.0
+ exit 1
"

nghiphu, 2018-07-09:
When i used pre-built toolchains in https://cmu.app.boxcn.net/s/hnpvf1n72uccnhyfe307rc2nb9rfxmjp,
it occued this error:
" 
 CC      fs/unionfs/mmap.o
  CC      fs/unionfs/whiteout.o
  LD      fs/unionfs/unionfs.o
  LD      fs/unionfs/built-in.o
  LD      fs/built-in.o
Makefile:120: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
"
i'm using ubuntu 16.04

ddcc, 2018-07-09:
For the first error, it's likely that something in musl-cross has changed since I originally built it. You'll need to figure out if some configuration variables have been renamed or etc.

As for the second, you'll need to post the error from the failing sub-make step. Compilation occurs in parallel, so what you posted only shows what was built. The actual error is probably higher up.

ddcc, 2018-07-09:
You only need to post the actual error:

```
TIMEC kernel/timeconst.h
Can't use 'defined(@array)' (Maybe you should just omit the defined()?) at /home/np/kernel/kernel-v2.6.32/kernel/timeconst.pl line 373.
/home/np/kernel/kernel-v2.6.32/kernel/Makefile:129: recipe for target 'kernel/timeconst.h' failed
make[2]: *** [kernel/timeconst.h] Error 255
```

I'll suggest that if you plan to compile something as complicated as the Linux kernel, you should have some familiarity with C and Linux, or at least searching for error messages and trying to fix them. A better place to start would probably be to compile and run your own kernel for your current Linux machine, rather than a modified kernel for a research project.

For the current error, it looks like some Perl language feature was deprecated, and the compiler suggestion is the correct fix: https://lkml.org/lkml/2012/11/18/159 . I don't remember now if the 2.6.3x MIPS kernels were built using the 5.3.0 GCC toolchain, or a slightly older version.

nghiphu, 2018-07-09:
please try to find what gcc for compiling?

ddcc, 2018-07-14:
Fixed in https://github.com/firmadyne/kernel-v2.6.32/commit/82dda18ae0f2ba91d42c71ee30e72a16b89573cd

Screamfox, 2018-07-14:
This is my complete terminal output:

`
root@ns2:/Firmware/kernel-v2.6.32/build/mipsel# make ARCH=mips CROSS_COMPILE=/opt/cross/mipsel-linux-musl/bin/mipsel-linux-musl- O=./build/mipsel -j8 all
make -C /Firmware/kernel-v2.6.32 O=/Firmware/kernel-v2.6.32/build/mipsel/.
  GEN     /Firmware/kernel-v2.6.32/build/mipsel/Makefile
  CHK     include/linux/version.h
  CHK     include/linux/utsrelease.h
  SYMLINK include/asm -> include/asm-mips
  Using /Firmware/kernel-v2.6.32 as source for kernel
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/conmakehash
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/genksyms/lex.o
  HOSTCC  scripts/genksyms/genksyms.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/genksyms/parse.o
  HOSTCC  scripts/mod/mk_elfconfig
/Firmware/kernel-v2.6.32/scripts/kallsyms.c: In function ‘read_symbol’:
/Firmware/kernel-v2.6.32/scripts/kallsyms.c:112:4: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
    fgets(str, 500, in);
    ^~~~~~~~~~~~~~~~~~~
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/genksyms/genksyms
/Firmware/kernel-v2.6.32/scripts/mod/modpost.c: In function ‘get_markers’:
/Firmware/kernel-v2.6.32/scripts/mod/modpost.c:1562:4: warning: ignoring return value of ‘asprintf’, declared with attribute warn_unused_result [-Wunused-result]
    asprintf(&line, "%s\t%s\t%s\n", name, mod->name, fmt);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Firmware/kernel-v2.6.32/scripts/mod/modpost.c: In function ‘add_marker’:
/Firmware/kernel-v2.6.32/scripts/mod/modpost.c:1982:2: warning: ignoring return value of ‘asprintf’, declared with attribute warn_unused_result [-Wunused-result]
  asprintf(&line, "%s\t%s\t%s\n", name, mod->name, fmt);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  HOSTLD  scripts/mod/modpost
  CC      kernel/bounds.s
  GEN     include/linux/bounds.h
  CC      arch/mips/kernel/asm-offsets.s
  GEN     include/asm/asm-offsets.h
  CALL    /Firmware/kernel-v2.6.32/scripts/checksyscalls.sh
  CHK     include/linux/compile.h
  CC      init/main.o
  CC      init/do_mounts.o
  CC      init/do_mounts_rd.o
  CC      init/calibrate.o
  CC      init/initramfs.o
  CC      init/do_mounts_initrd.o
  HOSTCC  usr/gen_init_cpio
  CC      arch/mips/mti-malta/malta-amon.o
  CC      init/version.o
  CC      arch/mips/kernel/cpu-probe.o
  CC      arch/mips/mti-malta/malta-cmdline.o
  CC      arch/mips/mti-malta/malta-display.o
  CC      arch/mips/mm/cache.o
  GEN     usr/initramfs_data.cpio
  CC      arch/mips/mti-malta/malta-init.o
  CC      arch/mips/mti-malta/malta-int.o
  AS      usr/initramfs_data.o
  CC      arch/mips/mm/dma-default.o
  LD      usr/built-in.o
  CC      arch/mips/math-emu/cp1emu.o
  CC      arch/mips/kernel/branch.o
  AS      arch/mips/kernel/entry.o
  CC      arch/mips/mm/extable.o
  CC      arch/mips/mti-malta/malta-memory.o
  AS      arch/mips/kernel/genex.o
  LD      init/mounts.o
  LD      init/built-in.o
  CC      arch/mips/kernel/irq.o
[... 1507 lines cut ...]
  CC      fs/xfs/linux-2.6/xfs_lrw.o
  LD      fs/ocfs2/dlm/ocfs2_dlmfs.o
  LD      fs/ocfs2/dlm/built-in.o
  LD      fs/ocfs2/built-in.o
  CC      fs/xfs/linux-2.6/xfs_super.o
  CC      fs/xfs/linux-2.6/xfs_sync.o
  CC      fs/xfs/linux-2.6/xfs_xattr.o
  CC      fs/xfs/support/debug.o
  CC      fs/xfs/support/uuid.o
  LD      fs/xfs/xfs.o
  LD      fs/xfs/built-in.o
  LD      fs/built-in.o
Makefile:120: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
Makefile:19: recipe for target 'all' failed
make: *** [all] Error 2
`

- Gcc Version = G++ version
gcc (Ubuntu 7.2.0-8ubuntu3.2) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Rgds,

Screamfox, 2018-07-14:
@nghiphu 

How you have installed the gcc -5.3.0 ? 

rgds

closed 2019-12-10 by ddcc