firmadyne/firmadyne #83: Error when firmadyne try to do extraction file & errors in every next step comands !

Screamfox, 2018-07-12:
@FIRMADYNE

Hi Devs,

I have this issue when i try to extract zip filewhere we can find tha Amiko HD8150 firmware receiver incomporated in zip fil with _rzl_notes.txt_ and _AMIKO_HD8150_2.4.43_emu.abs_ .

Here he go the terminal output ;

```
`root@ns2:/Firmware/firmadyne# ./sources/extractor/extractor.py -b Amiko -sql 127.0.0.1 -np -nk "AMIKO_HD8150_2.4.43_emu.zip" images
>> Database Image ID: 2

/Firmware/firmadyne/AMIKO_HD8150_2.4.43_emu.zip
>> MD5: 17c266121d5568340ef927c211fcc4f4
>> Tag: 2
>> Temp: /tmp/tmpOqzn89
>> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
>>>> Zip archive data, at least v2.0 to extract, compressed size: 4454941, uncompressed size: 8388608, name: AMIKO_HD8150_2.4.43_emu.abs
>> Recursing into archive ...

/tmp/tmpOqzn89/_AMIKO_HD8150_2.4.43_emu.zip.extracted/rlz_notes.txt
        >> MD5: aa5c7dd374bf981442deb5fa67e7189e
        >> Skipping: text/plain...

/tmp/tmpOqzn89/_AMIKO_HD8150_2.4.43_emu.zip.extracted/AMIKO_HD8150_2.4.43_emu.abs
        >> MD5: dc6b86caa2c55d18a43ca9ad9053bc77
        >> Tag: 2
        >> Temp: /tmp/tmpBNDrbm
        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
        >> Recursing into archive ...
        >>>> Extraction failed!
        >>>> LZMA compressed data, properties: 0x6C, dictionary size: 8388608 bytes, uncompressed size: 11883876 bytes
        >> Recursing into compressed ...

/tmp/tmpBNDrbm/_AMIKO_HD8150_2.4.43_emu.abs.extracted/30080
                >> MD5: d20984f2f6c357c44979063654fc3849
                >> Tag: 2
                >> Temp: /tmp/tmpYvwvSH
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >>>> LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 18432 bytes
                >> Recursing into compressed ...

/tmp/tmpYvwvSH/_30080.extracted/8D4338
                        >> MD5: b36e778756ec82564479489bf7137838
                        >> Tag: 2
                        >> Temp: /tmp/tmpEAWxej
                        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                        >> Recursing into archive ...
                        >>>> Extraction failed!
                        >> Recursing into compressed ...
                        >> Cleaning up /tmp/tmpEAWxej...

/tmp/tmpYvwvSH/_30080.extracted/8D6358
                        >> MD5: 97643cd43f9041dddb1ca72cd4164df7
                        >> Tag: 2
                        >> Temp: /tmp/tmpvUcDNF
                        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                        >> Recursing into archive ...
                        >>>> Extraction failed!
                        >> Recursing into compressed ...
                        >> Cleaning up /tmp/tmpvUcDNF...

/tmp/tmpYvwvSH/_30080.extracted/8E7118
                        >> MD5: f2d87aa26cfc68b551537da72ec4b99e
                        >> Tag: 2
                        >> Temp: /tmp/tmpPhBIMG
                        >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                        >> Recursing into archive ...
                        >>>> Extraction failed!
                        >> Recursing into compressed ...
                        >> Cleaning up /tmp/tmpPhBIMG...
                >> Cleaning up /tmp/tmpYvwvSH...

/tmp/tmpBNDrbm/_AMIKO_HD8150_2.4.43_emu.abs.extracted/3B0080
                >> MD5: ee7c14835520aecfa0ca50063ab70577
                >> Tag: 2
                >> Temp: /tmp/tmpHqlSN5
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >> Recursing into compressed ...
                >> Cleaning up /tmp/tmpHqlSN5...

/tmp/tmpBNDrbm/_AMIKO_HD8150_2.4.43_emu.abs.extracted/560080
                >> MD5: 13bf1c71618a23f999f5579b93bac42a
                >> Tag: 2
                >> Temp: /tmp/tmpT1EhnI
                >> Status: Kernel: True, Rootfs: False, Do_Kernel: False,                 Do_Rootfs: True
                >> Recursing into archive ...
                >>>> Extraction failed!
                >> Recursing into compressed ...
                >> Cleaning up /tmp/tmpT1EhnI...
        >> Cleaning up /tmp/tmpBNDrbm...
>>>> Extraction failed!
>> Recursing into compressed ...

/tmp/tmpOqzn89/_AMIKO_HD8150_2.4.43_emu.zip.extracted/rlz_notes.txt
        >> MD5: aa5c7dd374bf981442deb5fa67e7189e
        >> Skipping: aa5c7dd374bf981442deb5fa67e7189e...

/tmp/tmpOqzn89/_AMIKO_HD8150_2.4.43_emu.zip.extracted/AMIKO_HD8150_2.4.43_emu.abs
        >> MD5: dc6b86caa2c55d18a43ca9ad9053bc77
        >> Skipping: dc6b86caa2c55d18a43ca9ad9053bc77...
>> Cleaning up /tmp/tmpOqzn89...`
```

But when i use this command i can extract correctly the .abs file so firmadyne developers can give me some help do performe my emulation with this firmware ?

```
$ binwalk -e AMIKO_HD8150_2.4.43_emu.abs 

DECIMAL       HEXADECIMAL     DESCRIPTION

--------------------------------------------------------------------------------
196736        0x30080         LZMA compressed data, properties: 0x6C, dictionary size: 8388608 bytes, uncompressed size: 11883876 bytes
3866752       0x3B0080        LZMA compressed data, properties: 0x6C, dictionary size: 8388608 bytes, uncompressed size: 3255512 bytes
5636224       0x560080        LZMA compressed data, properties: 0x6C, dictionary size: 8388608 bytes, uncompressed size: 87904 bytes
```
Or i can get other files with same *.abs (Firmware files) "list below" with Ali Main Code Decompressor also knowed ([FirmwareDecrypter v8.9](http://www.satedu.cba.pl/index.php?action=downloadfile&filename=FirmwareDecrypter_v8.9.zip&directory=Test%20Folder&))

Other tested Ali Tools, Ali__tools_Console_v4.0 ([Ali tools Console v4.0 software](http://www.satedu.cba.pl/index.php?action=downloadfile&filename=Ali__tools_Console_v4.0__CRC_FIXER.rar&directory=Test%20Folder&))

    `Software used: Ali Main Code Decrypter 8.9`

```
Files unpacked:

    bootloader
    MemCfg
    maincode(AV)
    seecode
    default_lang
    cipluskey
    countryband
    logo_user
    logo_menu
    logo_radio
    logo_boot
    patch
    defaultdb(PRC)
    userdb(64+64)
```
`

### Other Addicional info,

```
# hexdump -C 
...
..
.
00b51a30  00 00 00 00 4c 69 62 63  6f 72 65 20 76 65 72 73  |....Libcore vers|
00b51a40  69 6f 6e 20 31 33 2e 31  36 2e 30 40 53 44 4b 34  |ion 13.16.0@SDK4|
00b51a50  2e 30 66 61 2e 31 33 2e  31 36 5f 32 30 31 36 31  |.0fa.13.16_20161|
00b51a60  30 31 39 28 67 63 63 20  76 65 72 73 69 6f 6e 20  |019(gcc version |
00b51a70  33 2e 34 2e 34 20 6d 69  70 73 73 64 65 2d 36 2e  |3.4.4 mipssde-6.|
00b51a80  30 36 2e 30 31 2d 32 30  30 37 30 34 32 30 29 28  |06.01-20070420)(|
00b51a90  41 64 6d 69 6e 69 73 74  72 61 74 6f 72 40 20 46  |Administrator@ F|
00b51aa0  72 69 2c 20 4a 75 6c 20  32 38 2c 20 32 30 31 37  |   ri, Jul 28, 2017   |
00b51ab0  20 31 32 3a 35 33 3a 32  38 20 41 4d 29 0a 00 00 | 12:53:28 AM)...    |
00b51ac0  44 4d 58 5f 53 33 36 30  31 5f 30 00 00 a1 03 18  |  DMX_S3601_0  .....|
```
When I use readelf it says this file isn't an ELF file. 

So I tried to open the firmware in the qemu (Win64) but it doesn't start the emulation.
```
C:\Program Files\qemu>qemu-system-mips.exe -machine mips -cpu mips32r6-generic -drive file=C:\30080.bin,index=0,media=disk,format=raw

qemu-system-mips.exe: warning: could not load MIPS bios 'mips_bios.bin'
```
With my best regards,
David Martins alias Screamfox

ddcc, 2018-07-12:
Does binwalk actually extract a valid root filesystem? Because the extractor script is picking up the same LZMA data as binwalk.

Screamfox, 2018-07-12:
When i use binwalk directly with this command, 

`binwalk -e`

the command extract correctly 3 files from the firmware (*.abs) file so probably a issue with the script or some command i need to add to enable the script run and extract completely the files.

i leaved the files in the precedent message to help @FIRMADYNE to reproduce the error. 

rgds,

Screamfox, 2018-07-12:
so do you suggest i report the bug to binwalk ? because the *.abs files are using LZAM compression mode but when the files are extracted with Ali decompressor i have a diferente root file system. so probably binwalk don't decompress the *.abs file correctly so probably is better to reporting this, what did you think ?

Rgds,

ddcc, 2018-07-13:
I think there's some confusion regarding the term "root filesystem". For the purposes of firmadyne, which emulates Linux-based firmware, we only care about a Linux root filesystem (e.g. /bin, /etc, /var, etc.). The list of unpacked files that you have above don't have this directory structure, so it doesn't appear to be Linux-based, and cannot be emulated by firmadyne.

Screamfox, 2018-07-13:
so you don't have a idea how can i "load" this files with something like firmadyne or qemu ?
or simule a NOR or NAND flashed with this files and load the NOR or NAND on a emulator like qemu or firmadyne or also kvm ?

ddcc, 2018-07-13:
It's not Linux-based, so firmadyne doesn't support it.

Generally, emulating embedded devices isn't straightforward because they're not plug-and-play like x86. Unless the emulator happens to support the exact hardware the firmware is for, it's not going to work.

Screamfox, 2018-07-13:
@ddcc thank you so much to help me so i Googled a little and i a found a other emulator can probably run this Non linux embedded firmware. 

http://www.unicorn-engine.org

well i thing is the Unicorn framework seems like the way ahead for now .

thank you so much for the support.

closed 2018-07-13 by Screamfox