firmadyne/firmadyne #85: Kernel panic - not syncing: Attempted to kill init

yindeming, 2018-08-08:
I'm running this firmware:
`[?] Enter the name or absolute path of the firmware you want to analyse : IPC_REALTEK_LDS_LINUX_V2.07.bin
[?] Enter the brand of the firmware : lds
[+] Now going to extract the firmware. Hold on..
[+] Firmware : IPC_REALTEK_LDS_LINUX_V2.07.bin
[+] Brand : lds
[+] Database image ID : 2
[+] Identifying architecture
[+] Architecture : mipsel
[+] Storing filesystem in database
[!] Filesystem already exists
[+] Building QEMU disk image
[+] Setting up the network connection, please standby
[+] Network interfaces : []
[+] Running the firmware finally
[+] command line : sudo /root/Downloads/firmware-analysis-toolkit/firmadyne/scratch/2/run.sh
[*] Press ENTER to run the firmware...
Starting firmware emulation... use Ctrl-a + x to exit
`

And later it give error:
`[    2.868000] init/1: potentially unexpected fatal signal 4.
[    2.868000] 
[    2.872000] Cpu 0
[    2.872000] $ 0   : 00000000 1000a400 00000000 00000000
[    2.876000] $ 4   : 0000000a 7fa42e90 00000000 7fa42e90
[    2.880000] $ 8   : 00000000 2aaaf2b4 0000096c 00000000
[    2.884000] $12   : 00401ca4 7fa42e80 00401974 00000000
[    2.884000] $16   : 2ab3e890 00000000 2ab3f191 2ab21da1
[    2.892000] $20   : 2ab3ab10 2ab21dd0 2ab21dd8 00000004
[    2.892000] $24   : 2ab23ba0 2ab0b6b0                  
[    2.896000] $28   : 2ab42b30 7fa42e30 7fa42f28 2aaf4d10
[    2.900000] Hi    : 000000d3
[    2.900000] Lo    : 000c092d
[    2.904000] epc   : 2aaf4db8 0x2aaf4db8
[    2.908000]     Not tainted
[    2.908000] ra    : 2aaf4d10 0x2aaf4d10
[    2.912000] Status: 0000a413    USER EXL IE 
[    2.912000] Cause : 00800028
[    2.912000] PrId  : 00019300 (MIPS 24Kc)
[    2.924000] Kernel panic - not syncing: Attempted to kill init!
`

I've tried to delete the corresponding folder in scrach, and run the makeImage.sh script, but the script gives error:
`./makeImage.sh: line 15: $1: unbound variable`

Would like some help on this issue. Thanks.

ddcc, 2018-08-08:
Have you looked at the [usage instructions](https://github.com/firmadyne/firmadyne#usage) for the sequence and arguments of the scripts? You need to pass the image id as a parameter to makeImage.sh .

ddcc, 2018-08-08:
But it looks like the issue with the init process will require some manual debugging, so it won't be straightforward to emulate this firmware.

yindeming, 2018-08-08:
Thank you Dominic. I've tried the v2.0.3 for Netgear WNAP320 firmware, works fantastic.

I followed your advice and tried again, and got a different error:
> [    2.492000] UDF-fs: No partition found (1)
[    2.496000] List of all partitions:
[    2.500000] 1f00            1024 mtdblock0 (driver?)
[    2.504000] 1f01            1024 mtdblock1 (driver?)
[    2.508000] 1f02            1024 mtdblock2 (driver?)
[    2.512000] 1f03            1024 mtdblock3 (driver?)
[    2.516000] 1f04            1024 mtdblock4 (driver?)
[    2.516000] 1f05            1024 mtdblock5 (driver?)
[    2.520000] 1f06            1024 mtdblock6 (driver?)
[    2.520000] 1f07            1024 mtdblock7 (driver?)
[    2.524000] 1f08            1024 mtdblock8 (driver?)
[    2.528000] 1f09            1024 mtdblock9 (driver?)
[    2.528000] 1f0a          120832 mtdblock10 (driver?)
[    2.532000] 0800         1048576 sda driver: sd
[    2.536000]   0801         1047552 sda1
[    2.536000] No filesystem could mount root, tried:  ext3 ext2 ext4 cramfs squashfs vfat iso9660 romfs udf
[    2.540000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(8,1)

After some digging, it seems that the kernel cannot mount the / partition due to missing drivers. If that's firmware problem, please just close the issue.

ddcc, 2018-08-09:
That error suggests that the filesystem was not created correctly, or somehow became corrupt, which would be solved by deleting the scratch directory and rerunning the makeImage.sh script.

I would guess that the first error you encountered is the actual emulation error, and the current one is due to the filesystem not being created correctly when you tried again.

liuliqaz, 2020-11-16:
same problem, can it be solved? thanks a lot