firmadyne/firmadyne #86: Firmware can't taking IP on Ubuntu 18.04

HtHuanChen, 2018-08-15:
I tried to use the `./scripts/inferNetwork.sh` to set up the network connection ,but I only got an error and timeout.

```
$ sudo ./scripts/inferNetwork.sh 2 mipsel
Running firmware 2: terminating after 60 secs...
qemu-system-mipsel: -net nic,vlan=0: 'vlan' is deprecated. Please use 'netdev' instead.
qemu-system-mipsel: terminating on signal 2 from pid 8732 (timeout)
```
And the version of `qemu-system-mipsel` is:
```
$ qemu-system-mipsel -version
QEMU emulator version 2.11.1(Debian 1:2.11+dfsg-1ubuntu7.4)
Copyright (c) 2003-2017 Fabrice Bellard and the QEMU Project developers
```

Appreciate your assist on where i can start troubleshooting.

ddcc, 2018-08-15:
Have you tried the [troubleshooting steps](https://github.com/firmadyne/firmadyne#runsh-is-not-generated)?

HtHuanChen, 2018-08-23:
Emmmm, sorry, I overlooked it. The problem have now been sorted. You can close the issue.
Thanks for your help.

closed 2018-08-23 by HtHuanChen

c0mix, 2018-08-29:
@HtHuanChen I have the same problem, how did you solved it?